esphome/issues #1689: AC_DIMMER no on/off

CSnipper, 2020-12-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome and HA supervised on debian

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini pro

**ESPHome version (latest production, beta, dev branch)**

Latest

**Affected component:**

AC_DIMMER

**Description of problem:**
All configured and I can't use it as a simple on/off switch. Tried mode: leading, INPUT, INPUT_PULLUP,inverted: yes and no

**Problem-relevant YAML-configuration entries:**
```yaml

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: D6
    method: leading
    zero_cross_pin:
      number: D7
      mode: INPUT_PULLUP
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: Dimmer

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-12-15:
This works reasonably well on my esp8266 / robotdyn combo. It does flicker a bit sometimes when changing brightness but it's good enough and it can turn on and off the light no problem.

Can you provide logs?

Set log level to VERBOSE and make sure you see the mains frequency reported properly. e.g.:

````
[20:25:00][C][ac_dimmer:200]: AcDimmer:
[20:25:00][C][ac_dimmer:201]:   Output Pin: GPIO12 (Mode: OUTPUT)
[20:25:00][C][ac_dimmer:202]:   Zero-Cross Pin: GPIO13 (Mode: INPUT_PULLUP, INVERTED)
[20:25:01][C][ac_dimmer:203]:    Min Power: 10.0%
[20:25:01][C][ac_dimmer:204]:    Init with half cycle: YES
[20:25:01][C][ac_dimmer:208]:    Method: leading
[20:25:01][V][ac_dimmer:213]:   Estimated Frequency: 49.955Hz
````

rradar, 2020-12-25:
I actually had same problem initally using the provide example code form the docs. Playing around with the config and with help from the discord channel gave a satisfying results for dimmable lights (including turning it totally off).

Can you try adding this config @CSnipper and report back if it works?
```yaml
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: D6
    zero_cross_pin:
      number: D6
      mode: INPUT
      inverted: yes
    init_with_half_cycle: true
```
Specially the last line could be the  :key:  I think.

closed 2021-05-08 by stale[bot]