esphome/issues #1694: ESPHome not connecting to SSID with spaces

rpvelloso, 2020-12-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):*pip* 

**ESP (ESP32/ESP8266, Board/Sonoff):*esp01* 

**ESPHome version (latest production, beta, dev branch)*v1.15.0*

**Affected component:*WIFI*

**Description of problem:*ESPHome does not connects to an SSID with spaces. I have 3 APs, two have spaces in the SSIDs and ESPHome does not connect on these two APs, only to the AP with no spaces in the SSID. I have supplied only one AP at a time to test it. I've tested all SSIDs with Tasmota and works fine with the three APs (so the problem is not in the ESP01 board, but in the firmware)*

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  ssid: "Casa"
  password: "thisSSIDworksFine"

wifi:
  ssid: "Casa Sala"
  password: "thisSSIDdoesNot"
```

**Logs (if applicable):**

```
no logs :(

```

**Additional information and things you've tried:**

rradar, 2020-12-14:
not sure but did you already tried something like this @rpvelloso ?

```yaml
wifi:
  ssid: "Casa%20Sala"
  password: "doesThisSSDnowWORK?"
```

rpvelloso, 2020-12-14:
@rradar Nope, I have tried "Casa\x20Sala" and "Casa\ Sala", I'll try %20

rpvelloso, 2020-12-14:
@rradar same problem, no success :(

alexyao2015, 2020-12-15:
Its definitely working for me... I think it must be an issue with your setup.

rpvelloso, 2020-12-15:
I've checked the generated main.cpp file from my config and it is correctly generated, but it does not connect. I'll try to grab the logs from serial interface.

rpvelloso, 2020-12-15:
```  wifi_wificomponent = new wifi::WiFiComponent();
  wifi_wificomponent->set_use_address("ir_sala.local");
  wifi::WiFiAP wifi_wifiap = wifi::WiFiAP();
  wifi_wifiap.set_ssid("Casa Sala");
  wifi_wifiap.set_password("password");
  wifi_wifiap.set_priority(0.0f);
  wifi_wificomponent->add_sta(wifi_wifiap);
  wifi_wificomponent->set_reboot_timeout(900000);
  wifi_wificomponent->set_power_save_mode(wifi::WIFI_POWER_SAVE_NONE);
  wifi_wificomponent->set_fast_connect(false);
  wifi_wificomponent->set_output_power(20.0f);
  App.register_component(wifi_wificomponent);
```

alexyao2015, 2020-12-15:
That is very strange. Been using esphome for over 2 years now. My SSID had always had spaces and never once did I have an issue connecting. Likely due to some other setting. 

Try removing the password from the network with spaces and try connecting

rpvelloso, 2020-12-15:
Just checked the serial logs and it is running properly. For some reason the esphome device is not listed at the AP's DHCP client list (no ideia why), because of that I thought it was not connected to the AP, my bad. Sorry for the inconvenience. Closing the issue.

closed 2020-12-15 by rpvelloso