esphome/issues #1696: Teleinfo linky standard mode: bad crc

Mryck, 2020-12-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.16.0-dev

**Affected component:**

teleinfo : https://next.esphome.io/components/sensor/teleinfo.html

**Description of problem:**

I use the PiTInfo with the ESP32, the PiTInfo is connected to the Linky counter and the Linky is in Standard mode.
Then in the log I have arc 
**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: teleinfo
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "SSID"
  password: "***"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Teleinfo Fallback Hotspot"
    password: "***"

uart:
  id: uart_bus
  rx_pin: GPIO3
  baud_rate: 9600
  parity: EVEN
  data_bits: 7

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s
    unit_of_measurement: dB
    accuracy_decimals: 0
    force_update: false
    icon: mdi:wifi
  - platform: teleinfo
    tags:
     - tag_name: "ADCO"
       sensor:
        name: "ADCO"
        unit_of_measurement: ""
        icon: mdi:eye
     - tag_name: "BASE"
       sensor:
        name: "Index"
        unit_of_measurement: "kWh"
        icon: mdi:home-analytics
        filters:
          - multiply: 0.001
     - tag_name: "ISOUSC"
       sensor:
        name: "Intensité souscrite"
        unit_of_measurement: "A"
        icon: mdi:information
     - tag_name: "IMAX"
       sensor:
        name: "Intensité maximale"
        unit_of_measurement: "A"
        icon: mdi:information
     - tag_name: "IINST"
       sensor:
        name: "Intensité"
        unit_of_measurement: "A"
        icon: mdi:power-plug
     - tag_name: "PAPP"
       sensor:
        name: "Puissance"
        unit_of_measurement: "VA"
        icon: mdi:flash
    update_interval: 2s

binary_sensor:
  - platform: status
    name: "Status"

captive_portal:

web_server:
  port: 80

# Enable logging
logger:     
  baud_rate: 0   # disable logging via UART, help to avoid numerous crash with ESP_LOGD
  level: INFO

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
[17:23:49][I][app:105]: ESPHome version 1.16.0-dev compiled on Dec 15 2020, 17:14:08
[17:23:51][E][teleinfo:036]: bad crc: got 60 except 72
[17:23:53][W][teleinfo:058]: Internal buffer full
[17:23:55][E][teleinfo:036]: bad crc: got 37 except 82
[17:23:57][E][teleinfo:036]: bad crc: got 48 except 79
[17:23:59][E][teleinfo:036]: bad crc: got 95 except 55
[17:24:01][E][teleinfo:036]: bad crc: got 48 except 79
[17:24:03][E][teleinfo:036]: bad crc: got 60 except 72
[17:24:05][W][teleinfo:058]: Internal buffer full
[17:24:07][E][teleinfo:036]: bad crc: got 39 except 84
[17:24:09][E][teleinfo:036]: bad crc: got 48 except 79
[17:24:11][E][teleinfo:036]: bad crc: got 95 except 55

```

**Additional information and things you've tried:**

I tried to use an other rx pin (GPIO16) but same issue.

Mryck, 2020-12-19:
I found the problem, first I switched to an D1 mini then I use the correct label for the standard protocol.

closed 2020-12-19 by Mryck

Zodiac5, 2021-03-11:
I have exactly the same problem, just wondering if you can expand on how you fixed it. I'm not sure what you mean by "correct label for the standard protocol" If you can post your working config that would be a great help to me.

Mryck, 2021-03-12:
Hi sure here is the config I use for the sensor

```
uart:
  id: uart_bus
#  tx_pin: GPIO15
  rx_pin: GPIO3
  baud_rate: 9600
  parity: EVEN
  data_bits: 7
  stop_bits: 1
  
sensor:
  - platform: teleinfo
    tags:
     - tag_name: "ADSC"
       sensor:
        name: "ADSC"
        unit_of_measurement: ""
        icon: mdi:eye
     - tag_name: "EAST"
       sensor:
        name: "Index"
        unit_of_measurement: "kWh"
        icon: mdi:home-analytics
        accuracy_decimals: 3
        filters:
          - multiply: 0.001
     - tag_name: "EAST"
       sensor:
        name: "Index (Wh)"
        unit_of_measurement: "Wh"
        icon: mdi:home-analytics
     - tag_name: "PREF"
       sensor:
        name: "Intensité souscrite"
        unit_of_measurement: "A"
        icon: mdi:information
     - tag_name: "SINSTS"
       sensor:
        name: "Puissance"
        unit_of_measurement: "VA"
        icon: mdi:flash
    update_interval: 2s

# Enable Home Assistant API
api:

ota:

logger:     
  baud_rate: 0   # disable logging via UART, help to avoid numerous crash with ESP_LOGD
  level: INFO
```

Also I use the PiTinfo and I had to solder a 1k resistance between GND and pin 3 of the LTV814 (in parallel with R1)

Hope it helps

baramuse, 2021-04-20:
Hi,

I just hit the same issue.
First I had the issue using a D1 Mini and not setting the correct labels, then it's been working fine thanks to the tip.

Although now I tried to use a Shelly1 v3 instead of the D1Mini and I'm getting the "Bad CRC" errors, witht the same conf.
I plugged back in the D1 Mini and it worked again.
So looks like it's related to the Shelly1 v3, but both are ESP8622, I'm using the same RX pin, and 3.3v.
I'm using the PiTinfo 1.3.

Here's my conf :
```
uart:
  id: uart_bus
  rx_pin: RX
  baud_rate: 9600
  parity: EVEN
  data_bits: 7
  stop_bits: 1

sensor:
  - platform: teleinfo
    tags:
     - tag_name: "SINSTS"
       sensor:
        name: "${device_name}_puissance_apparente"
        unit_of_measurement: "VA"
        icon: mdi:flash
    update_interval: 15s
    uart_id : uart_bus
  - platform: wifi_signal
# Enable logging
logger:
  baud_rate: 0
```

Is that something to do with the 1k resistance you've added ?

Thanks a lot !

Zodiac5, 2021-04-20:
Thanks for the info. I didn't get it working with the d1 mini, so would be interested to know if you manage it, post an update if you do :)

baramuse, 2021-04-20:
Fot the D1Mini (genuine) here's my config : 
```
esphome:
  name: d1_mini_01
  platform: ESP8266
  board: d1_mini

# Example configuration entry
uart:
  id: uart_bus
  rx_pin: RX
  baud_rate: 9600
  parity: EVEN
  data_bits: 7

sensor:
  - platform: teleinfo
    tags:
     - tag_name: "SINSTS"
       sensor:
        name: "puissance_apparente"
        unit_of_measurement: "VA"
        icon: mdi:flash
     - tag_name: "SINSTS"
       sensor:
        name: "puissance_apparente_kva"
        unit_of_measurement: "KVA"
        icon: mdi:flash  
        accuracy_decimals: 1
        filters:
          - multiply: 0.001
     - tag_name: "EAIT" #Energie active injectée totale
       sensor:
         name: "energie_active_injectee_totale"
         unit_of_measurement: "Wh"
         icon: mdi:flash
     - tag_name: "SINSTI" #Puissance app. Instantanée injectée
       sensor:
         name: "puissance_apparente_intantanee_injectee"
         unit_of_measurement: "VA"
         icon: mdi:flash
    update_interval: 5s
    uart_id : uart_bus
  - platform: wifi_signal
    name: "Teleinfo WiFi Signal"
    update_interval: 60s
    unit_of_measurement: dB
    accuracy_decimals: 0
    force_update: false
    icon: mdi:wifi
  - platform: uptime
    name: "Teleinfo uptime"
    update_interval: 60s
    unit_of_measurement: s
    accuracy_decimals: 0
    force_update: false
    icon: mdi:timer

binary_sensor:
  - platform: status
    name: "Teleinfo status"

# Enable logging
logger:
 hardware_uart : UART1

# Enable Home Assistant API
api:

ota:
```
FYI : I've used the UART1 for logging as it was too much to handle for the esp8266 when using a software uart and UART0 was already "taken" by the teleinfo component.
Apart from that, I've patched the 3.3v, GND and TX pins to the mini's 3.3v, GND and RX pin and tha'ts it.
What kind of octocoupler do you use between the D1 and the meter ?