esphome/issues #1698: Is the Home Assistant API blocking? missing-template

lancelon, 2020-12-16:
Sorry, I know this isn't a 'bug' but it's very much an issue I'm facing. Struggling to find an answer anywhere.

I have an odd situation whereby my ESPHome node (which is an ESP32-POE) stops responding to two of the buttons.

Nine times out of ten the buttons work perfectly. And whenever I come into the building after hours to try and debug what on Earth is going on I can never reproduce the issue.

I'm using the internal pullups and an API connection to Home Assistant.

I was wondering if my network connection is iffy (which it is, I suspect the main router is faulty as I often can't access the web GUI and the internet connection fairly often fails - say twice a day during the busy part of the day) could it be that the ESPHome node stops responding to button presses because the API connection to Home Assistant is blocking?

Oddly though the ESP32-POE and Home Assistant are both strung off the same unmanaged POE switch, so I didn't really think the main router would sort of 'get involved' or inhibit their communication.

Thank you!

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Flashed with PIP on Mac
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-POE
**ESPHome version (latest production, beta, dev branch)**

1.15.3
**Affected component:**

The whole thing....!

**Description of problem:**
ESP32-POE stops responding to button presses BUT continues to listen out for and respond to MQTT commands.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: traffic
  platform: ESP32
  board: esp32-poe

mqtt:
  broker: 192.168.1.10
  client_id: TL
  topic_prefix: Traffic
  will_message:
    topic: traffic/avail
    payload: offline
  on_message:
    - topic: traffic/green/on
      then:
        - switch.turn_on: relay1
        - switch.turn_on: gindicator
    - topic: traffic/green/off
      then:
        - switch.turn_off: relay1
        - switch.turn_off: gindicator

    - topic: traffic/red/on
      then:
        - switch.turn_on: relay2
        - switch.turn_on: rindicator
    - topic: traffic/red/off
      then:
        - switch.turn_off: relay2
        - switch.turn_off: rindicator

logger:

api:
  password: "x"

ota:
  password: "x"

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO12

switch:
  - platform: gpio
    name: "Green Light"
    pin: 13
    id: relay1
    interlock: [relay2]
    on_turn_on:
    - switch.turn_on: gindicator
    on_turn_off:
    - switch.turn_off: gindicator
  - platform: gpio
    name: "Red Light"
    id: relay2
    pin: 15
    interlock: [relay1]
    on_turn_on:
    - switch.turn_on: rindicator
    on_turn_off:
    - switch.turn_off: rindicator
  - platform: gpio
    name: "Buzzer"
    id: buzzer
    pin: 14
  - platform: gpio
    name: "Green Indicator"
    pin: 3
    id: gindicator
    interlock: [rindicator]
  - platform: gpio
    name: "Red Indicator"
    pin: 2
    id: rindicator
    interlock: [gindicator]

binary_sensor:
  - platform: gpio
    pin:
      number: 39
      mode: INPUT_PULLUP
      inverted: True
    name: "Buzzer"
    filters:
      - delayed_on: 5ms
    on_press:
      then:
        - switch.turn_on: buzzer
    on_release:
      then:
        - switch.turn_off: buzzer
  - platform: gpio
    pin:
      number: 36
      mode: INPUT_PULLUP
      inverted: True
    name: "ALLOW ACCESS"
    filters:
      - delayed_on: 5ms
    on_press:
      then:
        - switch.turn_on: relay1
        - switch.turn_on: gindicator
  - platform: gpio
    pin:
      number: 34
      mode: INPUT_PULLUP
      inverted: True
    name: "DENY ACCESS"
    filters:
      - delayed_on: 5ms
    on_press:
        - switch.turn_on: relay2
        - switch.turn_on: rindicator

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ronytomen, 2020-12-17:
If you have the means, you should leave a active log running so you can see what is occurring when the problem happens.  You can tail to logs realtime from the CLI utilities for ESPHome over the network.

lancelon, 2020-12-17:
@ronytomen thank you I have done this - presume you mean using MQTT? I have managed to isolate an instance where the buttons were not working. This was at 2:09pm. Looking at the MQTT log for the deny_access binary sensor, there is no debug message sent at 2:09. However the button started responding again a minute later and so for 2L10pm there IS a message posted to the MQTT debug for the binary sensor.

I am minded to remove the HA API access and fall back to MQTT - see if that clears the problem up.

Interestingly the buzzer continues to work throughout any button blackouts so I am wondering if it could be related to the interlock somehow?!

Not sure. But it is a frustrating intermittent fault.

closed 2021-05-01 by stale[bot]