esphome/issues #1702: Dev build crashes !!!!

SeByDocKy, 2020-12-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

Building process of the dev is now crashing : 

```

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
File "/usr/lib/python3.6/distutils/core.py", line 163, in setup
        raise SystemExit("error: " + str(msg))
    SystemExit: error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
 
    During handling of the above exception, another exception occurred:
 
    Traceback (most recent call last):
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1163, in run_setup
        run_setup(setup_script, args)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 253, in run_setup
        raise
      File "/usr/lib/python3.6/contextlib.py", line 99, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python3.6/contextlib.py", line 99, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 166, in save_modules
        saved_exc.resume()
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 141, in resume
        six.reraise(type, exc, self._tb)
      File "/usr/lib/python3/dist-packages/setuptools/_vendor/six.py", line 685, in reraise
        raise value.with_traceback(tb)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 154, in save_modules
        yield saved
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 250, in run_setup
        _execfile(setup_script, ns)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 45, in _execfile
        exec(code, globals, locals)
      File "/tmp/easy_install-6y4962jd/cffi-1.14.4/setup.py", line 242, in <module>
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 163, in setup
        raise SystemExit("error: " + str(msg))
    SystemExit: error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
 
    During handling of the above exception, another exception occurred:
 
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-bwmzjp_w/cryptography/setup.py", line 121, in <module>
        "src/_cffi_src/build_padding.py:ffi",
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 128, in setup
        _install_setup_requires(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 123, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 513, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 774, in resolve
        replace_conflicting=replace_conflicting
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1057, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1069, in obtain
        return installer(requirement)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 580, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 698, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 724, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 909, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1177, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1165, in run_setup
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
 
        =============================DEBUG ASSISTANCE=============================
        If you are seeing a compilation error please try the following steps to
        successfully install cryptography:
        1) Upgrade to the latest pip and try again. This will fix errors for most
           users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
        2) Read https://cryptography.io/en/latest/installation.html for specific
           instructions for your platform.
        3) Check our frequently asked questions for more information:
           https://cryptography.io/en/latest/faq.html
        =============================DEBUG ASSISTANCE=============================
 
 
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-bwmzjp_w/cryptography/
[19:41:10] FATAL: Failed installing esphome pinned version.
[cont-init.d] 30-esphome.sh: exited 1.
[cont-finish.d] executing container finish scripts...
[cont-finish.d] 99-message.sh: executing... 
-----------------------------------------------------------
                Oops! Something went wrong.
 
 We are so sorry, but something went terribly wrong when
 starting or running this add-on.
 
 Be sure to check the log above, line by line, for hints.
-----------------------------------------------------------
[cont-finish.d] 99-message.sh: exited 0.
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.

```

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

SeByDocKy, 2020-12-20:
I tested with the 765e641 commit and it's working fine.... so the problem is introduced with the last commit.

Enzokot, 2020-12-22:
@SeByDocKy are you using the add-on for HA?

stone0904, 2020-12-22:
> @SeByDocKy are you using the add-on for HA?

Hi,

I use the HA Add-on and have exact the same issue. Yesterday I updated HAOS from 5.8 to 5.9 without successes.

BR Sven

Glaucer, 2020-12-22:
I upgraded mine today to HassOs 5.9 and started this problem

Enzokot, 2020-12-22:
The same issue on raspbian.

szymanskip, 2020-12-22:
The workaround for now is to specify the exact version of esphome in the add-on configuration like that:
```
esphome_version: 765e641d08dc30db15501eff7a66f33ec06314f7
```

glmnet, 2020-12-27:
So problem is the esptool 3? ESPHome uses it only for esp32. I've tried on my desk and it worked ok.

gitolicious, 2020-12-31:
Same happens on my Odroid N2.

    System: Debian GNU/Linux 10 (buster)  (aarch64 / odroid-n2)
    Home Assistant Core: 2020.12.2
    Home Assistant Supervisor: 2020.12.7

Rollingback as proposed by @szymanskip works fine.

Enzokot, 2021-01-01:
> So problem is the esptool 3? ESPHome uses it only for esp32. I've tried on my desk and it worked ok.

Problem with installing add-on HA with option 
esphome_version: dev

Lumpusz, 2021-01-02:
> > So problem is the esptool 3? ESPHome uses it only for esp32. I've tried on my desk and it worked ok.
> 
> Problem with installing add-on HA with option
> esphome_version: dev

I have the same issue, it is cause by the esptool bump from 2.8 to 3.0.

jp181278, 2021-01-05:
hy I have solved the issue indicating the esp version in the configuration of the tool 
esphome_version: ab48e4a466d907a5d466fe0859b57159524732d6
doesn't work form me and I'm not using the dev version but the release one but after indicating the dev version esphome_version: 765e641d08dc30db15501eff7a66f33ec06314f7 everything seems to go ok

SeByDocKy, 2021-01-11:
When can we expect a final fix ? coz actually we locked into the last dev working version without any hope to enjoy last progress of the dev version

glmnet, 2021-01-11:
Could you test branch https://github.com/esphome/esphome/tree/revert-esptool

```
esphome_version: revert-esptool
```

Enzokot, 2021-01-11:
> Could you test branch https://github.com/esphome/esphome/tree/revert-esptool
> 
> ```
> esphome_version: revert-esptool
> ```

Yes, this config is work good

closed 2021-01-11 by glmnet

gitolicious, 2021-01-11:
Everything works as before, thanks!