esphome/issues #1705: Light automation '- light.turn_on:' wont allow two (or more) 'transformers'

3ative, 2020-12-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

Latest - v1.15.3

**Affected component:**

### **light.turn_on Action**
https://esphome.io/components/light/index.html?highlight=light%20turn_on#light-turn-on-action

**Description of problem:**
When using _'light.turn_on Action'_, an error occurs when 'effect' and 'transition_length' are used together - either on their own work as expected. The '**Configuration options:**' (_docs page_) suggests any combination is allowed. 

**Problem-relevant YAML-configuration entries:**
[image]

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
When sending an API command from Node-Red, I.E. 'Call service Node' >Domain > Light > 'turn_on'** with Data 'effect).
For example:
```yaml
    data:
    {"effect":"flicker"}
```
The **default_transition_length:** of the light is respected - So it would be great if this was also the case for this "light.turn_on Action"

3ative, 2021-04-19:
Anyone ??
Bueller... Bueller...

glmnet, 2021-04-21:
looks like you want a pulsating effect?

3ative, 2021-04-21:
> looks like you want a pulsating effect?

No, I would like to add 'Flicker' effect and another parameter - as mentioned above, ESPHome "light.turn_on:' only allows one.

glmnet, 2021-04-26:
Yes seems like two different effects 

What you want to accomplish exactly?

Transition length is for fading from one brightness or color to another, no fading from or to effects is supported

3ative, 2021-04-27:
@glmnet 
Within ESPHome code there is a 'Default Transition Length' and if I command this light to come on, from outside - E.G. HA or Node-Red, both the 'Flicker' effect and default transition is honoured.
However, using the ESPHome automation, this _default transition_ is ignored - so, I tried to include it in the ESPHome automation... as you can see, it doesn't like it.

closed 2021-09-08 by stale[bot]