esphome/issues #1708: st7789v seems to ignore spi_id

fwestenberg, 2020-12-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 2020.12.0

**ESP (ESP32/ESP8266, Board/Sonoff):**

featheresp32

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

https://esphome.io/components/display/st7789v.html

**Description of problem:**
The st7789v display is (factory) attached to pin 19 (mosi), pin 18 (clk) and pin 5 (cs). When for example the max31855 board is also connected, this board also need a spi configuration. This is possible in ESPHome by adding spi_id to the configured spi interfaces like:

```
spi:
  - clk_pin: GPIO18
    mosi_pin: GPIO19
     id: spi_display
    
  - clk_pin: GPIO25
    miso_pin: GPIO27
    id: spi_pid
```

And the display is use this spi_id also:

```
display:
  - platform: st7789v
    spi_id: spi_display
   ...
```

The display only turns on, when the second SPI configuration is removed. Seems like the st7789v ignores the spi_id parameter.

**Problem-relevant YAML-configuration entries:**
```yaml
spi:
  - clk_pin: GPIO18
    mosi_pin: GPIO19
    id: spi_display
    
  - clk_pin: GPIO25
    miso_pin: GPIO27
    id: spi_pid

display:
  - platform: st7789v
    backlight_pin: GPIO4
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO23
    rotation: 90
    spi_id: spi_display

sensor:
  - platform: max31855
    name: "PID"
    cs_pin: 33
    spi_id: spi_pid
    id: pid_sensor
    update_interval: 60s
```

**Logs (if applicable):**

```
Nothing special in the log, no errors or warnings.

```

**Additional information and things you've tried:**
- Remove the spi_id from the display- and spi config - no effect
- Remove the sensor config and spi of the sensor - display turns
- Change the sequence of the spi configuration - no effect

DocDoo, 2021-04-04:
### Operating environment/Installation (Hass.io/Docker/pip/etc.): 

`Docker v1.16.2`

### ESP (ESP32/ESP8266, Board/Sonoff):

`featheresp32`

### ESPHome version (latest production, beta, dev branch)

`Latest production: core-2021.3.4`

### Affected component:

https://esphome.io/components/display/st7789v.html

### Description of problem:
Like @fwestenberg mentions it appears like the st7789v display on the TTGO-Display is not respecting the **spi_id:** value provided.

I have a MAX31865 module (named "spi_pt100") connected to the TTGO-Display (which is named "spi_display"). When I configure them using the below SPI definition, then the MAX31865 module does not return correct temperature readings (it returns `Sending state -242.02000 °C with 2 decimals of accuracy` instead), but the ST7789V module (the display) works as expected.

```
spi:
    # Definition of IN/OUT (IO) pins til brug for Serial Programmable Interface (SPI) bus for MAX31865 module
  - clk_pin: GPIO25
    miso_pin: GPIO27
    mosi_pin: GPIO26
    id: spi_pt100

    # Definition of IN/OUT (IO) pins til brug for Serial Programmable Interface (SPI) bus for ST7789V module
  - clk_pin: GPIO18
    mosi_pin: GPIO19
    id: spi_display
```

### Logs (if applicable):

Nothing special in the log, no errors or warnings. Only the incorrect temperature reading of `Sending state -242.02000 °C with 2 decimals of accuracy` from the MAX31865 module.

### Additional information and things you've tried:

I have found a solution/workaround by changing the order by which I define the two busses to the following (the ST7789V module before the MAX31865 module) as seen below.

```
spi:
    # Definition of IN/OUT (IO) pins til brug for Serial Programmable Interface (SPI) bus for ST7789V module
  - clk_pin: GPIO18
    mosi_pin: GPIO19
    id: spi_display

    # Definition of IN/OUT (IO) pins til brug for Serial Programmable Interface (SPI) bus for MAX31865 module
  - clk_pin: GPIO25
    miso_pin: GPIO27
    mosi_pin: GPIO26
    id: spi_pt100
```

Then both devices work as expected 🤨

Regardless of which order I define the two SPI busses, I use the below configuration of the two modules. As you can see, I configure the MAX31865 module to use the **spi_pt100** bus and the ST7789V module to use the **spi_display** bus.

If I, for the ST7789V module, configure it to use the **spi_pt100** bus, I would expect it to fail, but it keeps working, which is why I concur with @fwestenberg that the problem is with the handling of the configuration of the ST7789V module.

That my workaround with defining the SPI bus for the ST7789V module before the SPI bus for the MAX31865 module, works I expect is because the ST7789V module simply connects to the first defined SPI bus and the MAX31865 module connects to the SPI bus which it is configured for. 🧐

```
sensor:
  - platform: max31865
    id: coffee_maker_temperature
    spi_id: spi_pt100
    name: "Coffee maker temperature"
    cs_pin: GPIO33
    mains_filter: 50 Hz
    rtd_wires: 3
    reference_resistance: 430 Ω
    rtd_nominal_resistance: 100 Ω
    update_interval: 5 s

display:
  - platform: st7789v
    spi_id: spi_display
    backlight_pin: GPIO4
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO23
    rotation: 90
    lambda: |-
      it.print(0, 0, id(helvetica_24), "Hello World!");
```

DocDoo, 2021-08-03:
I don't think this issue has been resolved and I don't have the competences to fix it myself I'm afraid. What can possibly be done here?

kbx81, 2021-08-03:
I can take a look and try to fix -- we also need to look at https://github.com/esphome/issues/issues/1944

NdK73, 2022-11-13:
I was searching before opening a new issue and noticed this workaround. Seems it fixes my use case too (ESP32-2432S028 board, one SPI bus for XPT2046 touch controller and another for ILI9341 display).
My working configuration is:
```
spi:
  # Order of SPI busses is important: if spi_touch is defined first, touch ctrl is not recognized
  - id: spi_disp
    clk_pin: 14
    mosi_pin: 13
    miso_pin: 12
  - id: spi_touch
    clk_pin: 25
    mosi_pin: 32
    miso_pin: 39

touchscreen:
  platform: xpt2046
  id: my_touch
  cs_pin: 33
  spi_id: spi_touch
  interrupt_pin: 36
  update_interval: 50ms
  report_interval: 1s
  threshold: 400
  calibration_x_min: 290
  calibration_x_max: 3840
  calibration_y_min: 220
  calibration_y_max: 3730
  swap_x_y: true
  on_touch:
  - lambda: |-
        ESP_LOGI("cal", "x=%d, y=%d, x_raw=%d, y_raw=%0d",
            id(my_touch)->x,
            id(my_touch)->y,
            id(my_touch)->x_raw,
            id(my_touch)->y_raw
            );

display:
  - platform: ili9341
    model: TFT 2.4
    spi_id: spi_disp
    rotation: 90
    cs_pin: 15
    dc_pin: 2
    led_pin: 21
```
Maybe a problem with index 0 in some common code?
Hope it helps.

kbx81, 2023-04-09:
If anyone is still following this issue, please test https://github.com/esphome/esphome/pull/4669 and let me know if it fixes the issue for you.

Add this to your config to give it a try:

```yaml
external_components:
  - source: github://pr#4669
    components: [ spi ]
```

closed 2023-04-12 by jesserockz