esphome/issues #1709: ESP32 - M5STACK Analog PIN Definition not working ?

Tokn59, 2020-12-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
M5STACK CORE

**ESPHome version (latest production, beta, dev branch)**
dev latest

**Affected component:**
https://esphome.io/components/sensor/adc.html

**Description of problem:**
PIN definition does not seem to work. I tried A0, GPIO32 and others but I keep getting the error : ERROR Error while reading config: Invalid YAML syntax:

mapping values are not allowed here
  in "/config/esphome/m5stackaqkoen.yaml", line 44, column 8:
        PIN: GPIO32
           ^

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: m5stack_air_quality
  platform: ESP32
  board: m5stack-core-esp32
  libraries:
    - "uFire SHT20"
    - "ArduinoJson"
  includes:
    - include/sht20.h

wifi:
  ssid: "Elsjeshome"
  password: "xxxxxxxxxxx"

  ap:
    ssid: "m5stackkoen"
    password: "xxxxxxxxxx"

api:

ota:

logger:

# Define I/O: UART, I²C and SPI
uart:
  rx_pin: 16
  baud_rate: 9600

i2c:
  sda: 21
  scl: 22
  scan: true
  frequency: 200kHz

spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

# Particulate matter sensor and temperature sensor
sensor:
- platform: adc
    PIN: GPIO32
    name: "Air Quality Indicator"
    update_interval: 60s
```

**Logs (if applicable):**
Error while reading config: Invalid YAML syntax:

mapping values are not allowed here
  in "/config/esphome/m5stackaqkoen.yaml", line 44, column 8:
        PIN: GPIO32
           ^
```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

koenvervloesem, 2020-12-20:
Your indentation is off. It should be:

```
sensor:
- platform: adc
  PIN: GPIO32
  name: "Air Quality Indicator"
  update_interval: 60s
```
Moreover, the syntax is case-sensitive. So it shouldn't be `PIN`, but `pin`:

```
sensor:
- platform: adc
  pin: GPIO32
  name: "Air Quality Indicator"
  update_interval: 60s
```

Tokn59, 2020-12-20:
Indeed Koen ! Blind as a mole looking straight at it and not seeing it ! Thanks a lot !!!!

Tokn59, 2020-12-26:
Koen, one more question if I may as you seem to be very knowledgable on M5Stack-ESPHome combo. Can I use both UART in the same ESPHome "sketch" ? I want to use the PMS7003 module (which uses GPIO16/17) and a MHZ-19 sensor on GPIO01/03 ...

I have tested MHZ-19 on GPIO01/03 and that doesn't seem to work ?

Thanks for your valuable support !

koenvervloesem, 2020-12-26:
Yes you can use both with their own UART. As I had the MH-Z19 too running at home on another ESP device, I just connected it to the M5Stack Air Quality Kit to try it, and this works. Just define two UARTs:

```yaml
uart:
  - id: uart_pms
    rx_pin: 16
    baud_rate: 9600
  - id: uart_co2
    rx_pin: 21
    tx_pin: 22
    baud_rate: 9600
```

Note that I used pins 21 and 22 for the second UART, which are normally used for I²C. I just did this because they are exposed to the outside on the M5Stack Core module so this was the easiest to do for me to test. I removed the I²C bus and the SHT20 sensor definition in the YAML configuration because the sensor isn't accurate anyway. If you use GPIO1 and GPIO3, you should probably disable UART logging so it won't interfere with your MH-Z19:

```yaml
logger:
  baud_rate: 0
```

Then in the definition of the sensors, you add the right UART id:

```yaml
sensor:
  - platform: pmsx003
    type: PMSX003
    uart_id: uart_pms
    ...
  - platform: mhz19
    id: co2_sensor
    uart_id: uart_co2
    ...
```

As your original issue has been solved, I think you can close the issue. And this was also another topic than the original one, so maybe open a separate issue next time, that makes it easier for other to find problems and their solutions.

Tokn59, 2021-01-03:
Working fine but it seems that readiing the Pmsx003 and the Mhz-19 sensor poses issues when trying to ue GPIO ad well. The pmsx003 integration does not seem to allof for a "update_interval:" setting causing timing issues.

I added the following to te code :

 - platform: gpio
    pin:
      number: 26
      mode: INPUT_PULLUP
      inverted: true
    id: proximity_ensor    
    name: "m5stack_co2_prox"

This sensor doesn't get it's status reported back to HA when going Low or High. Maybe due to the fact that the ESP32 is constantly busy with the Pmsx7003 ?

koenvervloesem, 2021-01-03:
Just to be sure, because I can't see it in your code: you have the following indentation, right?

```yaml
  - platform: gpio
    pin:
      number: 26
      mode: INPUT_PULLUP
      inverted: true
    id: proximity_ensor
    name: "m5stack_co2_prox"
```

If you want to get to the bottom of this, maybe temporarily add more verbosity to the [logger](https://esphome.io/components/logger.html) (although this will proably influence the timings too...):

```yaml
  logger:
    level: VERY_VERBOSE
```

Tokn59, 2021-01-03:
Yes, te identation is correct and I have the "switch" in HA .... Only I see no change of it's status. Not sure if I can set the logger to VERY_VERBOSe Ias I disable it :

# Enable logging
logger:
# Disable UART Logging
  baud_rate: 0
  hardware_uart: UART1

koenvervloesem, 2021-01-03:
You can remove the `baud_rate: 0` to re-enable it, hopefully it works on UART1.

Tokn59, 2021-01-03:
Hmmmm ... If I do that I can strangely enough no longer connect and no values are read anymore from the sensors :

INFO Reading configuration /config/esphome/m5stack_co2.yaml...
INFO Starting log output from m5stack_co2.local using esphome API
INFO Connecting to m5stack_co2.local:6053 (192.168.1.225)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.225: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to m5stack_co2.local:6053 (192.168.1.225)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.225: [Errno 111] Connection refused). Re-Trying in 1 seconds

Changing again to :

logger:
#  level: VERY_VERBOSE
# Disable UART Logging
  baud_rate: 0
  hardware_uart: UART1

results in everything working again .... Weird ....

closed 2021-05-18 by stale[bot]