esphome/issues #1711: Binary sensor sometimes not reporting the correct state

estevez-dev, 2020-12-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):** Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)** 1.15.3

**Affected component:** https://esphome.io/components/binary_sensor/gpio.html

**Description of problem:**
I have Wemos D1 Mini with relay and binary sensor. Relay is emulating button press and release on the sound amplifier, and the sensor is detecting 5V current in the factory USB port. Here is the build:
[image]

Most of the time all works fine, switch state is represent the state of the sensor. But sometimes the sensor doesn't report ON state when the amplifier is on (USB port should be powered when the amplifier is on). So the switch in HA is always OFF.
I'm calling `switch.turn_on` with `entity_id: switch.amplifier` in Home Assistant. It can take several times turning the amplifier on and off till the state starts upgrading.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: amplifier
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "********"
  password: "********"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Cambridge AXA25"
    password: "********"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "********"

ota:
  password: "********"

binary_sensor:
  - platform: gpio
    id: amplifier_power
    name: "Amplifier Power"
    pin:
      number: D2
      mode: INPUT_PULLDOWN_16
    on_state:
      then:
      - switch.template.publish:
          id: amplifier
          state: !lambda 'return x;'

switch:
  - platform: gpio
    id: relay
    pin: D1
    restore_mode: ALWAYS_OFF
  - platform: template
    name: "Amplifier"
    id: amplifier
    turn_on_action:
      - if:
          condition:
            binary_sensor.is_off: amplifier_power
          then:
          - switch.turn_on: relay
          - delay: 300ms
          - switch.turn_off: relay
    turn_off_action:
      - if:
          condition:
            binary_sensor.is_on: amplifier_power
          then:
          - switch.turn_on: relay
          - delay: 300ms
          - switch.turn_off: relay
```

**Logs (if applicable):**

```
[16:31:19][D][switch:021]: 'Amplifier' Turning ON.
[16:31:19][D][switch:021]: 'relay' Turning ON.
[16:31:19][D][switch:045]: 'relay': Sending state ON
[16:31:20][D][switch:025]: 'relay' Turning OFF.
[16:31:20][D][switch:045]: 'relay': Sending state OFF
``` 
Here are the correct functioning logs:
```
[16:42:37][D][switch:021]: 'Amplifier' Turning ON.
[16:42:37][D][switch:021]: 'relay' Turning ON.
[16:42:37][D][switch:045]: 'relay': Sending state ON
[16:42:37][D][switch:025]: 'relay' Turning OFF.
[16:42:37][D][switch:045]: 'relay': Sending state OFF
[16:42:37][D][binary_sensor:036]: 'Amplifier Power': Sending state ON
[16:42:37][D][switch:045]: 'Amplifier': Sending state ON
```

**Additional information and things you've tried:**
I have another similar device with ESPHome and Wemos D1 Mini that detects the state of the appliance by detecting 5V curret from the factory USB port and it works just fine. The only difference is there is no relay there. It uses IR led instead to transmit commands.

glmnet, 2021-01-05:
Most likely wiring issue. Can you attach pics?

glmnet, 2021-01-05:
Also this is wrong 

```

pin:
      number: D2
      mode: INPUT_PULLDOWN_16

```

estevez-dev, 2021-01-05:
> Also this is wrong
> 
> ```
> 
> pin:
>       number: D2
>       mode: INPUT_PULLDOWN_16
> ```

[image]

estevez-dev, 2021-01-05:
The issue is in USB itself. It is indeed not always powered, so all works as intended =)

closed 2021-01-05 by estevez-dev

glmnet, 2021-01-05:
> > Also this is wrong
> 
> > 
> 
> > ```
> 
> > 
> 
> > pin:
> 
> >       number: D2
> 
> >       mode: INPUT_PULLDOWN_16
> 
> > ```
> 
> 
> 
> [image]
> 
> 

Just FYI that's ok. 

Read docs for details: https://esphome.io/guides/configuration-types.html#pin-schema