esphome/issues #1712: ESP32 first WiFi connect alwasy fail after reset/power-on

Erriez, 2020-12-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

Version: 1.15.3

**Affected component:**

WiFi

**Description of problem:**
First WiFi connect alwas failed with error message after reset:
`Connecting to WiFi network failed. Are the credentials wrong?`
Second WiFi connect is successful.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP32
  board: lolin_d32

wifi:
  ssid: "abc"
  password: "password1234567890abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRS"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "avK3DYZfk1kz"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "thepassword"

ota:
  password: "thepassword"
```

**Logs (if applicable):**

```
[16:36:37][I][logger:166]: Log initialized
[16:36:37][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[16:36:37][I][app:029]: Running through setup()...
[16:36:37][C][wifi:033]: Setting up WiFi...
[16:36:37][D][wifi:324]: Starting scan...
[16:36:39][D][wifi:339]: Found networks:
[16:36:39][I][wifi:385]: - 'abc' (C8:0E:14:78:3A:56) ▂▄▆█
[16:36:39][D][wifi:386]:     Channel: 8
[16:36:39][D][wifi:387]:     RSSI: -66 dB
[16:36:39][I][wifi:194]: WiFi Connecting to 'abc'...
[16:36:40][W][wifi_esp32:386]: Event: Disconnected ssid='abc' bssid=C8:0E:14:78:3A:56 reason='Authentication Failed'
[16:36:40][W][wifi:504]: Connecting to WiFi network failed. Are the credentials wrong?
[16:36:40][I][wifi:194]: WiFi Connecting to 'abc'...
[16:36:40][W][wifi:504]: Connecting to WiFi network failed. Are the credentials wrong?
[16:36:45][D][wifi:324]: Starting scan...
[16:36:47][D][wifi:339]: Found networks:
[16:36:47][I][wifi:385]: - 'abc' (C8:0E:14:78:3A:56) ▂▄▆█
[16:36:47][D][wifi:386]:     Channel: 8
[16:36:47][D][wifi:387]:     RSSI: -66 dB
[16:36:48][I][wifi:194]: WiFi Connecting to 'abc'...
[16:36:48][W][wifi:452]: Incomplete connection.
[16:36:48][W][wifi_esp32:386]: Event: Disconnected ssid='abc' bssid=C8:0E:14:78:3A:56 reason='Association Leave'
[16:36:48][I][wifi:194]: WiFi Connecting to 'abc'...
[16:36:48][I][wifi:457]: WiFi Connected!
[16:36:48][C][wifi:303]:   SSID: 'abc'
[16:36:48][C][wifi:304]:   IP Address: 10.0.0.231
[16:36:48][C][wifi:306]:   BSSID: C8:0E:14:78:3A:56
[16:36:48][C][wifi:307]:   Hostname: 'test'
[16:36:48][C][wifi:311]:   Signal strength: -68 dB ▂▄▆█
[16:36:48][C][wifi:315]:   Channel: 8
[16:36:48][C][wifi:316]:   Subnet: 255.255.255.0
[16:36:48][C][wifi:317]:   Gateway: 10.0.0.138
[16:36:48][C][wifi:318]:   DNS1: 10.0.0.138
[16:36:48][C][wifi:319]:   DNS2: 0.0.0.0
[16:36:48][D][wifi:466]: Disabling AP...
[16:36:48][C][ota:029]: Over-The-Air Updates:
[16:36:48][C][ota:030]:   Address: test.local:3232
[16:36:48][C][ota:032]:   Using Password.
[16:36:48][C][api:022]: Setting up Home Assistant API server...
[16:36:48][I][app:059]: setup() finished successfully!
[16:36:48][I][app:105]: ESPHome version 1.15.3 compiled on Dec 21 2020, 16:32:52
[16:36:48][C][wifi:443]: WiFi:
[16:36:48][C][wifi:303]:   SSID: 'abc'
[16:36:48][C][wifi:304]:   IP Address: 10.0.0.231
[16:36:48][C][wifi:306]:   BSSID: C8:0E:14:78:3A:56
[16:36:48][C][wifi:307]:   Hostname: 'test'
[16:36:48][C][wifi:311]:   Signal strength: -66 dB ▂▄▆█
[16:36:48][C][wifi:315]:   Channel: 8
[16:36:48][C][wifi:316]:   Subnet: 255.255.255.0
[16:36:48][C][wifi:317]:   Gateway: 10.0.0.138
[16:36:48][C][wifi:318]:   DNS1: 10.0.0.138
[16:36:48][C][wifi:319]:   DNS2: 0.0.0.0
[16:36:48][C][logger:185]: Logger:
[16:36:48][C][logger:186]:   Level: DEBUG
[16:36:48][C][logger:187]:   Log Baud Rate: 115200
[16:36:48][C][logger:188]:   Hardware UART: UART0
[16:36:48][C][captive_portal:169]: Captive Portal:
[16:36:48][C][ota:029]: Over-The-Air Updates:
[16:36:48][C][ota:030]:   Address: test.local:3232
[16:36:48][C][ota:032]:   Using Password.
[16:36:48][C][api:095]: API Server:
[16:36:48][C][api:096]:   Address: test.local:6053
```

**Additional information and things you've tried:**
I've placed the ESP32 close to the WiFi (FritzBox 5490) and same behavior as FritzWlan Repeater 1750E.

Lymkin, 2020-12-22:
I've fixed this issue before on my personal ESP programs by starting a 5 min timer at boot and if the ESP has no connection within 5 mins it auto reboots.  The ESP's boot way faster than an access point, so I have to believe this is a common issue with power outages.
`  WebServer.begin();

  if (WiFi.status() == WL_CONNECTED) {
    digitalWrite (StatusLed, HIGH);
    StatusLedIsOn = true;
  } 

  String CurrentAPIP = WiFi.softAPIP().toString();

  if (CurrentAPIP == "0.0.0.0") {
    DebugPrintln("WiFi Station Mode");
    GlobalPause = false;
    InAPMode = false;
  }
  else {
    DebugPrintln("WiFi AP Mode....Auto restart in 5 minutes.");
    GlobalPause = true;
    InAPMode = true;
    APModeTimer = millis();
  }

void loop() {
  unsigned long curTime = millis();

  if (InAPMode == true) {  //Reset ESP after 5 mins to restore connectivity after possible power outage
    if (curTime - APModeTimer >= 300000) {
      DebugPrintln("Resetting ESP to restore connectivity....");
      ESP.restart();
    }
  }
`

Erriez, 2020-12-22:
Thanks, I'll give it a try.

Lymkin, 2020-12-22:
@Erriez - This is not ESPHome code, it is the arduino code they would need to implement to restart the esp.

closed 2021-05-06 by stale[bot]