esphome/issues #1714: Simple Cathode LED Error

cwberardi, 2020-12-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

Arguably not a bug, but no matter what I try in my config file I cannot get esphome to light a basic cathode LED.  I have checked the voltage of the pin output which show I am getting a signal.  Does anyone have a working schematic and code where they are turning on/off a single cathode LED?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bedroom_sensor
  platform: ESP8266
  board: 
  
packages:
  wifi: !include wifi.yaml

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: True

sensor:
  - platform: adc
    pin: A0
    name: "Bedroom Ambient Light"
    update_interval: 120s
  - platform: htu21d
    temperature:
      name: "Bedroom Temperature"
      id: htu21d_temp
      filters:
        - lambda: return x * (9.0/5.0) + 32.0;
        - offset: -2.0
      unit_of_measurement: "°F"
    humidity:
      name: "Bedroom Humidity"
      id: htu21d_humidity
    update_interval: 120s
  - platform: sgp30
    eco2:
      name: "Bedroom eCO2"
      accuracy_decimals: 1
    tvoc:
      name: "Bedroom TVOC"
      accuracy_decimals: 1
    update_interval: 120s
    compensation:
      temperature_source: htu21d_temp
      humidity_source: htu21d_humidity
  - platform: wifi_signal
    name: "Bedroom Sensor WiFi Signal"
    update_interval: 120s

binary_sensor:
  - platform: gpio
    pin: GPIO12
    name: "PIR Sensor"
    device_class: motion
    on_press:
      then:
        - light.turn_on: single_led
    on_release:
      then:
        - light.turn_off: single_led
  - platform: status
    name: "Bedroom Sensor Status"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: Bedroom Sensor IP Address
    ssid:
      name: Bedroom Sensor SSID
    mac_address:
      name: Bedroom Sensor Mac Wifi Address

light:
  - platform: binary
    name: "LED"
    output: gpio_led
    id: single_led

output:
  - platform: binary
    pin: GPIO16
    id: gpio_led

switch:
  - platform: restart
    name: "Bedroom Sensor Restart"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret esp_ota

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I've tried multiple pins on my D1 mini as well as multiple LEDs and resistors, nothing has yielded light.

cwberardi, 2020-12-22:
Mistake in yaml config above.  It actually reads
output:
  - platform: gpio
    pin: GPIO16
    id: gpio_led

glmnet, 2020-12-22:
so you want it to turn on with the pir sensor, are you sure the PIR is giving signal properly?
I would setup the pir input as input pullup and then connect the pin to ground via a jumper just to make sure all the led stuff works

cwberardi, 2020-12-22:
> so you want it to turn on with the pir sensor, are you sure the PIR is giving signal properly?
> I would setup the pir input as input pullup and then connect the pin to ground via a jumper just to make sure all the led stuff works

Correct.  The ESPHome logs show the PIR working and that the LED is getting turned on/off, but no light from the LED.  Given this would you still recommend changing the PIR to input pullup?
```
[08:56:30][D][binary_sensor:036]: 'PIR Sensor': Sending state ON
[08:56:30][D][light:265]: 'LED' Setting:
[08:56:30][D][light:274]:   State: ON
[08:56:33][D][binary_sensor:036]: 'PIR Sensor': Sending state OFF
[08:56:33][D][light:265]: 'LED' Setting:
[08:56:33][D][light:274]:   State: OFF

glmnet, 2020-12-22:
hmm no, given those logs it's pretty clear the pin is being commanded, so no need for that, next step would be attach a multimeter to the GPIO output pin and measure voltages on the pin

cwberardi, 2020-12-22:
> hmm no, given those logs it's pretty clear the pin is being commanded, so no need for that, next step would be attach a multimeter to the GPIO output pin and measure voltages on the pin

Good idea. I just did that and it shows 3.3V when PIR is triggered.  I guess from this I can infer it is nothing wrong with ESPHome since I am seeing the correct voltage on the pin.  Perhaps there is not enough current to drive the LED?  Regardless, it must be something wrong with my wiring, appreciate your help.

closed 2020-12-22 by cwberardi