esphome/issues #1716: ESPHome-Dev under Home Assistant now fails to build

nickolas1969, 2020-12-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

RPi 4 8GB running Hass.io image, and recently updated to current (as of yesterday)

Supervisor version 2020.12.7
Home Assistant OS 5.9

**ESP (ESP32/ESP8266, Board/Sonoff):**

No issues with my ESP8266 modules, this is just with the add-on in HA.

**ESPHome version (latest production, beta, dev branch)**

ESPHome-Dev current version as of today. 15ef4d2f_esphome-dev

**Affected component:**

**Description of problem:**

The ESPHome-dev version is no longer building itself under Hass.io on RPi 8GB 64-bit image. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 154, in save_modules
        yield saved
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 250, in run_setup
        _execfile(setup_script, ns)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 45, in _execfile
        exec(code, globals, locals)
      File "/tmp/easy_install-vwfq3fh5/cffi-1.14.4/setup.py", line 242, in <module>
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 163, in setup
        raise SystemExit("error: " + str(msg))
    SystemExit: error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1163, in run_setup
        run_setup(setup_script, args)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 253, in run_setup
        raise
      File "/usr/lib/python3.6/contextlib.py", line 99, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python3.6/contextlib.py", line 99, in __exit__
        self.gen.throw(type, value, traceback)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 166, in save_modules
        saved_exc.resume()
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 141, in resume
        six.reraise(type, exc, self._tb)
      File "/usr/lib/python3/dist-packages/setuptools/_vendor/six.py", line 685, in reraise
        raise value.with_traceback(tb)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 154, in save_modules
        yield saved
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 195, in setup_context
        yield
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 250, in run_setup
        _execfile(setup_script, ns)
      File "/usr/lib/python3/dist-packages/setuptools/sandbox.py", line 45, in _execfile
        exec(code, globals, locals)
      File "/tmp/easy_install-vwfq3fh5/cffi-1.14.4/setup.py", line 242, in <module>
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 163, in setup
        raise SystemExit("error: " + str(msg))
    SystemExit: error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-rpj2l89g/cryptography/setup.py", line 121, in <module>
        "src/_cffi_src/build_padding.py:ffi",
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 128, in setup
        _install_setup_requires(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 123, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 513, in fetch_build_eggs
        replace_conflicting=True,
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 774, in resolve
        replace_conflicting=replace_conflicting
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1057, in best_match
        return self.obtain(req, installer)
      File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1069, in obtain
        return installer(requirement)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 580, in fetch_build_egg
        return cmd.easy_install(req)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 698, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 724, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 909, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1177, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/usr/lib/python3/dist-packages/setuptools/command/easy_install.py", line 1165, in run_setup
        raise DistutilsError("Setup script exited with %s" % (v.args[0],))
    distutils.errors.DistutilsError: Setup script exited with error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
    
        =============================DEBUG ASSISTANCE=============================
        If you are seeing a compilation error please try the following steps to
        successfully install cryptography:
        1) Upgrade to the latest pip and try again. This will fix errors for most
           users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
        2) Read https://cryptography.io/en/latest/installation.html for specific
           instructions for your platform.
        3) Check our frequently asked questions for more information:
           https://cryptography.io/en/latest/faq.html
        =============================DEBUG ASSISTANCE=============================
    
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-rpj2l89g/cryptography/
[09:57:05] FATAL: Failed installing esphome pinned version.
[cont-init.d] 30-esphome.sh: exited 1.

```

**Additional information and things you've tried:**

Tried uninstalling the Add-On, deleting the ESPHome directory, then re-installing fresh. The logs above are the result.

nickolas1969, 2020-12-22:
Snagged this in the logs as well, previous log errors scrolled it passed me.
 
 Development Version! Manage and program ESP8266/ESP32 microcontrollers through YAML configuration files
-----------------------------------------------------------
 Add-on version: dev
 You are running the latest version of this add-on.
 System: Home Assistant OS 5.9  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2020.12.1
 Home Assistant Supervisor: 2020.12.7
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.

nickolas1969, 2020-12-22:
Just attempted the ESPHome stable 1.15.3 with my old ESPHome directory (3 devices) and it loads fine, the devices are accessible. (Tag Reader and 2 KMC Flashed wall plugs)

pms0811, 2020-12-23:
Same for me.
Yesterday I could be install esphome-dev
But today I can't install esphome-dev

File "/usr/lib/python3.6/distutils/core.py", line 163, in setup
        raise SystemExit("error: " + str(msg))
    SystemExit: error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

and same log as nickolas1969

Hardware : Rpi 3b+
Operating System : Home Assistant OS 5.9
hassOS Version : 2020.12.1
Supervisor Version : 2020.12.7

Esphome(stable) is fine.

BertrumUK, 2020-12-23:
I am seeing the same. Thought it was a problem at my end after I was playing with Docker settings yesterday. Running supervised Raspbian GNU/Linux 10 (buster) Rpi4 4GB.

Removed `esphome_version: dev` from the config to roll back to 1.15.3.

Enzokot, 2020-12-23:
temporary solution
https://github.com/esphome/issues/issues/1702#issuecomment-749720677

BarBaar44, 2021-01-04:
Same here running an Odroid C2 with hassos 5.10

QbaF, 2021-01-10:
This same on RPi4 with Home Assistant Operating System 5.10 and HA 2021.1.1

glmnet, 2021-01-11:
Duplicate of https://github.com/esphome/issues/issues/1716

QbaF, 2021-01-11:
> Duplicate of #1716

This is #1716 :)

QbaF, 2021-05-12:
Still persist.

facaf, 2021-06-30:
RPi 3b+ hassio os 6.1, esphome 1.19.4.
Same problem, freezes on compilation. I can only restore the system by manually restarting the RPi.
I tried reinstalling hassio, using yaml from devices that were already working and had no luck.