esphome/issues #1717: ESP32 wrong ADC measurement

farbefreak, 2020-12-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
PIP
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**
both with:
Firmware: 1.15.2 (Dec 22 2020, 20:34:32)
and
Firmware: 1.15.3 (Dec 22 2020, 20:47:35)

**Affected component:**

ADC and therefor NTC

**Description of problem:**

ADC mesures wrong values. 
I connected a NTC and calibrated it. 
However the value that is reported does not match the calibrated values, so i tried to investigate.
(shows 19,5C, should be 25C)

Shorting the NTC leads to a ADC value of 0V -> all good.
However disconnecting the NTC shows:

[20:54:44][D][switch:021]: 'ntc_vcc' Turning ON.
[20:54:44][D][adc:056]: 'ntc_voltage': Got voltage=3.90V
[20:54:44][D][sensor:092]: 'ntc_voltage': Sending state 3.90000 V with 2 decimals of accuracy
[20:54:44][D][resistance:037]: 'ntc_resistance' - Resistance -61719.0Ω
Indicating it measured 3,90V (the maximum possible with ESP32 and the 11db setting) 

So i edited the yaml to measure the voltage going in the ADC pin (with still disconnected NTC but connected 10k Pullup):
3,27V -> everything fine, a little lower than the Vcc supply voltage but i assume its lower because of my DMM impedance in conjunction with the 10k pullup that's used.

Seems like the ADC is mesuring 3,9V even if it should be measuring 3,3V.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: ntc
    sensor: resistance_resistancesensor
    calibration:
      #b_constant: 3977
      #reference_temperature: 25°C
      #reference_resistance: 9.97kOhm
      ## using my calibration values
      - 6.28kOhm -> 35.7°C
      - 28.6kOhm -> 0°C
      - 10kOhm -> 24.8°C
    name: Bad Temperatur

  - platform: resistance
    sensor: ntc_voltage
    configuration: DOWNSTREAM
    resistor: 9.97kOhm
    reference_voltage: 3.27V
    internal: true
    name: ntc_resistance

# Example source sensor:
  - platform: adc
    id: ntc_voltage
    pin: GPIO34
    attenuation: 11db
    internal: true
    update_interval: never

switch:
  #### NTC vcc geschalten
  - platform: gpio
    pin: 2
    id: ntc_vcc

interval:
  - interval: 60s
    then:
      - switch.turn_on: ntc_vcc
      - component.update: ntc_voltage
      ## dont turn off the pin to make it possible to messure ntc_vcc with DMM
      ##- switch.turn_off: ntc_vcc
```

**Logs (if applicable):**
```
[20:48:01][I][app:105]: ESPHome version 1.15.3 compiled on Dec 22 2020, 20:47:35
[20:48:01][C][wifi:443]: WiFi:
[20:48:01][C][wifi:303]:   SSID: *****
[20:48:01][C][wifi:304]:   IP Address: 192.168.0.31
[20:48:01][C][wifi:306]:   BSSID: AC:******
[20:48:01][C][wifi:307]:   Hostname: 'rgbww'
[20:48:01][C][wifi:311]:   Signal strength: -63 dB ▂▄▆█
[20:48:01][C][wifi:315]:   Channel: 6
[20:48:01][C][wifi:316]:   Subnet: 255.255.255.0
[20:48:01][C][wifi:317]:   Gateway: 192.168.0.1
[20:48:01][C][wifi:318]:   DNS1: 192.168.0.1
[20:48:01][C][wifi:319]:   DNS2: 0.0.0.0
[20:48:01][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Bewegung Bad'
[20:48:01][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[20:48:01][C][gpio.binary_sensor:016]:   Pin: GPIO35 (Mode: INPUT)
[20:48:01][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Bewegung Toilette'
[20:48:01][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[20:48:01][C][gpio.binary_sensor:016]:   Pin: GPIO32 (Mode: INPUT)
[20:48:01][C][switch.gpio:042]: GPIO Switch 'ntc_vcc'
[20:48:01][C][switch.gpio:043]:   Pin: GPIO2 (Mode: OUTPUT)
[20:48:01][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[20:48:01][C][ledc.output:032]: LEDC Output:
[20:48:01][C][ledc.output:033]:   Pin GPIO14 (Mode: OUTPUT)
[20:48:01][C][ledc.output:034]:   LEDC Channel: 0
[20:48:01][C][ledc.output:035]:   Frequency: 1220.0 Hz
[20:48:01][C][ledc.output:032]: LEDC Output:
[20:48:01][C][ledc.output:033]:   Pin GPIO12 (Mode: OUTPUT)
[20:48:01][C][ledc.output:034]:   LEDC Channel: 1
[20:48:01][C][ledc.output:035]:   Frequency: 1220.0 Hz
[20:48:01][C][ledc.output:032]: LEDC Output:
[20:48:01][C][ledc.output:033]:   Pin GPIO27 (Mode: OUTPUT)
[20:48:01][C][ledc.output:034]:   LEDC Channel: 2
[20:48:01][C][ledc.output:035]:   Frequency: 1220.0 Hz
[20:48:01][C][ledc.output:032]: LEDC Output:
[20:48:01][C][ledc.output:033]:   Pin GPIO13 (Mode: OUTPUT)
[20:48:01][C][ledc.output:034]:   LEDC Channel: 3
[20:48:01][C][ledc.output:035]:   Frequency: 1220.0 Hz
[20:48:01][C][ledc.output:032]: LEDC Output:
[20:48:01][C][ledc.output:033]:   Pin GPIO26 (Mode: OUTPUT)
[20:48:01][C][ledc.output:034]:   LEDC Channel: 4
[20:48:01][C][ledc.output:035]:   Frequency: 1220.0 Hz
[20:48:02][C][ledc.output:032]: LEDC Output:
[20:48:02][C][ledc.output:033]:   Pin GPIO25 (Mode: OUTPUT)
[20:48:02][C][ledc.output:034]:   LEDC Channel: 5
[20:48:02][C][ledc.output:035]:   Frequency: 1220.0 Hz
[20:48:02][C][ledc.output:032]: LEDC Output:
[20:48:02][C][ledc.output:033]:   Pin GPIO33 (Mode: OUTPUT)
[20:48:02][C][ledc.output:034]:   LEDC Channel: 6
[20:48:02][C][ledc.output:035]:   Frequency: 1220.0 Hz
[20:48:02][C][logger:185]: Logger:
[20:48:02][C][logger:186]:   Level: DEBUG
[20:48:02][C][logger:187]:   Log Baud Rate: 115200
[20:48:02][C][logger:188]:   Hardware UART: UART0
[20:48:02][C][light:178]: Light 'Badezimmer'
[20:48:02][C][light:180]:   Default Transition Length: 0.5s
[20:48:02][C][light:181]:   Gamma Correct: 1.80
[20:48:02][C][light:184]:   Min Mireds: 153.8
[20:48:02][C][light:185]:   Max Mireds: 370.4
[20:48:02][C][light:178]: Light 'Toilette'
[20:48:02][C][light:180]:   Default Transition Length: 1.0s
[20:48:02][C][light:181]:   Gamma Correct: 1.80
[20:48:02][C][light:184]:   Min Mireds: 153.8
[20:48:02][C][light:185]:   Max Mireds: 370.4
[20:48:02][C][ntc:014]: NTC Sensor 'Bad Temperatur'
[20:48:02][C][ntc:014]:   Unit of Measurement: '°C'
[20:48:02][C][ntc:014]:   Accuracy Decimals: 1
[20:48:02][C][ntc:014]:   Icon: 'mdi:thermometer'
[20:48:02][C][resistance:010]: Resistance Sensor 'ntc_resistance'
[20:48:02][C][resistance:010]:   Unit of Measurement: 'Ω'
[20:48:02][C][resistance:010]:   Accuracy Decimals: 1
[20:48:02][C][resistance:010]:   Icon: 'mdi:flash'
[20:48:02][C][resistance:011]:   Configuration: DOWNSTREAM
[20:48:02][C][resistance:012]:   Resistor: 9970.00Ω
[20:48:02][C][resistance:013]:   Reference Voltage: 3.3V
[20:48:02][C][adc:026]: ADC Sensor 'ntc_voltage'
[20:48:02][C][adc:026]:   Unit of Measurement: 'V'
[20:48:02][C][adc:026]:   Accuracy Decimals: 2
[20:48:02][C][adc:026]:   Icon: 'mdi:flash'
[20:48:02][C][adc:035]:   Pin: 34
[20:48:02][C][adc:047]:  Attenuation: 11db (max 3.9V)
[20:48:02][C][adc:051]:   Update Interval: 4294967.5s
[20:48:02][C][ota:029]: Over-The-Air Updates:
[20:48:02][C][ota:030]:   Address: rgbww.local:3232
[20:48:02][C][api:095]: API Server:
[20:48:02][C][api:096]:   Address: rgbww.local:6053
[20:48:04][D][api.connection:604]: Client 'Home Assistant 0.116.2 (192.168.0.25)' connected successfully!
[20:48:44][D][switch:021]: 'ntc_vcc' Turning ON.
[20:48:44][D][adc:056]: 'ntc_voltage': Got voltage=1.77V
[20:48:44][D][sensor:092]: 'ntc_voltage': Sending state 1.77333 V with 2 decimals of accuracy
[20:48:44][D][resistance:037]: 'ntc_resistance' - Resistance 11813.0Ω
[20:48:44][D][sensor:092]: 'ntc_resistance': Sending state 11813.00684 Ω with 1 decimals of accuracy
[20:48:44][D][ntc:026]: 'Bad Temperatur' - Temperature: 21.3°C
[20:48:44][D][sensor:092]: 'Bad Temperatur': Sending state 21.25627 °C with 1 decimals of accuracy
... (disconnected NTC)
[20:50:44][D][switch:021]: 'ntc_vcc' Turning ON.
[20:50:44][D][adc:056]: 'ntc_voltage': Got voltage=3.90V
[20:50:44][D][sensor:092]: 'ntc_voltage': Sending state 3.90000 V with 2 decimals of accuracy
[20:50:44][D][resistance:037]: 'ntc_resistance' - Resistance -61719.0Ω
[20:50:44][D][sensor:092]: 'ntc_resistance': Sending state -61719.03516 Ω with 1 decimals of accuracy
```

**Additional information and things you've tried:**

farbefreak, 2020-12-22:
Seems like issue: 1640 mentions it.

I do have a work around, however it still does not work right:

  - platform: adc
    id: ntc_voltage
    pin: GPIO34
    attenuation: 11db
    internal: true
    update_interval: never
    filters:
      - calibrate_linear:
         - 0.0 -> 0.0
         - 1.75 -> 1.65

1,75V is read by the ADC if 1,65V are actually present. 

So filtering it this way fixes it sort of. 
Sort of because this linear scale would mean it should measure 3,5V with 3,3V present. However as tested before, it measures >3,9V with 3,3V present. So the error is not linear.
only adding 0.0 -> 0,0 and 3,9V -> 3,3V also screws up the temperature value.  (because its not linear..)
Something seems wrong with the ESP32 ADC code.

rradar, 2020-12-25:
The analog inputs on the esp's (both 82xx and 32) are **well known to not be very accurate**. Specially results on the start and the end of range tent to drift a lot.

Could you try what measurements you get when using a [ads1115](https://esphome.io/components/sensor/ads1115.html) for example?

farbefreak, 2020-12-26:
I do have a ads1115 connected to a esp8266 but i didn't check the accuracy because i only have moisture sensors connected that are calibrated as a complete system (add -> sensor -> earth moisture).
But i will check if it measures right, however it will take a while to do.

I didnt know that the ADC could be off by this much. Thats >100mV in the middle of the range and > 600mV! on the end of the range, unusable in my opinion.

Do you know if the offset is at least stable? Or does it move?

I will check the ADC of my board by connecting a power supply and drawing a curve, this will be interesting.

CarlosGS, 2021-11-04:
This was fixed in https://github.com/esphome/esphome/pull/2574 :smiley: 

Also, I'm adapting the documentation to remove the misleading 3.9V (https://github.com/esphome/esphome/pull/2663)