esphome/issues #1719: MQTT persistence? (or am I missing a point?)

jeanfabrice, 2020-12-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Eclipse Mosquitto 2020.12.1

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP WROOM 32

**ESPHome version (latest production, beta, dev branch)**
1.15.3

**Affected component:**

https://esphome.io/components/mqtt.html

**Description of problem:**
Messages posted to MQTT topic during deep_sleep are not processed at wake-up.

**Problem-relevant YAML-configuration entries:**
```yaml
mqtt:
  broker: mqtt.local
  username: esp32
  password: <redacted>
  topic_prefix: homeassistant/esp32
  discovery_prefix: discovery
 
deep_sleep:
  run_duration: 10s
  sleep_duration: 20s
  id: deep_sleep_1

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I would like my esp32 to process messages received in the subscribed topic while being asleep.
My goal is to build a cover switch, powered on battery, so taking advantages of deep_sleep capacity. Esp32 will be in deep_sleep 20s and run 10s. If I ask for cover closing during sleep time, I would like it to close the cover as soon as it wakes up.

Looking at this guide <http://www.steves-internet-guide.com/mqtt-clean-sessions-example/>, my understanding it that in order to process messages received while being disconnected, mqtt client has to use `clean_session=False` to ask for persistence on broker side.

ESPhome is using  [Paho mqtt client](https://github.com/eclipse/paho.mqtt.python) and does not seem to provide any specifics regarding `clean_session` : <https://github.com/esphome/esphome/blob/dev/esphome/mqtt.py#L46>.

When clean_session is not set, it defaults to `None` on Paho side, and `None` equals `True`: <https://github.com/eclipse/paho.mqtt.python/blob/v1.5.1/src/paho/mqtt/client.py#L554>

So my understanding is that esphome is not using persistent connection to the MQTT broker and there is currently no way to set it. Is that correct? Since I'm very new to esphome/hass.io, am I missing something?

M95D, 2020-12-24:
Maybe you didn't post the message with the "retain" option, or the broker doesn't allow retained messages.

jeanfabrice, 2020-12-24:
My understanding is that `retain` delivers the very last topic posted message to any (past or new) subscriber.
Using retain with a switch  'ON' payload will ask esp32 to open the cover every time it wakes up (until I clear the retained message)

This is not what I need.
I simply need subscriber to be delivered every messages he missed while being asleep, that is, having persistence enabled broker side.

jeanfabrice, 2020-12-24:
edit:
the Paho mqtt client and the esphome code linked in the issue are not the one used by MQTT Component in esphome.

Esphome MQTT Component uses the [AsyncMqttClient platform.io](https://platformio.org/lib/show/346/AsyncMqttClient) library 
Anyway , this library also sets `clean_session` to `true` (means: no persistence) by default  (<https://github.com/marvinroger/async-mqtt-client/blob/master/docs/2.-API-reference.md#asyncmqttclient-setcleansessionbool-cleansession>) and I could not find any reference to that setting into `esphome/components/mqtt/*`

closed 2021-05-08 by stale[bot]