esphome/issues #1721: OTA Error sending data: [Errno 104] Connection reset by peer

Erriez, 2020-12-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos D1 Mini / NodeMCU

**ESPHome version (latest production, beta, dev branch)**

Version: 1.15.3

**Affected component:**

https://esphome.io/components/ota.html

**Description of problem:**

OTA update results in failure: `Error sending data: [Errno 104] Connection reset by peer`

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_rf
  platform: ESP8266
  board: nodemcuv2
  #board: d1_mini_lite

wifi:
  ssid: "ssid"
  password: "passwordwith63characters"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ESP RF Fallback Hotspot"
    password: "FogrmAnfG0ln"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "password"

ota:
  password: "password"

remote_receiver:
  pin: 
    number: D2
    inverted: True
  dump: all
  idle: 20ms
  buffer_size: 2kb

remote_transmitter:
  pin: D1
  carrier_duty_percent: 50%

switch:
  - platform: template
    name: IR Humax power
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x0008
        command: 0x00FF
  - platform: template
    name: IR Humax volume up
[... 797 lines cut ...]
[11:45:18]load 0x4010f000, len 3584, room 16 
[11:45:18]tail 0
[11:45:18]chksum 0xb0
[11:45:18]csum 0xb0
[11:45:18]v2843a5ac
[11:45:18]~ld
[11:45:18][I][logger:166]: Log initialized
```

**Additional information and things you've tried:**

* The OTA upload error occurs randomly after changing `remote_receiver` or `remote_transmitter` in the `yaml` file.
* I assume there is sufficient RAM and flash available:

```
RAM:   [=====     ]  47.7% (used 39040 bytes from 81920 bytes)
Flash: [====      ]  40.8% (used 425848 bytes from 1044464 bytes)
```

* Update over USB is stable, but the goal is to upload in the field.
* ESP8266 is close to the WiFi access point.
* OTA update with `remote_receiver` and `remote_transmitter` removed works (when this already flashed)
* Works with ESP32.

Is it finally time to say goodbye to ESP8266?

soostdijck, 2020-12-30:
I had the same issue, turns out I had originally flashed with a `-fs 1MB` instead of the actual 4MB that it has. A manual re-flash using esptool and the correct flash size fixed things for me.

`esptool.py --port /dev/tty.usbserial-0001 write_flash -fm keep -fs 4MB 0x0 ~/Downloads/se
nsor_test_1.bin`

Erriez, 2020-12-30:
It looks like the application above won't fit in flash or RAM, but there is no error message.
I assume that `board: d1_mini_lite`  automatically configures the flash size, in this case 4MB, right?

Erriez, 2020-12-31:
Fully reproducible when adding / removing IR `turn_on_actions`.

The stack dump displays: `WARNING Decoded 0x40100a78: pvPortZalloc` so it looks like running out of stack.

agittins, 2021-01-01:
I've also been experiencing the symptom of OTA updates stalling with the same error when using just a single remote_receiver module. I'm testing an RF receiver which is a cheap SRX887, which is a superregenerative receiver, meaning that it's _always_ sending pulses through. I find if I tie it's output to ground I can then send OTA updates more reliably (100% reliably if it boots that way, less so if it's only grounded some time after bootup), so my guess is that the interrupt servicing (or polling?) is causing a race / timeout / resource exhaustion condition.

I'm using a Tuya ESP8266 module, the TYWE3S, RF input on GPIO12.

Perhaps somewhat related, I noticed that there's an "idle" setting which defines the end of a received code block and a filter param for high-passing, but for RF it might be helpful to have a setting for how long we should see a logic low before actively listening for / decoding a code - since we get noise all the time, but we get a quiet mark/sync block when valid carrier is present (depending on protocol, of course). I haven't looked at the code though, so I might be making some invalid assumptions there.

Erriez, 2021-01-01:
> I'm testing an RF receiver which is a cheap SRX887, which is a superregenerative receiver, meaning that it's always sending pulses through

Yes, these RF receivers generates interrupts continuously on the data pin. I've a SRX882 receiver connected with similar behavior. All interrupts must be disabled and allocated memory freed before executing an OTA update, but I'm not sure if this is implemented. I've also stack dumps and endless reboots when removing the receiver and add over 100 RF transmit functions.

I reached the limit of around 50 RF transmit functions with OTA enabled. Adding more (which is required for my application) results in stack overflow or unpredictable behavior because RAM is full. I can add around 25 RF transmits with OTA disabled. 

I don't understand why all generated `remote_transmitter` / `remote_receive` strings are located in RAM instead of flash. I think the efficiency of the generated code reaches the limit for my application.

agittins, 2021-01-01:
> I reached the limit of around 50 RF transmit functions with OTA enabled. Adding more (which is required for my application) results in stack overflow or unpredictable behavior because RAM is full. I can add around 25 RF transmits with OTA disabled.

Would your use-case support having the transmit functions handled inside HA instead of on-device? I'm using the built-in HA event component to send all received codes into HA, which I then handle using nodered. That way the ESP doesn't need to store any remote codes onboard, and I don't need to reflash every time I change my setups. Downside of course is that it then depends on HA being available to work, but I'm OK with that for the services I'm controlling. Here's what I'm using inside of the remote_receiver definition (shows both an IR and the RF ones, I have the IR one commented out currently while I test RF, but it does work):

```
remote_receiver:
  - id: ir_rx
    pin: 
      number: GPIO5
      inverted: True
      mode: INPUT_PULLUP
    dump: all
    on_nec:
      then:
        - homeassistant.event:
            event: esphome.remote_nec
            data_template:
              address: !lambda 'return x.address;'
              command: !lambda 'return x.command;'
  - id: rf_rx
    pin:
      number: GPIO12
    dump: rc_switch
    tolerance: 70%
    filter: 100us
    idle: 8ms
    buffer_size: 2kb
    on_rc_switch:
      then:
        - homeassistant.event:
            event: esphome.remote_rc_switch
            data_template:
              code: !lambda 'return x.code;'
              protocol: !lambda 'return x.protocol;'
```

I've got a similar thing setup for TX, but I haven't tested that, yet - let me know if you'd like the (untested) config though.

Erriez, 2021-01-01:
Thanks for sharing your idea. I understand what you mean and was not aware that this is possible. Currently I defined all RF codes in the ESPHome yaml, so all codes are located inside the ESP8266. I'll give it a try.

closed 2021-05-23 by stale[bot]

agittins, 2021-05-24:
This is still an issue, it seems that the interrupts from an RF receiver will hog enough time from the cpu that OTA updates stall.

At a guess, per @Erriez 's note above, esphome should be disabling interrupts from remote_receiver when it commences an OTA update, if it's not already.