esphome/issues #1722: Esphome with M5Stack and 2 X UART

Tokn59, 2020-12-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
M5STACK CORE

**ESPHome version (latest production, beta, dev branch)**
1.16.0-dev 

**Affected component:**

**Description of problem:**
Trying to use 2 UART's on M5Stack, not reading any values from UART defined as GPIO01/GPIO03. Tried setting logger level to NONE as suggested.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: m5stack_co2
  platform: ESP32
  board: m5stack-core-esp32
  libraries:
    - "uFire SHT20"
    - "ArduinoJson"
  includes:
    - include/sht20.h

wifi:
  ssid: "IOT"
  password: "xxxxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "M5Stack Co2 Fallback Hotspot"
    password: "5Cypmvmmdhbs"

captive_portal:

# Enable logging

logger:
  level: NONE

# Enable Home Assistant API
api:
  password: "xxxxxxxxxxx"

ota:
  password: "xxxxxxxxx"

web_server:
  port: 80

i2c:
  sda: 21
  scl: 22
  scan: true
  frequency: 200kHz

spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19
  
uart:
  - id: uart_pms 
    tx_pin: GPIO17
    rx_pin: GPIO16
    baud_rate: 9600  
  - id: uart_co2 
    tx_pin: GPIO03
    rx_pin: GPIO01
    baud_rate: 9600
sensor:
  - platform: mhz19
    co2:
      name: "MH-Z19 CO2 Value"
      id: co2value
    temperature:
      name: "MH-Z19 Temperature"
      id: co2temp
      filters:
       - offset: -3.0
    automatic_baseline_calibration: false  
    update_interval: 30s
    uart_id: uart_co2

#  - platform: pulse_width
#    pin: GPIO15
#    name: Pulse Width Sensor
#    update_interval: 10s
#    filters:
#    - sliding_window_moving_average:
#        window_size: 15
#        send_every: 15
#    - lambda: return (x * 256) + 1.0;
  - platform: pmsx003
    type: PMSX003
    uart_id: uart_pms
    pm_1_0:
      name: "PM1"
      id: sensor_pm1
      filters:
        - sliding_window_moving_average:
            window_size: 180
            send_every: 60
    pm_2_5:
      name: "PM2.5"
      id: sensor_pm25
      filters:
        - sliding_window_moving_average:
            window_size: 180
            send_every: 60
    pm_10_0:
      name: "PM10"
      id: sensor_pm10
      filters:
        - sliding_window_moving_average:
            window_size: 180
            send_every: 60
  - platform: custom
    lambda: |-
      auto sht20 = new SHT20();
      App.register_component(sht20);
      return {sht20->temperature_sensor, sht20->humidity_sensor, sht20->vpd_sensor, sht20->dew_point_sensor};
    sensors:
      - name: "Temperature"
        id: sensor_temperature
        unit_of_measurement: °C
        accuracy_decimals: 2
      - name: "Humidity"
        id: sensor_humidity
        unit_of_measurement: "%"
        accuracy_decimals: 2
      - name: "Vapour-pressure deficit"
        id: sensor_vpd
        unit_of_measurement: "kPa"
        accuracy_decimals: 2
      - name: "Dew point"
        id: sensor_dew_point
        unit_of_measurement: °C
        accuracy_decimals: 2

# Button to toggle the display backlight
binary_sensor:
  - platform: gpio
    id: M5_BtnB
    pin:
      number: 38
      inverted: true
    on_click:
      then:
        - switch.toggle: backlight

# GPIO pin of the display backlight
switch:
  - platform: gpio
    pin: 32
    name: "Backlight"
    id: backlight
    restore_mode: ALWAYS_ON

# Download Roboto font from https://fonts.google.com/specimen/Roboto
font:
  - file: "fonts/Roboto-Medium.ttf"
    id: font_roboto_medium22
    size: 22
    glyphs: '!"%()+,-_.:°0123456789ABCDEFGHIJKLMNOPQRSTUVWXYZ abcdefghijklmnopqrstuvwxyz/³µ'

display:
  - platform: ili9341
    id: m5stack_display
    model: M5Stack
    cs_pin: 14
    dc_pin: 27
    led_pin: 32
    reset_pin: 33
    rotation: 0
    lambda: |-
      Color RED(1,0,0);
      Color BLUE(0,0,1);
      Color WHITE(1,1,1);

      it.rectangle(0,  0, it.get_width(), it.get_height(), BLUE);
      it.rectangle(0, 22, it.get_width(), it.get_height(), BLUE);   // header bar
      it.print(it.get_width() / 2, 11, id(font_roboto_medium22), RED, TextAlign::CENTER, "Particulate matter");

      it.print(11,  33, id(font_roboto_medium22), WHITE, TextAlign::LEFT, "PM1");
      it.print(11,  55, id(font_roboto_medium22), WHITE, TextAlign::LEFT, "PM2.5");
      it.print(11,  77, id(font_roboto_medium22), WHITE, TextAlign::LEFT, "PM10");

      it.printf(it.get_width() - 11,  33, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "%.0f µg/m³", id(sensor_pm1).state);
      it.printf(it.get_width() - 11,  55, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "%.0f µg/m³", id(sensor_pm25).state);
      it.printf(it.get_width() - 11,  77, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "%.0f µg/m³", id(sensor_pm10).state);

      it.rectangle(0, 110, it.get_width(), 22, BLUE);   // header bar
      it.print(it.get_width() / 2, 121, id(font_roboto_medium22), RED, TextAlign::CENTER, "Environment");

      it.print(11, 143, id(font_roboto_medium22), WHITE, TextAlign::LEFT, "T");
      it.print(11, 165, id(font_roboto_medium22), WHITE, TextAlign::LEFT, "H");
      it.print(11, 187, id(font_roboto_medium22), WHITE, TextAlign::LEFT, "VPD");
      it.print(11, 209, id(font_roboto_medium22), WHITE, TextAlign::LEFT, "Tdp");

      it.printf(it.get_width() - 77, 143, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "%.0f", id(sensor_temperature).state);
      it.printf(it.get_width() - 77, 165, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "%.0f", id(sensor_humidity).state);
      it.printf(it.get_width() - 77, 187, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "%.0f", id(sensor_vpd).state);
      it.printf(it.get_width() - 77, 209, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "%.0f", id(sensor_dew_point).state);

      it.printf(it.get_width() - 11, 143, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "°C");
      it.printf(it.get_width() - 11, 165, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "%%");
      it.printf(it.get_width() - 11, 187, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "kPa");
      it.printf(it.get_width() - 11, 209, id(font_roboto_medium22), WHITE, TextAlign::RIGHT, "°C");    ```

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
PASTE DEBUG LOG HERE

[E][uart_esp32:147]: Reading from UART timed out at byte 0!
[W][mhz19:034]: Reading data from MHZ19 failed!
[D][pmsx003:111]: Got PM1.0 Concentration: 2 µg/m^3, PM2.5 Concentration 3 µg/m^3, PM10.0 Concentration: 3 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.64396 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 40.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 13.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 2 µg/m^3, PM2.5 Concentration 3 µg/m^3, PM10.0 Concentration: 3 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.65469 °C with 2 decimals of accuracy```

**Additional information and things you've tried:**

MHZ-19 works fine, tested.

koenvervloesem, 2020-12-26:
My suggestion was actually to [set the baud rate to 0](https://github.com/esphome/issues/issues/488#issuecomment-546479186), which seems to do something else than setting the log level to none.

Tokn59, 2020-12-26:
logger:
# Disable UART Logging
  baud_rate: 0

Done .. But still :

[D][pmsx003:111]: Got PM1.0 Concentration: 4 µg/m^3, PM2.5 Concentration 6 µg/m^3, PM10.0 Concentration: 8 µg/m^3
[E][uart_esp32:147]: Reading from UART timed out at byte 0!
[W][mhz19:034]: Reading data from MHZ19 failed!
[D][pmsx003:111]: Got PM1.0 Concentration: 4 µg/m^3, PM2.5 Concentration 6 µg/m^3, PM10.0 Concentration: 8 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.03263 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 4 µg/m^3, PM2.5 Concentration 6 µg/m^3, PM10.0 Concentration: 8 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.03263 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 4 µg/m^3, PM2.5 Concentration 6 µg/m^3, PM10.0 Concentration: 8 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.04336 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 3 µg/m^3, PM2.5 Concentration 6 µg/m^3, PM10.0 Concentration: 8 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.05408 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 4 µg/m^3, PM2.5 Concentration 7 µg/m^3, PM10.0 Concentration: 9 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.05408 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 5 µg/m^3, PM2.5 Concentration 7 µg/m^3, PM10.0 Concentration: 9 µg/m^3
[D][pmsx003:111]: Got PM1.0 Concentration: 5 µg/m^3, PM2.5 Concentration 7 µg/m^3, PM10.0 Concentration: 9 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.06481 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][sensor:092]: 'Temperature': Sending state 28.06481 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 5 µg/m^3, PM2.5 Concentration 7 µg/m^3, PM10.0 Concentration: 9 µg/m^3
[D][pmsx003:111]: Got PM1.0 Concentration: 5 µg/m^3, PM2.5 Concentration 7 µg/m^3, PM10.0 Concentration: 9 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.07553 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 4 µg/m^3, PM2.5 Concentration 7 µg/m^3, PM10.0 Concentration: 9 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.07553 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 4 µg/m^3, PM2.5 Concentration 7 µg/m^3, PM10.0 Concentration: 9 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.07553 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[D][sensor:092]: 'Dew point': Sending state 12.00000 °C with 2 decimals of accuracy
[D][pmsx003:111]: Got PM1.0 Concentration: 4 µg/m^3, PM2.5 Concentration 7 µg/m^3, PM10.0 Concentration: 9 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.08626 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 39.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000 kPa with 2 decimals of accuracy
[... cut ...]
[D][pmsx003:111]: Got PM1.0 Concentration: 1 µg/m^3, PM2.5 Concentration 1 µg/m^3, PM10.0 Concentration: 3 µg/m^3
[E][uart_esp32:147]: Reading from UART timed out at byte 0!
[W][mhz19:034]: Reading data from MHZ19 failed!
[D][pmsx003:111]: Got PM1.0 Concentration: 1 µg/m^3, PM2.5 Concentration 1 µg/m^3, PM10.0 Concentration: 3 µg/m^3
[D][sensor:092]: 'Temperature': Sending state 28.97644 °C with 2 decimals of accuracy
[D][sensor:092]: 'Humidity': Sending state 37.00000 % with 2 decimals of accuracy
[D][sensor:092]: 'Vapour-pressure deficit': Sending state 2.00000

koenvervloesem, 2020-12-26:
Does it work if you use another UART for the logger? For instance UART1 with GPIO9 and GPIO10:

```yaml
logger:
  hardware_uart: UART1
```

Tokn59, 2020-12-26:
YEZZZZZZZ !!!! You are a genius Koen !!!!!!!! (from Belgium as well ??)

So final code for the logger : 

# Disable UART Logging
logger:
  baud_rate: 0
  hardware_uart: UART1

THANK YOU !!!!!!!

closed 2020-12-27 by glmnet