esphome/issues #1723: ESP overload - crashes on boot while connecting to WiFi (with too many active HA sensors)

htvekov, 2020-12-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 - nodemcuv2

**ESPHome version (latest production, beta, dev branch)**

ESPhome 1.16.0.dev0

**Affected component:**

[WiFi component](https://esphome.io/components/wifi.html)
(most likely)

**Description of problem:**
Adding extra sensors/switches to my setup, makes ESPHome repeatedly crash on boot (during WiFI connection)

Reference issue: #1090

I've spend last two days trying to figure out what is wrong with my Sonos display set up with a PIR sensor and two filter delays.
Have read up on numerous issues regarding ESP crashes and entering eternal bootloop during WiFi setup.
Error reporting seems to be 'all over the place' and not sharing identical specifics - besides the stack trace errors on ESPHome boot.

My initial display lambda was quite extensive and time consuming (0,25 sec. according to verbose log). So first step was to bump ic2 speed to 100khz. That helped, but still i couldn't run code with 2 binary template sensors with two different filter off delays. Rewrote/tweaked my switch action that triggered the template sensors and finally got ESPHome to boot almost 100% stable.
But I can't add an on_boot action to turn on the display on each boot without ESP crashes again. Removing one of my template sensors and then it works ? As I need both my template sensors I had to tweak the code and instead add the turn on display command at my display page control code section.

Finally my current code is stable and boots on almost every attempt. But I have the feeling that I'm at 99,8% CPU usage first seconds after boot. Adding or moving anything in my code could easily cause ESP to crash on boot again.

Two simultaneous template sensors with delayed off schedules seems in most cases to be more than ESPHome can handle during boot ? I believe that ESPHome must have some core timing issues on boot. In my case if feels like to many things are initialized simultaneously on boot than ESPHome can handle.

Just had a brief count on sensors, switches etc. used in my current configuration:
2 x switches, 2 x HA sensors, 6 x HA text sensors, 6 x HA binary sensors (2 of those with delayed on/off filters), 1 x interval timer and an extensive i2c display lambda.

- Adding just one more sensor/switch to my current setup, makes ESPHome to crash on boot.
- Without fast_connect: true I have to remove numerous sensors and schedulers to get ESPHome to boot.

So my current setup really stretches ESPHome to max on an ESP8266 !

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: sonos_display_ssd1306
  friendly_name: Sonos Display ssd1306
  device_description: Sonos SSD1306/1309 display for Sonos groups

esphome:
  name: $devicename
  comment: ${device_description}
  platform: ESP8266
  board: nodemcuv2
  
wifi:
  ssid: "My SSID"
  password: "MYPASSWORD"
  fast_connect: true # Needed for fast WiFi connect to avoid stack trace error on boot
  manual_ip:
    static_ip: 192.168.X.XX # Enter your static IP address. Needed for fast WiFi connect to avoid stack trace error on boot 
    gateway: 192.168.X.X # Enter your gateway
    subnet: 255.255.255.0 # Enter your subnet

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fallback SSID"
    password: "FALLBACKPASSWORD"

captive_portal:

# Enable logging
[... 728 lines cut ...]
[14:30:14]<<<stack<<<
[14:30:14]
[14:30:14]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[14:30:14]
[14:30:14] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[14:30:14]
[14:30:14]load 0x4010f000, len 3584, room 16 
[14:30:14]tail 0
[14:30:14]chksum 0xb0
[14:30:14]csum 0xb0
[14:30:14]v2843a5ac
[14:30:14]~ld
[14:30:14][I][logger:166]: Log initialized
[14:30:14][C][ota:366]: There have been 3 suspected unsuccessful boot attempts.
[14:30:14][I][app:029]: Running through setup()...
[14:30:14][D][binary_sensor:034]: 'Sonos Display ssd1306 PIR': Sending initial state OFF
[14:30:14][D][binary_sensor:034]: 'Sonos Display ssd1306 Status': Sending initial state OFF
[14:30:14][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[14:30:14]
[14:30:14]User exception (panic/abort/assert)
[14:30:14]--------------- CUT HERE FOR EXCEPTION DECODER ---------------

```

**Additional information and things you've tried:**

htvekov, 2021-01-11:
Have just updated to v1.17.0.dev to see if  [fix safe_mode  - 1421](https://github.com/esphome/esphome/pull/1421) had any impact on my issue .

No difference I'm afraid with 1.17.0.dev

alexyao2015, 2021-01-15:
Can you attach main.cpp?

htvekov, 2021-01-16:
main.cpp attached
[main_copy.zip](https://github.com/esphome/issues/files/5824211/main_copy.zip)

alexyao2015, 2021-01-16:
From the main.cpp, it looks like it should boot into safe mode automatically after 10 unsuccessful boots. Is that what you are seeing?

htvekov, 2021-01-16:
hi' @alexyao2015 

Unfortunately not.
From serial or OTA update (with a hardware reset) doesn't really matter.

Eternal boot loop....
ESPHome tries to configure safe_boot after 10 loops, but still just strace track errors (see log below):

```
WARNING Decoded 0x40224cb0: loop_wrapper() at core_esp8266_main.cpp
[20:23:50]<<<stack<<<
[20:23:50]
[20:23:50]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[20:23:50]
[20:23:50] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[20:23:50]
[20:23:50]wdt reset
[20:23:50]load 0x4010f000, len 3584, room 16 
[20:23:50]tail 0
[20:23:50]chksum 0xb0
[20:23:50]csum 0xb0
[20:23:50]v2843a5ac
[20:23:50]~ld
[20:23:50][I][logger:166]: Log initialized
[20:23:50][C][ota:366]: There have been 10 suspected unsuccessful boot attempts.
[20:23:50][E][ota:371]: Boot loop detected. Proceeding to safe mode.
[20:23:50][I][app:029]: Running through setup()...
[20:23:50][C][wifi:033]: Setting up WiFi...
[20:23:50][I][wifi:194]: WiFi Connecting to 'UniFi'...
[20:23:50][I][wifi:457]: WiFi Connected!
[20:23:50][C][wifi:303]:   SSID: [redacted]
[20:23:50][C][wifi:304]:   IP Address: 192.168.0.71
[20:23:50][C][wifi:306]:   BSSID: [redacted]
[20:23:50][C][wifi:307]:   Hostname: 'sonos_display_ssd1306'
[20:23:50][C][wifi:311]:   Signal strength: -59 dB ▂▄▆█
[20:23:50][C][wifi:315]:   Channel: 11
[20:23:50][C][wifi:316]:   Subnet: 255.255.255.0
[20:23:50][C][wifi:317]:   Gateway: 192.168.0.1
[20:23:50][C][wifi:318]:   DNS1: (IP unset)
[20:23:50][C][wifi:319]:   DNS2: (IP unset)
[20:23:50][D][wifi:466]: Disabling AP...
[20:23:50]
[20:23:50] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[20:23:50]
[20:23:50]wdt reset
[20:23:50]load 0x4010f000, len 3584, room 16 
[20:23:50]tail 0
[20:23:50]chksum 0xb0
[20:23:50]csum 0xb0
[20:23:50]v2843a5ac
[20:23:50]~ld
[20:23:50][I][logger:166]: Log initialized
[20:23:50][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[20:23:50][I][app:029]: Running through setup()...
[20:23:50][D][binary_sensor:034]: 'Sonos Display ssd1306 PIR': Sending initial state OFF
[20:23:50][D][binary_sensor:034]: 'Sonos Display ssd1306 Status': Sending initial state OFF
[20:23:50][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[20:23:50]
[20:23:50]User exception (panic/abort/assert)
[20:23:50]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[20:23:50]
[20:23:50]Panic core_esp8266_main.cpp:206 loop_task
[20:23:50]
[20:23:50]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[20:23:50]
[20:23:50]ctx: sys
[20:23:50]sp: 3fffeee0 end: 3fffffb0 offset: 0000
[20:23:50]3fffeee0:  ffffffff ffffffff ffffffff ffffffff  
[20:23:50]3fffeef0:  000000fe 00000000 00000000 00000000  
[... 448 lines cut ...]
[20:23:55][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[20:23:55]
[20:23:55]User exception (panic/abort/assert)
[20:23:55]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[20:23:55]
[20:23:55]Panic core_esp8266_main.cpp:206 loop_task
[20:23:55]
[20:23:55]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[20:23:55]
[20:23:55]ctx: sys
[20:23:55]sp: 3fffeee0 end: 3fffffb0 offset: 0000
[20:23:55]3fffeee0:  ffffffff ffffffff ffffffff ffffffff  
[20:23:55]3fffeef0:  000000fe 00000000 00000000 00000000  
[20:23:55]3fffef00:  00000000 00000000 00000000 3fff1540  
[20:23:55]3fffef10:  3fff160c 3fff160c ffffffff 3fff1500  
[20:23:55]3fffef20:  3fffdad0 00000000 3ffe857c 402250f6  
WARNING Decoded 0x402250f6: raise_exception at core_esp8266_postmortem.cpp
[20:23:55]3fffef30:  ffffffff ffffffff ffffffff 40225153  
WARNING Decoded 0x40225153: __panic_func
[20:23:55]3fffef40:  00000000 3fffdad0 3fff1500 40224b2a  
WARNING Decoded 0x40224b2a: loop_task(ETSEventTag*) at core_esp8266_main.cpp
[20:23:55]3fffef50:  400301e9 3fffdab0 3fffdab0 40000f49  
[20:23:55]3fffef60:  40000e19 00000005 000720d0 00000000
```

alexyao2015, 2021-01-16:
Can you `alexyao2015:feature/safe-mode-brick` in your development addon's `esphome_version` config option? 

Then add this config option to ota. See if it still is a permanent bootloop.

```yaml
ota:
   store_bootloops_in_flash_and_brick: true
```

htvekov, 2021-01-16:
Hi' @alexyao2015 

Just tested.
Still stack trace error and constant reboots I'm afraid.
But at least it tries to boot into safe mode (after 10 runs) and also fetches some platformio IDE-data

```
[23:37:56]wdt reset
[23:37:56]load 0x4010f000, len 3584, room 16 
[23:37:56]tail 0
[23:37:56]chksum 0xb0
[23:37:56]csum 0xb0
[23:37:56]v2843a5ac
[23:37:56]~ld
[23:37:56][I][logger:166]: Log initialized
[23:37:56][C][ota:369]: There have been 10 suspected unsuccessful boot attempts.
[23:37:56][E][ota:376]: Boot loop detected. Proceeding to safe mode.
[23:37:56][I][app:029]: Running through setup()...
[23:37:56][C][wifi:033]: Setting up WiFi...
[23:37:56][I][wifi:194]: WiFi Connecting to 'UniFi'...
[23:38:00][I][wifi:457]: WiFi Connected!
[23:38:00][C][wifi:303]:   SSID: [redacted]
[23:38:00][C][wifi:304]:   IP Address: 192.168.0.71
[23:38:00][C][wifi:306]:   BSSID: [redacted]
[23:38:00][C][wifi:307]:   Hostname: 'sonos_display_ssd1306'
[23:38:00][C][wifi:311]:   Signal strength: -58 dB ▂▄▆█
[23:38:00][C][wifi:315]:   Channel: 11
[23:38:00][C][wifi:316]:   Subnet: 255.255.255.0
[23:38:00][C][wifi:317]:   Gateway: 192.168.0.1
[23:38:00][C][wifi:318]:   DNS1: (IP unset)
[23:38:00][C][wifi:319]:   DNS2: (IP unset)
[23:38:00][D][wifi:466]: Disabling AP...
[23:38:09]
[23:38:09] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[23:38:09]
[23:38:09]wdt reset
[23:38:09]load 0x4010f000, len 3584, room 16 
[23:38:09]tail 0
[23:38:09]chksum 0xb0
[23:38:09]csum 0xb0
[23:38:09]v2843a5ac
[23:38:09]~ld
[23:38:09][I][logger:166]: Log initialized
[23:38:09][C][ota:369]: There have been 10 suspected unsuccessful boot attempts.
[23:38:09][E][ota:376]: Boot loop detected. Proceeding to safe mode.
[23:38:09][I][app:029]: Running through setup()...
[23:38:09][C][wifi:033]: Setting up WiFi...
[23:38:09][I][wifi:194]: WiFi Connecting to 'UniFi'...
[23:38:12][I][wifi:457]: WiFi Connected!
[23:38:12][C][wifi:303]:   SSID: [redacted]
[23:38:12][C][wifi:304]:   IP Address: 192.168.0.71
[23:38:13][C][wifi:306]:   BSSID: [redacted]
[23:38:13][C][wifi:307]:   Hostname: 'sonos_display_ssd1306'
[23:38:13][C][wifi:311]:   Signal strength: -58 dB ▂▄▆█
[23:38:13][C][wifi:315]:   Channel: 11
[23:38:13][C][wifi:316]:   Subnet: 255.255.255.0
[23:38:13][C][wifi:317]:   Gateway: 192.168.0.1
[23:38:13][C][wifi:318]:   DNS1: (IP unset)
[23:38:13][C][wifi:319]:   DNS2: (IP unset)
[23:38:13][D][wifi:466]: Disabling AP...
[23:38:21]
[23:38:21] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[23:38:21]
[23:38:21]wdt reset
[23:38:21]load 0x4010f000, len 3584, room 16 
[23:38:21]tail 0
[23:38:21]chksum 0xb0
[23:38:21]csum 0xb0
[23:38:21]v2843a5ac
[23:38:21]~ld
[... 818 lines cut ...]
[23:38:38]3fffef80:  4010512b 000720d0 00000000 3ffe9ecd  
WARNING Decoded 0x4010512b: call_user_start_local
[23:38:38]3fffef90:  4010000d 00000027 00000002 00040000  
WARNING Decoded 0x4010000d: call_user_start
[23:38:38]3fffefa0:  4024c8b8 3fffef3c 4024c869 3fffeee8  
WARNING Decoded 0x4024c8b8: cont_ret at cont.S.o
WARNING Decoded 0x4024c869: cont_continue at cont.S.o
[23:38:38]3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
[23:38:38]3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
[23:38:38]3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
[23:38:38]3fffefe0:  00000003 00000000 0000000a 00000000  
[23:38:38]3fffeff0:  00000003 00000000 0000000a 00000000  
[23:38:38]3ffff000:  feefeffe feefeffe feefeffe feefeffe  
[23:38:38]3ffff010:  00000000 a0000000 00000000 0000001c  
[23:38:38]3ffff020:  00003000 feefeffe 00003000 00000000  
[23:38:38]3ffff030:  3ffff190 00000000 3ffff190 40231a32  
WARNING Decoded 0x40231a32: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:233
[23:38:38]3ffff040:  0000a000 3ffff133 0000000a 00000000  
[23:38:38]3ffff050:  00000001 00000000 4022c0f5 00000003  
WARNING Decoded 0x4022c0f5: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:194 (discriminator 1)
[23:38:39]3ffff060:  3ffff190 00000003 3ffff190 40231a32  
WARNING Decoded 0x40231a32: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:233
[23:38:39]3ffff070:  00000004 00000000 0000000a 00000000  
[23:38:39]3ffff080:  00000004 00000000 0000000a 00000000
```

alexyao2015, 2021-01-23:
I just remembered, I didn't merge the automation fix into that safe mode brick branch. Try it now. It shouldn't have an eternal bootloop anymore.

htvekov, 2021-01-23:
Hi' @alexyao2015 

Just tested again.
I'm afraid that the eternal boot loop is still present 🤔

I had my working yaml running for long enough to reset the boot loop counter.
Then i reflashed with 'issue' version via serial.
Once thing that I find curious is that on very first boot the 10 suspected boot attemtps message appears in log ?
Unless your code does that deliberately ?

```
Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:21:04];$$\xdc\xdf<\x8cl\xe0|\xc4d\xe4c|\x87\xc3\xe4\xdb;\x9bc\x84c\xc4\xfb''\x9fln'\xdc\xe3\xeccx\xc4\xc7$;$;dp\xfbg\xe0\x83lĜc'\xe2|\xec$\x8f\xc7c\x8c\xfbn'\xeel\x8c\x8f$ \x90ood os\x87\x9b\x93g#d`;ۛ'#d`\xdc#\xc7\x92$;\x9b \xfc\x83'\x9c[I][logger:166]: Log initialized
[15:21:04][C][ota:369]: There have been 10 suspected unsuccessful boot attempts.
[15:21:04][E][ota:376]: Boot loop detected. Proceeding to safe mode.
[15:21:04][I][app:029]: Running through setup()...
[15:21:04][C][wifi:033]: Setting up WiFi...
[15:21:04][I][wifi:194]: WiFi Connecting to 'UniFi'...
[15:21:07][W][wifi_esp8266:441]: Event: Disconnected ssid='UniFi' reason='Probe Request Unsuccessful'
[15:21:07][W][wifi:485]: Error while connecting to network.
[15:21:07][W][wifi:522]: Restarting WiFi adapter...
[15:21:07][I][wifi:194]: WiFi Connecting to 'UniFi'...
[15:21:11][I][wifi:457]: WiFi Connected!
[15:21:11][C][wifi:303]:   SSID: [redacted]
[15:21:11][C][wifi:304]:   IP Address: 192.168.0.71
[15:21:11][C][wifi:306]:   BSSID: [redacted]
[15:21:11][C][wifi:307]:   Hostname: 'sonos_display_ssd1306'
[15:21:11][C][wifi:311]:   Signal strength: -57 dB ▂▄▆█
[15:21:11][C][wifi:315]:   Channel: 11
[15:21:11][C][wifi:316]:   Subnet: 255.255.255.0
[15:21:11][C][wifi:317]:   Gateway: 192.168.0.1
[15:21:11][C][wifi:318]:   DNS1: (IP unset)
[15:21:11][C][wifi:319]:   DNS2: (IP unset)
[15:21:11][D][wifi:466]: Disabling AP...
[15:21:11][C][ota:029]: Over-The-Air Updates:
[15:21:11][C][ota:030]:   Address: 192.168.0.71:8266
[15:21:11][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[15:21:11][I][app:059]: setup() finished successfully!
[15:21:11][I][ota:386]: Waiting for OTA attempt.
[15:21:11]
[15:21:11]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:21:11]
[15:21:11]Exception (28):
WARNING Exception type: Access to invalid address: LOAD (wild pointer?)
[15:21:11]epc1=0x4021769a epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000298b depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /config/esphome/sonos_display_ssd1306 -t idedata
WARNING Decoded 0x4021769a: WiFiClient::operator=(WiFiClient const&)
[15:21:16]
[15:21:16]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:21:16]
[15:21:16]ctx: sys
[15:21:16]sp: 3fffeda0 end: 3fffffb0 offset: 0190
[15:21:16]3fffef30:  40253e7a 3fffdab0 00000000 3ffe857c  
WARNING Decoded 0x40253e7a: ets_timer_handler_isr
[15:21:16]3fffef40:  3fff2c7c 3fff2c7c 3ffe857c 40208ad2  
WARNING Decoded 0x40208ad2: esphome::ota::OTAComponent::handle_()
[15:21:16]3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
[15:21:16]3fffef60:  40000e19 00000005 000720d0 00000000  
[15:21:16]3fffef70:  00000000 aa55aa55 000000ed 40105125  
WARNING Decoded 0x40105125: call_user_start_local
[15:21:16]3fffef80:  4010512b 000720d0 00000000 3ffe9ecd  
WARNING Decoded 0x4010512b: call_user_start_local
[15:21:16]3fffef90:  4010000d 3ffe9ecd 3ffe9ecd 3ffe9ecd  
WARNING Decoded 0x4010000d: call_user_start
[15:21:16]3fffefa0:  4024c8b8 3fffef3c 4024c869 3fffeee8  
[... 2385 lines cut ...]
WARNING Exception type: Access to invalid address: LOAD (wild pointer?)
[15:22:04]epc1=0x4021769a epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000298b depc=0x00000000
WARNING Decoded 0x4021769a: WiFiClient::operator=(WiFiClient const&)
WARNING Decoded 0x4021769a: WiFiClient::operator=(WiFiClient const&)
[15:22:05]
[15:22:05]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:22:05]
[15:22:05]ctx: sys
[15:22:05]sp: 3fffeda0 end: 3fffffb0 offset: 0190
[15:22:05]3fffef30:  40253e7a 3fffdab0 00000000 3ffe857c  
WARNING Decoded 0x40253e7a: ets_timer_handler_isr
[15:22:05]3fffef40:  3fff2c7c 3fff2c7c 3ffe857c 40208ad2  
WARNING Decoded 0x40208ad2: esphome::ota::OTAComponent::handle_()
[15:22:05]3fffef50:  40000f49 40000f49 3fffdab0 40000f49  
[15:22:05]3fffef60:  40000e19 00000005 000720d0 00000000  
[15:22:05]3fffef70:  00000000 aa55aa55 000000ed 40105125  
WARNING Decoded 0x40105125: call_user_start_local
[15:22:05]3fffef80:  4010512b 000720d0 00000000 00000030  
WARNING Decoded 0x4010512b: call_user_start_local
[15:22:05]3fffef90:  4010000d 00000025 00000002 00040000  
WARNING Decoded 0x4010000d: call_user_start
[15:22:05]3fffefa0:  4024c8b8 3fffef3c 4024c869 3fffeee8  
WARNING Decoded 0x4024c8b8: cont_ret at cont.S.o
```

htvekov, 2021-01-23:
Hi´ @alexyao2015 

Reflashed to known working ESPHome yaml configuration.
This won't start with your branch version. Also enters eternal boot loop ?
Normally this configuration starts at max. 3 boot attempts 

```
[15:32:16]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:32:16]
[15:32:16] ets Jan  8 2013,rst cause:4, boot mode:(3,7)
[15:32:16]
[15:32:16]wdt reset
[15:32:16]load 0x4010f000, len 3584, room 16 
[15:32:16]tail 0
[15:32:16]chksum 0xb0
[15:32:16]csum 0xb0
[15:32:16]v2843a5ac
[15:32:16]~ld
[15:32:16][I][logger:166]: Log initialized
[15:32:16][C][ota:369]: There have been 10 suspected unsuccessful boot attempts.
[15:32:16][E][ota:376]: Boot loop detected. Proceeding to safe mode.
[15:32:16][I][app:029]: Running through setup()...
[15:32:16][C][wifi:033]: Setting up WiFi...
[15:32:16][I][wifi:194]: WiFi Connecting to 'UniFi'...
[15:32:19][I][wifi:457]: WiFi Connected!
[15:32:19][C][wifi:303]:   SSID: [redacted]
[15:32:19][C][wifi:304]:   IP Address: 192.168.0.71
[15:32:19][C][wifi:306]:   BSSID: [redacted]
[15:32:19][C][wifi:307]:   Hostname: 'sonos_display_ssd1306'
[15:32:19][C][wifi:311]:   Signal strength: -59 dB ▂▄▆█
[15:32:19][C][wifi:315]:   Channel: 11
[15:32:19][C][wifi:316]:   Subnet: 255.255.255.0
[15:32:19][C][wifi:317]:   Gateway: 192.168.0.1
[15:32:19][C][wifi:318]:   DNS1: (IP unset)
[15:32:19][C][wifi:319]:   DNS2: (IP unset)
[15:32:19][D][wifi:466]: Disabling AP...
[15:32:19][C][ota:029]: Over-The-Air Updates:
[15:32:19][C][ota:030]:   Address: 192.168.0.71:8266
[15:32:19][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[15:32:19][I][app:059]: setup() finished successfully!
[15:32:19][I][ota:386]: Waiting for OTA attempt.
[15:32:19]
[15:32:19]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[15:32:19]
[15:32:19]Exception (28):
WARNING Exception type: Access to invalid address: LOAD (wild pointer?)
[15:32:19]epc1=0x402174aa epc2=0x00000000 epc3=0x00000000 excvaddr=0x00071e66 depc=0x00000000
WARNING Decoded 0x402174aa: WiFiClient::operator=(WiFiClient const&)
[15:32:19]
[15:32:19]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:32:19]
[15:32:19]ctx: sys
[15:32:19]sp: 3fffedb0 end: 3fffffb0 offset: 0190
[15:32:19]3fffef40:  00000000 3fffdad0 3fff14c8 40224940  
WARNING Decoded 0x40224940: loop_task(ETSEventTag*) at core_esp8266_main.cpp
[15:32:19]3fffef50:  3fff2c7c 00000000 3fffef70 40208ad2  
WARNING Decoded 0x40208ad2: esphome::ota::OTAComponent::handle_()
[15:32:19]3fffef60:  40000e19 00000005 00071e40 00000000  
[15:32:19]3fffef70:  3fffefc0 aa55aa55 000000ed 40105125  
WARNING Decoded 0x40105125: call_user_start_local
[15:32:19]3fffef80:  4010512b 00071e40 00000000 ffff0208  
WARNING Decoded 0x4010512b: call_user_start_local
[15:32:19]3fffef90:  4010000d 0000009f 00010000 00000001  
WARNING Decoded 0x4010000d: call_user_start
[15:32:19]3fffefa0:  4024c678 3fffef3c 4024c629 3fffeef8  
WARNING Decoded 0x4024c678: cont_ret at cont.S.o
WARNING Decoded 0x4024c629: cont_continue at cont.S.o
[15:32:19]3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
[15:32:19]3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
[... 743 lines cut ...]
[15:32:31]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:32:31]
[15:32:31]ctx: sys
[15:32:31]sp: 3fffedb0 end: 3fffffb0 offset: 0190
[15:32:31]3fffef40:  00000000 3fffdad0 3fff14c8 40224940  
WARNING Decoded 0x40224940: loop_task(ETSEventTag*) at core_esp8266_main.cpp
[15:32:31]3fffef50:  3fff2c7c 00000000 3fffef70 40208ad2  
WARNING Decoded 0x40208ad2: esphome::ota::OTAComponent::handle_()
[15:32:31]3fffef60:  40000e19 00000005 00071e40 00000000  
[15:32:31]3fffef70:  3fffefc0 aa55aa55 000000ed 40105125  
WARNING Decoded 0x40105125: call_user_start_local
[15:32:31]3fffef80:  4010512b 00071e40 00000000 ffff0208  
WARNING Decoded 0x4010512b: call_user_start_local
[15:32:31]3fffef90:  4010000d 0000009f 00010000 00000001  
WARNING Decoded 0x4010000d: call_user_start
[15:32:31]3fffefa0:  4024c678 3fffef3c 4024c629 3fffeef8  
WARNING Decoded 0x4024c678: cont_ret at cont.S.o
WARNING Decoded 0x4024c629: cont_continue at cont.S.o
[15:32:31]3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
[15:32:31]3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
[15:32:31]3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
[15:32:31]3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
[15:32:31]3fffeff0:  00000003 00000000 0000000a 00000000
```

htvekov, 2021-01-23:
Typical boot with ESPHome v.1.17.0 dev
This time it booted on first attempt.

```
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[15:41:19]sll\x9c\x9e|\x84d\xe0<\x8c$\xe4#<\xc7\xc3\xe4\033\x93r\xdb#\xc4c\x8c\xfbn'\xdflgo\x9c\xe3\xecc8ďlsl;$8\xfb'\xe0\xc3d\x8c\x9cco\xe3|\xe4l\x87\x8e#\x84\xf2on\xefd\x84\x87l`\x98\033go$ o{\x87ۛ'#d ;\xdb\xdb'#$ \x9cb\x8e\x93l;\xdb \xfc\xc3'\xdc[I][logger:166]: Log initialized
[15:41:19][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[15:41:19][I][app:029]: Running through setup()...
[15:41:19][D][binary_sensor:034]: 'Sonos Display ssd1306 PIR': Sending initial state OFF
[15:41:19][D][binary_sensor:034]: 'Sonos Display ssd1306 Status': Sending initial state OFF
[15:41:19][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[15:41:19][C][wifi:033]: Setting up WiFi...
[15:41:19][I][wifi:194]: WiFi Connecting to 'UniFi'...
[15:41:19][D][switch:021]: 'Sonos Display ssd1306 On/Off' Turning ON.
[15:41:19][D][switch:045]: 'Sonos Display ssd1306 On/Off': Sending state ON
[15:41:24][I][wifi:457]: WiFi Connected!
[15:41:24][C][wifi:303]:   SSID: [redacted]
[15:41:24][C][wifi:304]:   IP Address: 192.168.0.71
[15:41:24][C][wifi:306]:   BSSID: [redacted]
[15:41:24][C][wifi:307]:   Hostname: 'sonos_display_ssd1306'
[15:41:24][C][wifi:311]:   Signal strength: -62 dB ▂▄▆█
[15:41:24][C][wifi:315]:   Channel: 11
[15:41:24][C][wifi:316]:   Subnet: 255.255.255.0
[15:41:24][C][wifi:317]:   Gateway: 192.168.0.1
[15:41:24][C][wifi:318]:   DNS1: (IP unset)
[15:41:24][C][wifi:319]:   DNS2: (IP unset)
[15:41:24][D][wifi:466]: Disabling AP...
[15:41:24][C][ota:029]: Over-The-Air Updates:
[15:41:24][C][ota:030]:   Address: 192.168.0.71:8266
[15:41:24][C][api:022]: Setting up Home Assistant API server...
[15:41:24][I][app:059]: setup() finished successfully!
[15:41:24][D][sensor:092]: 'Sonos Display ssd1306 WiFi Signal': Sending state -62.00000 dB with 0 decimals of accuracy
[15:41:24][I][app:105]: ESPHome version 1.17.0-dev compiled on Jan 23 2021, 15:40:17
[15:41:24][C][wifi:443]: WiFi:
[15:41:24][C][wifi:303]:   SSID: [redacted]
[15:41:24][C][wifi:304]:   IP Address: 192.168.0.71
[15:41:24][C][wifi:306]:   BSSID: [redacted]
[15:41:24][C][wifi:307]:   Hostname: 'sonos_display_ssd1306'
[15:41:24][C][wifi:311]:   Signal strength: -62 dB ▂▄▆█
[15:41:24][C][wifi:315]:   Channel: 11
[15:41:24][C][wifi:316]:   Subnet: 255.255.255.0
[15:41:24][C][wifi:317]:   Gateway: 192.168.0.1
[15:41:24][C][wifi:318]:   DNS1: (IP unset)
[15:41:24][C][wifi:319]:   DNS2: (IP unset)
[15:41:24][C][i2c:028]: I2C Bus:
[15:41:24][C][i2c:029]:   SDA Pin: GPIO5
[15:41:24][C][i2c:030]:   SCL Pin: GPIO4
[15:41:24][C][i2c:031]:   Frequency: 100000 Hz
[15:41:24][I][i2c:033]: Scanning i2c bus for active devices...
[15:41:24][I][i2c:040]: Found i2c device at address 0x3C
[15:41:24][C][template.switch:058]: Template Switch 'Sonos Display ssd1306 On/Off'
[15:41:24][C][template.switch:059]:   Restore State: NO
[15:41:24][C][template.switch:060]:   Optimistic: NO
[15:41:24][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Sonos Display ssd1306 PIR'
[15:41:24][C][gpio.binary_sensor:015]:   Device Class: 'motion'
[15:41:24][C][gpio.binary_sensor:016]:   Pin: GPIO14 (Mode: INPUT)
[15:41:24][C][template.binary_sensor:018]: Template Binary Sensor 'dim_display'
[15:41:24][C][template.binary_sensor:018]: Template Binary Sensor 'display_off'
[15:41:24][C][logger:185]: Logger:
[15:41:24][C][logger:186]:   Level: DEBUG
[15:41:24][C][logger:187]:   Log Baud Rate: 115200
[15:41:24][C][logger:188]:   Hardware UART: UART0
[15:41:24][C][homeassistant.text_sensor:011]: Homeassistant Text Sensor 'sonos_status'
[15:41:24][C][homeassistant.text_sensor:012]:   Entity ID: 'media_player.kokken'
[15:41:24][C][homeassistant.text_sensor:011]: Homeassistant Text Sensor 'media_source'
[15:41:24][C][homeassistant.text_sensor:012]:   Entity ID: 'sensor.media_source'
[15:41:25][C][homeassistant.text_sensor:011]: Homeassistant Text Sensor 'media_artist'
[15:41:25][C][homeassistant.text_sensor:012]:   Entity ID: 'sensor.media_artist'
[... 44 lines cut ...]
[15:41:35][D][api.connection:604]: Client 'Home Assistant 2020.12.2 (192.168.0.19)' connected successfully!
[15:41:35][D][time:040]: Synchronized time: Sat Jan 23 15:41:35 2021
[15:41:35][D][homeassistant.text_sensor:016]: 'media_player.kokken': Got state 'paused'
[15:41:35][D][text_sensor:015]: 'sonos_status': Sending state 'paused'
[15:41:35][D][homeassistant.text_sensor:016]: 'sensor.media_source': Got state 'DR P4 Fyn'
[15:41:35][D][text_sensor:015]: 'media_source': Sending state 'DR P4 Fyn'
[15:41:35][D][homeassistant.text_sensor:016]: 'sensor.media_artist': Got state 'DR P4 Fyn'
[15:41:35][D][text_sensor:015]: 'media_artist': Sending state 'DR P4 Fyn'
[15:41:35][D][homeassistant.text_sensor:016]: 'sensor.media_title': Got state 'unknown'
[15:41:35][D][text_sensor:015]: 'media_title': Sending state 'unknown'
[15:41:35][D][homeassistant.text_sensor:016]: 'sensor.sonos_master_friendly': Got state 'Køkken'
[15:41:35][D][text_sensor:015]: 'sonos_master': Sending state 'Køkken'
[15:41:35][D][homeassistant.text_sensor:016]: 'sensor.sonos_slaves_friendly': Got state 'Alrum, Stue'
[15:41:35][D][text_sensor:015]: 'sonos_slaves': Sending state 'Alrum, Stue'
[15:41:35][D][homeassistant.binary_sensor:021]: 'binary_sensor.is_volume_muted': Got state OFF
[... cut ...]
[15:41:35][D][homeassistant.sensor:019]: 'sensor.0x00158d0002d813f0_temperature': Got state 7.10
[15:41:35][D][sensor:092]: 'outdoor_temp': Sending state 7.10000  with 1 decimals of accuracy
[15:41:35][D][homeassistant.sensor:019]: 'sensor.volume': Got state 10.00
[15:41:35][D][sensor:092]: 'sonos_volume': Sending state 10.00000  with 1 decimals of accuracy
[15:42:19][D][sensor:092]: 'Sonos Display ssd1306 WiFi Signal': Sending state -61.00000 dB with 0 decimals of accuracy
[15:42:31][D][binary_sensor:036]: 'dim_display': Sending state OFF
```

alexyao2015, 2021-01-23:
Can you perhaps edit all your messages and attach all logs instead of pasting them here directly? I can't see any of your comments because they are filled with enormous logs.

alexyao2015, 2021-01-23:
In the meantime, how did you flash via serial? Did you first erase the flash then write? Are you able to post your main.cpp file as an attachment?

htvekov, 2021-01-23:
> Can you perhaps edit all your messages and attach all logs instead of pasting them here directly? I can't see any of your comments because they are filled with enormous logs.

👍
Done !

htvekov, 2021-01-23:
Hi' @alexyao2015

All flashing has been done using exactly  same method.
DIrectly from HACS version of ESPHome  via WebUI (Upload button).

I can't get the working known yaml to boot (output just like log#03.txt) using your branch with `store_bootloops_in_flash_and_brick: true`, when first `boot loop` has been detected.
I have to flash a known working yaml with `store_bootloops_in_flash_and_brick: false` and let the ESPHome sw reset the `boot loop` bit first (after a few minutes run time).

Then I can reflash known working yaml again with `store_bootloops_in_flash_and_brick: true` and then it will boot.

After a succesfull boot with your branch and  `store_bootloops_in_flash_and_brick: true` i revised yaml once again and added an extra switch. Reflashed as usual as it ended up i `boot loop`  (see log#05.txt)

[log#05.txt](https://github.com/esphome/issues/files/5860971/log.05.txt)

Attached below main.ccp with `store_bootloops_in_flash_and_brick:` set as `false` and `true`

Don't really know what else to test and where to go from here ?

[main_working_yaml_store_bootloops_in_flash_and_brick_true.zip](https://github.com/esphome/issues/files/5860981/main_working_yaml_store_bootloops_in_flash_and_brick_true.zip)

[main_working_yaml_store_bootloops_in_flash_and_brick_false.zip](https://github.com/esphome/issues/files/5860982/main_working_yaml_store_bootloops_in_flash_and_brick_false.zip)

htvekov, 2021-01-23:
Hi' @alexyao2015

All flashing has been done using exactly same method.
DIrectly from HACS version of ESPHome via WebUI (Upload button).

I can't get the working known yaml to boot (output just like log#03.txt) using your branch with store_bootloops_in_flash_and_brick: true, when first boot loop has been detected.
I have to flash a known working yaml with store_bootloops_in_flash_and_brick: false and let the ESPHome sw reset the boot loop bit first (after a few minutes run time).

Then I can reflash known working yaml again with store_bootloops_in_flash_and_brick: true and then it will boot.

After a succesfull boot with your branch and store_bootloops_in_flash_and_brick: true i revised yaml once again and added an extra switch. Reflashed as usual as it ended up in eternal boot loop (see log#05.txt)

log#05.txt

Attached below main.ccp with store_bootloops_in_flash_and_brick: set as false and true

Don't really know what else to test and where to go from here ?

alexyao2015, 2021-01-24:
Can you see if you can get it crashing in the same way without including a display or font in your yaml? I see you did add many many fonts into your config. 

That setting basically stores the boot loop counter in flash to hopefully prevent it from having an eternal boot loop by keeping it in safe mode until it receives an OTA. I did just rebase that branch onto the new dev branch so that all the changes should now be in it. So try it again with and without the display + font.

gretel, 2021-01-29:
the only way me could work around this issue is using far less (8) `median` filters. mostly the same configuration is working perfectly fine on a `esp32`. as long as the filters were declared any use of `interval` resulted in crashing on the first loop.

closed 2021-06-13 by stale[bot]