esphome/issues #1725: Device show offline but everyone working fine

jp181278, 2020-12-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

rpi3b+ 
## System Health

version | 2020.12.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.6
os_name | Linux
os_version | 5.4.79-v7
arch | armv7l
timezone | Europe/Rome

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.9.0
Stage | running
Available Repositories | 711
Installed Repositories | 22

host_os | Home Assistant OS 5.9
update_channel | stable
supervisor_version | 2020.12.7
docker_version | 19.03.13
disk_total | 109.3 GB
disk_used | 11.7 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.2.0), Samba share (9.3.0), Check Home Assistant configuration (3.6.0), Zigbee2mqtt (1.16.2.2), Mosquitto broker (5.1), ESPHome (1.15.3), SSH & Web Terminal (7.8.0), Duck DNS (1.12.4), Network UPS Tools (0.3.1), Home Assistant Google Drive Backup (0.102.0), Grafana (5.3.6), InfluxDB (3.7.9), Spotify Connect (0.8.2)

**ESP (ESP32/ESP8266, Board/Sonoff):**

sonoff mini, sonoff basic
**ESPHome version (latest production, beta, dev branch)**

esphome 1.15.3
**Affected component:**

Dashboard Interface

**Description of problem:**
[image]
the dashboard tells that all the devices are offline but they work correctly
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2021-01-05:
Maybe related to your network and mdns. Did you try the ping method already?

jp181278, 2021-01-05:
It was an issue of my network that after a power glitch activate the ap isolation

closed 2021-01-05 by jp181278