esphome/issues #1726: Tuya IR + IR Remote Climate and Daikin FTXB AC = does't work.

avbor, 2020-12-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Supervised 2020.12.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (Tuya IR Transmitter UFO-R1, TYWE3S chip)

**ESPHome version (latest production, beta, dev branch)**

Latest production from hassio addon store.
Version 1.15.3

**Affected component:**

IR Remote Climate

**Description of problem:**

I have ACs Daikin FTXB20C/RXB20C and FTXB25C/RXB25C, but when i try to use platform "daikin" from "IR Remote Climate" component - it does not work. Commands are sent, but the AC does not respond.
Log:
```
[18:21:48][D][climate:010]: 'Daikin AC - Living Room' - Setting
[18:21:48][D][climate:014]:   Mode: AUTO
[18:21:48][D][remote_transmitter:066]: Sending remote code...
[18:21:48][D][climate:262]: 'Daikin AC - Living Room' - Sending state:
[18:21:48][D][climate:265]:   Mode: AUTO
[18:21:48][D][climate:270]:   Fan Mode: ON
[18:21:48][D][climate:273]:   Swing Mode: OFF
[18:21:48][D][climate:276]:   Current Temperature: 23.30°C
[18:21:48][D][climate:282]:   Target Temperature: 22.00°C
[18:21:48][D][remote.raw:041]: Received Raw: 
[18:21:55][D][climate:010]: 'Daikin AC - Living Room' - Setting
[18:21:55][D][climate:014]:   Mode: OFF
[18:21:55][D][remote_transmitter:066]: Sending remote code...
[18:21:55][D][climate:262]: 'Daikin AC - Living Room' - Sending state:
[18:21:55][D][climate:265]:   Mode: OFF
[18:21:55][D][climate:270]:   Fan Mode: ON
[18:21:55][D][climate:273]:   Swing Mode: OFF
[18:21:55][D][climate:276]:   Current Temperature: 23.30°C
[18:21:55][D][climate:282]:   Target Temperature: 22.00°C
[18:21:55][D][remote.raw:041]: Received Raw:
```
IR transmitter works fine.
When i try to setup component as described here: https://github.com/esphome/feature-requests/issues/444#issuecomment-548166019 - I can partially control my AC (on\off at least).
Or, if i use Tuya app and setup AC Daikin control from Tuya db - all works fine.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ir_ac_daikin_livingroom
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: "IR AC Daikin LivingRoom"
    password: !secret wifi_fallback_ap_pass

captive_portal:
logger:

api:
  password: !secret esphome_api_password

ota:
  password: !secret esphome_api_password
[... 116 lines cut ...]
[18:30:46][D][remote.raw:041]: Received Raw: 491, -415, 462, -410, 459, -409, 461, -410, 461, -408, 463
[18:30:46][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE400A300
[18:30:47][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE4004200
[18:30:47][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE400001C
[18:30:49][D][remote.raw:041]: Received Raw: 489, -414, 460, -412, 461, -408, 461, -408, 461, -409, 460
[18:30:49][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE400A300
[18:30:49][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE4004200
[18:30:50][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE400001C
[18:30:54][D][remote.raw:041]: Received Raw: 489, -417, 461, -409, 439, -430, 458, -412, 458, -412, 433
[18:30:54][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE400A300
[18:30:54][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE4004200
[18:30:54][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE4000012
[18:30:55][D][remote.raw:041]: Received Raw: 490, -416, 437, -432, 449, -420, 462, -407, 460, -412, 463
[18:30:55][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE400A300
[18:30:55][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE4004200
[18:30:55][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE4000016
[18:30:56][D][remote.raw:041]: Received Raw: 467, -437, 460, -412, 435, -434, 460, -410, 462, -407, 459
[18:30:56][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE400A300
[18:30:56][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE4004200
[18:30:56][D][remote.panasonic:068]: Received Panasonic: address=0x885B, command=0xE4000010
```
I dont know why Panasonic here =)
I tried all platforms from https://esphome.io/components/climate/ir_climate.html, but none of them worked for me.

Is there a way to make this device fully control my ACs?

hcoohb, 2021-01-08:
Hi,

- As to why you receive some panasonic code: you put the `dump` option to `all` so esphome will try to recognise the received data from the list of protocols it knows. It apparently does not correctly recognise your remote protocol.
- There is currently an issue with the Daikin in esphome 1.15.3 #1595 , so if you want to test, maybe try to use esphome 1.15.2  for now.
- Daikin has actually many protocols for the different models of HVAC.
Unfortunately, the official component in esphome, only has the most common protocol, ARC43*** models.
I have developed a component for the Daikin176 protocol which I will try to upstream as an option. But from the models of your AC you mention, they might actually work on another protocol (Daikin128) that I am not aware has been implemented in esphome yet. So for now if 1.15.2 does not work for you, I would think the best solution is to use the IRremoteESP8266 lib as you [mentioned](https://github.com/esphome/feature-requests/issues/444#issuecomment-548166019), but maybe trying protocol Daikin128 so `IRDaikin128` instead of  `IRDaikinESP`

avbor, 2021-01-21:
@hcoohb thanx for reply.
Unfortunately, I am a complete beginner with ESPHome and all around programming =(
Where I can look for examples about using IRremoteESP8266?
With this - https://github.com/esphome/feature-requests/issues/444#issuecomment-548166019 - I have earned a very small part of the functions...

avbor, 2021-02-04:
Finally I was able to get this to work with the IRremoteESP8266 (IRDaikinESP), here you can see [how](https://github.com/avbor/HomeAssistantConfig/blob/master/esphome/include/daikin_ir.h).
But I still wanted to see support in the standard platform. With the version of ESPHome 1.16 - it still does not work.

closed 2021-06-18 by stale[bot]