esphome/issues #1727: Neopixel or FastLED breaks Remote receiver function

Yourigh, 2020-12-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Esphome on ESP32 custom board (tested also on devkit with same results), control from HA HASS OS RPi3+. 
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 custom board: https://github.com/Yourigh/BrimEffect_ESP32

Equivalent to ESP32 devkit. IR receiver was connected directly.
**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.15.2
**Affected component:**

* Remote Receiver
* FastLED
* NeoPixel

**Description of problem:**
Remote IR receiver is connected to GPIO16. 

Receiving of IR commands work if 
* LED strip is not configured in yaml.
* LED strip is set to `platform: neopixelbus` and `method: BIT_BANG`

Receiving of IR commands does **not** work (no logs about received data) if:
* LED strip is configured any other way (I2S methods or using FastLED)

The issue is that on BitBang method, there is some flickering of LEDs,  what is expected. But I am not able to set up any other method or to use FastLED the way that IR receiver would work too. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: workroom_brimeffect
  platform: ESP32
  board: featheresp32 #nodemcu-32s

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  services:
  - service: play_rtttl
    variables:
      song_str: string
    then:
      - rtttl.play:
          rtttl: !lambda 'return song_str;'

ota:

wled:

light:
  - platform: rgb
    name: "On-board LED"
    red: output_component1
    green: output_component2
    blue: output_component3
    default_transition_length : 0s
    # https://esphome.io/components/light/neopixelbus.html
  - platform: neopixelbus
    method: BIT_BANG
    # does not work with ESP32_I2S_1 - cannot scan wifi
    # ESP32_I2S_0 - boot ok, but LEDs do not change state
    # BIT_BANG works but flickets occationaly
    variant: WS2813
    pin: GPIO21
    num_leds: 59
    type: GRB
    name: "LED Stripe"
    default_transition_length : 0.1s
    effects:
      - wled:

output:
  - platform: ledc
    id: output_component1
    pin: 25
    inverted: true
    frequency: "1220Hz"
  - platform: ledc
    id: output_component2
    pin: 26
    inverted: true
    frequency: "1220Hz"
  - platform: ledc
    id: output_component3
    pin: 27
    inverted: true
    frequency: "1220Hz"
  - platform: ledc  
    # for buzzer output
    pin: GPIO23
    id: rtttl_out

rtttl:
  output: rtttl_out

i2c:
  - id: bus_a
    sda: 33
    scl: 32
    scan: True
    
remote_receiver:
  dump: nec
  pin:
    number: 16
    inverted: True
    mode: INPUT_PULLUP
  on_nec:
    - rtttl.play: "two short:d=4,o=5,b=100:16e6,16e6"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I tried Configuring different methods on Neopixel. The remote and receiver works well, when LED strip is not configured in YAML, so hardware is ok. I also tried using different GPIOs for IR receiver, with no change in behaviour.

I would like to use FastLED and have functional remote receiver.

MunkeyBalls, 2021-01-11:
I've run into the same issue with Wemos Mini32 ESP32 trying to get WS2812b leds and tsop38238 IR receiver working with fastled. As soon as I add fasted to my config, receiving IR no longer works. When switching to Neopixelbus the ESP32 doesn't even seem to boot.

mpneuried, 2021-02-25:
Hi, i discovered the same issue.
Until i added the fastled config the ir receiver stoped to work.
Using neopixelbus the esp32 failed to boot

closed 2021-07-09 by stale[bot]