esphome/issues #1728: new mqtt::MQTT*Sensors not generated

misery, 2020-12-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ArchLinux with ESPHome with pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, NodeMCU

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

MQTT

**Description of problem:**
If I provide the mqtt component in a _device_base_ package it won't generate all MQTT-Sensors. It generates a connection to MQTT broker but there won't be any ```App.register_component```.

If I move the mqtt component to the main yaml file it works.
Here the diff of generated sources.

```diff
--- with_include/src/main.cpp
+++ without_example/src/main.cpp
@@ -8,9 +8,13 @@
 using namespace mqtt;
 using namespace json;
 wifi_info::IPAddressWiFiInfo *wifi_info_ipaddresswifiinfo;
+mqtt::MQTTTextSensor *mqtt_mqtttextsensor;
 wifi_info::SSIDWiFiInfo *wifi_info_ssidwifiinfo;
+mqtt::MQTTTextSensor *mqtt_mqtttextsensor_2;
 wifi_info::BSSIDWiFiInfo *wifi_info_bssidwifiinfo;
+mqtt::MQTTTextSensor *mqtt_mqtttextsensor_3;
 wifi_info::MacAddressWifiInfo *wifi_info_macaddresswifiinfo;
+mqtt::MQTTTextSensor *mqtt_mqtttextsensor_4;
 // ========== AUTO GENERATED INCLUDE BLOCK END ==========="
 
 void setup() {
@@ -131,32 +135,44 @@
   //   platform: wifi_info
   //   ip_address:
   //     name: IP
+  //     mqtt_id: mqtt_mqtttextsensor
   //     id: wifi_info_ipaddresswifiinfo
   //   ssid:
   //     name: SSID
+  //     mqtt_id: mqtt_mqtttextsensor_2
   //     id: wifi_info_ssidwifiinfo
   //   bssid:
   //     name: BSSID
+  //     mqtt_id: mqtt_mqtttextsensor_3
   //     id: wifi_info_bssidwifiinfo
   //   mac_address:
   //     name: MAC
+  //     mqtt_id: mqtt_mqtttextsensor_4
   //     id: wifi_info_macaddresswifiinfo
   wifi_info_ipaddresswifiinfo = new wifi_info::IPAddressWiFiInfo();
   App.register_component(wifi_info_ipaddresswifiinfo);
   App.register_text_sensor(wifi_info_ipaddresswifiinfo);
   wifi_info_ipaddresswifiinfo->set_name("IP");
+  mqtt_mqtttextsensor = new mqtt::MQTTTextSensor(wifi_info_ipaddresswifiinfo);
+  App.register_component(mqtt_mqtttextsensor);
   wifi_info_ssidwifiinfo = new wifi_info::SSIDWiFiInfo();
   App.register_component(wifi_info_ssidwifiinfo);
   App.register_text_sensor(wifi_info_ssidwifiinfo);
   wifi_info_ssidwifiinfo->set_name("SSID");
+  mqtt_mqtttextsensor_2 = new mqtt::MQTTTextSensor(wifi_info_ssidwifiinfo);
+  App.register_component(mqtt_mqtttextsensor_2);
   wifi_info_bssidwifiinfo = new wifi_info::BSSIDWiFiInfo();
   App.register_component(wifi_info_bssidwifiinfo);
   App.register_text_sensor(wifi_info_bssidwifiinfo);
   wifi_info_bssidwifiinfo->set_name("BSSID");
+  mqtt_mqtttextsensor_3 = new mqtt::MQTTTextSensor(wifi_info_bssidwifiinfo);
+  App.register_component(mqtt_mqtttextsensor_3);
   wifi_info_macaddresswifiinfo = new wifi_info::MacAddressWifiInfo();
   App.register_component(wifi_info_macaddresswifiinfo);
   App.register_text_sensor(wifi_info_macaddresswifiinfo);
   wifi_info_macaddresswifiinfo->set_name("MAC");
+  mqtt_mqtttextsensor_4 = new mqtt::MQTTTextSensor(wifi_info_macaddresswifiinfo);
+  App.register_component(mqtt_mqtttextsensor_4);
   // =========== AUTO GENERATED CODE END ============
   // ========= YOU CAN EDIT AFTER THIS LINE =========
   App.setup();
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: example
  platform: ESP32
  board: nodemcu-32s

packages:
  device_base: !include device_base.yaml

# Works if provided here
# mqtt:
#  broker: 192.168.123.99

text_sensor:
- platform: wifi_info
  ip_address:
    name: IP
  ssid:
    name: SSID
  bssid:
    name: BSSID
  mac_address:
    name: MAC
```

**device_base.yaml**
```yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  ap:
    ssid: Fallback Hotspot ${name}
    password: !secret wifi_recover_password

# Only version sensors works if provided here
mqtt:
  broker: 192.168.123.99
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

koenvervloesem, 2020-12-29:
Isn't this the same bug as https://github.com/esphome/issues/issues/1487?

misery, 2020-12-29:
Yeah, seems so....

closed 2020-12-29 by misery