esphome/issues #1729: ESP32. Device randomly loost connection to API

Abbadon89, 2020-12-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Addon for hassio, latest

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 on NodeMCU

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

Unknown

**Description of problem:**

Device randomly loost connection to API

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  plug_name: esp_GPIO
esphome:
  name: esp_gpio
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "###########"
  fast_connect: true
  password: "##############"
  power_save_mode: none
  use_address: 192.168.2.115 
  manual_ip:
    static_ip: 192.168.2.115
    subnet: 255.255.255.0     
    gateway: 192.168.2.1
 # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp Gpio Fallback Hotspot"
    password: "#########"

captive_portal:
web_server:
  port: 80
  auth:
    username: ######
    password: #########

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  password: "########"

ota:
  password: "########"
  
dallas:
  - pin: GPIO4
    update_interval: 120s

time:
  - platform: homeassistant
    id: homeassistant_time

[... 253 lines cut ...]
}
[22:36:32][VV][api.service:220]: on_ping_request: PingRequest {}
[22:36:32][VV][api.service:032]: send_ping_response: PingResponse {}
**WARNING Disconnected from API: Timeout while waiting for message response!**
INFO Connecting to 192.168.2.115:6053 (192.168.2.115)
INFO Successfully connected to 192.168.2.115
[22:36:49][VV][api.service:192]: on_hello_request: HelloRequest {
  client_info: 'Home Assistant 2020.12.1'
}
[22:36:49][V][api.connection:588]: Hello from client: 'Home Assistant 2020.12.1 (192.168.2.3)'
[22:36:49][VV][api.service:012]: send_hello_response: HelloResponse {
  api_version_major: 1
  api_version_minor: 3
  server_info: 'esp_gpio (esphome v1.15.3)'
}
[22:36:49][VV][api.service:199]: on_connect_request: ConnectRequest {
  password: '#####'
}
[22:36:49][D][api.connection:604]: Client 'Home Assistant 2020.12.1 (192.168.2.3)' connected successfully!

```

**Additional information and things you've tried:**

Changing borad, reupload, changing power suply unit

hawkeye217, 2021-01-11:
I have the same issue on my ESP32 nodes only.  My ESP8266 nodes are rock solid.  Seemingly random wifi disconnects, even with ``power_save: none``.

broyuken, 2021-01-25:
I have random disconnects on my esp8266 wifi bulbs. They don't reboot as the uptime sensor doesn't reset but they lose connection to the API for anywhere from a few seconds up to a few hours. I've tried moving them right next to my AP without any change in behavior. I can't really hook this up to serial so I'm not sure what debugging I can do outside of providing logs that match the above.

```
**WARNING Disconnected from API: Timeout while waiting for message response!**
```

broyuken, 2021-01-25:
Just did some testing, light was up and stable for about 30 minutes while I had the logs window open in ESPHome addon. As soon as I closed it, within a minute my light went unavailable. Got the following from the webui logs.
```
[D][sensor:092]: 'TBD Light Uptime': Sending state 30.64370 minutes with 0 decimals of accuracy
[D][api:067]: Disconnecting ESPHome v1.15.3 (10.0.86.138)
[D][sensor:092]: 'TBD Light Uptime': Sending state 31.64372 minutes with 0 decimals of accuracy
[D][api:067]: Disconnecting Home Assistant 2021.1.5 (10.0.86.138)
[D][binary_sensor:036]: 'TBD Light Status': Sending state OFF
[D][binary_sensor:036]: 'TBD Light Status': Sending state ON
[D][api.connection:604]: Client 'Home Assistant 2021.1.5 (10.0.86.138)' connected successfully!
[D][sensor:092]: 'TBD Light Uptime': Sending state 32.64370 minutes with 0 decimals of accuracy
```

broyuken, 2021-01-26:
Also I have 2 bulbs showing the same behavior so I'm doubting it's the bulb itself.

broyuken, 2021-01-26:
Also working out of this issue. https://github.com/home-assistant/core/issues/35540 

I set a static IP on my esp and I haven't had a drop in 3+ hours. Still too early to say if this solved it but it's got promise.

broyuken, 2021-01-26:
Just dropped out

mmakaay, 2021-03-18:
I found an issue with random disconnects on my devices, which were caused by a race condition issue in the AsyncTCP library.
I filed a [bug report and a possible fix](https://github.com/me-no-dev/AsyncTCP/issues/116) for it with the upstream project, but I don't know how actively maintained that project is.

For testing, I applied the fix to the forked AsyncTCP-esphome library as well, and using that version my devices are rock solid. No connection drops for over 12 hours, while normally it would drop up to multiple times per minute, depending on the work load.
Check out [this posting](https://community.home-assistant.io/t/solved-esphome-dhcp-client-problems-temporary-solution-with-manual-fixed-ip/147312/37?u=mmakaay) where I describe the patch and how you can test your code with that version.

broyuken, 2021-03-18:
I went back to tasmota which has been rock solid for me.

closed 2021-07-31 by stale[bot]