esphome/issues #1731: Bootloop when BLE Tracker configured

sliwma, 2020-12-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

latest Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-WROOM-32D
**ESPHome version (latest production, beta, dev branch)**

1.15.3
**Affected component:**

esp32_ble_tracker

**Description of problem:**
When esp32_ble_tracker is configured, board does not connect to WiFi and keeps rebooting.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: lamp_midesklamp1s
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@1.11.0
    platform_packages: |-4
          framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.4
  
logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.12.21
    gateway: 192.168.12.1
    subnet: 255.255.255.0

  ap:
    ssid: "MiDeskLamp1S"
    password: !secret fallback

api:
  password: !secret api_lamp_midesklamp1s

ota:
  password: !secret api_lamp_midesklamp1s

esp32_ble_tracker:
```

**Logs (if applicable):**

```
Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][logger:166]: Log initialized
[C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[... 1261 lines cut ...]
load:0x40080400,len:5828
entry 0x400806ac
[I][logger:166]: Log initialized
[C][ota:366]: There have been 5 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][power_supply:010]: Setting up Power Supply...
[D][binary_sensor:034]: 'button': Sending initial state OFF
[C][light:097]: Setting up light 'MiDeskLamp1S'...
[D][light:265]: 'MiDeskLamp1S' Setting:
[D][light:278]:   Brightness: 100%
[D][light:282]:   Color Temperature: 1.0 mireds
[D][light:287]:   Red=100%, Green=100%, Blue=100%

Log from the ESPHome addon:
INFO Reading configuration /config/esphome/lamp_midesklamp1s.yaml...
INFO Starting log output from 192.168.12.21 using esphome API
INFO Connecting to 192.168.12.21:6053 (192.168.12.21)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.12.21: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to 192.168.12.21:6053 (192.168.12.21)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.12.21: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to 192.168.12.21:6053 (192.168.12.21)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.12.21: [Errno 111] Connection refused). Re-Trying in 2 seconds
```

**Additional information and things you've tried:**

Cougar, 2021-01-08:
There are a lot of `Brownout detector was triggered` messages. Brownout means that the supply voltage drops below the level that is required for MCU to work reliable but MCU still works to detect it. Usually it happens when the load is bigger than power supply is capable to feed. I would first try to use another power supply. Google gives a lot of hints too.

sliwma, 2021-01-23:
Hey Cougar, I tried with a stronger supply without a result. Weird, it's a Xiaomi Mi Desk Lamp and BLE worked on the stock firmware.

syssi, 2021-02-24:
The same applies to my device (Yeelight Ceiling Light YLXD76YL): https://github.com/syssi/esphome-yeelight-ceiling-light

closed 2021-07-09 by stale[bot]

sliwma, 2021-07-14:
1.19.4 - issue still exists.