esphome/issues #1732: ESP32 crashes when LEDc is used

toelke, 2020-12-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I am using docker (`:latest` and `:dev`).

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 on a ESP32dev-Clone from AZDelivery

**ESPHome version (latest production, beta, dev branch)**

`:latest` and `:dev` from dockerhub (pulled today).

**Affected component:**

https://esphome.io/components/output/ledc.html

**Description of problem:**

When I configure an output to use LEDc, the esp crashes immediately on use of that output.

My goal is to have a status LED for a light (on when the light is on and faded when the light is off). That is why I am changing the output in the `on_turn_on` of a switch.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: crash
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "XXX"
  password: "XXXXXXXX"

switch:
  - platform: gpio
    id: light
    pin: 32
    on_turn_on:
        then:
            - output.turn_on: gpio_2
    on_turn_off:
        then:
            - output.turn_off: gpio_2

output:
  - platform: ledc
    id: gpio_2
    pin: GPIO2

debug:

logger:
  level: DEBUG
```

**Logs (if applicable):**

```
[15:53:49]ets Jun  8 2016 00:22:57
[15:53:49]
[15:53:49]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[15:53:49]configsip: 0, SPIWP:0xee
[15:53:49]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[15:53:49]mode:DIO, clock div:2
[15:53:49]load:0x3fff0018,len:4
[15:53:49]load:0x3fff001c,len:1044
[15:53:49]load:0x40078000,len:8896
[15:53:49]load:0x40080400,len:5828
[15:53:49]entry 0x400806ac
[15:53:49][I][logger:166]: Log initialized
[15:53:49][I][app:029]: Running through setup()...
[15:53:49][C][switch.gpio:011]: Setting up GPIO Switch 'light'...
[15:53:49][D][switch:025]: 'light' Turning OFF.
[15:53:49][D][switch:045]: 'light': Sending state OFF
[15:53:49]/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
[15:53:49]abort() was called at PC 0x40088b65 on core 1
[15:53:49]
[15:53:49]Backtrace: 0x4008c730:0x3ffb1b70 0x4008c961:0x3ffb1b90 0x40088b65:0x3ffb1bb0 0x400d9227:0x3ffb1bf0 0x400d1ae1:0x3ffb1c10 0x400d2033:0x3ffb1c40 0x400d204f:0x3ffb1c60 0x401499a5:0x3ffb1c80 0x40149c95:0x3ffb1ca0 0x40149c73:0x3ffb1cc0 0x40149d23:0x3ffb1ce0 0x400d73c7:0x3ffb1d00 0x400d2355:0x3ffb1d20 0x400d1a39:0x3ffb1d50 0x400d2257:0x3ffb1da0 0x400d17a5:0x3ffb1dc0 0x400d17ce:0x3ffb1df0 0x40149ac1:0x3ffb1e10 0x40149b45:0x3ffb1e30 0x400d5c25:0x3ffb1e50 0x400d7325:0x3ffb1ea0 0x400d9d03:0x3ffb1fb0 0x40088e79:0x3ffb1fd0
[15:53:49]
[15:53:49]Rebooting...
```

**Additional information and things you've tried:**

I have googled the problem of an `abort()` in `queue.c` of the IDF, but it seems that that issue can have a number of reasons. Most people seem to have that problem when using an uninitialized mutex...

toelke, 2020-12-31:
I have the feeling that this is happening because the `switch` is configured and then tries to use the `output` before the `output` is configured...

toelke, 2021-01-01:
That hunch seems to have been correct.
I hacked something together where `write_state` checks if `setup` has been called and stores the state for `setup` to set it after initialization: https://github.com/toelke/esphome/tree/fix-uninitialized-ledc

JingleheimerSE, 2021-04-28:
I believe I encountered this defect today so Ill add my configuration and log for more data points. 

My goal is to control a solenoid valve (as a garage door lock) where the switch would first activate the solenoid with a high PWM, then transition to a lower PWM to reduce the heat build up in the coil while the solenoid is activated for a long period of time.

I tried to play around with pointing the switch to a dummy output (i.e. junk_output), but no matter what, when the switch's on_turn_on automation was uncommented, I would get stack traces.

```yaml
substitutions:
  deviceName: test_garage_lock
  prettyDeviceName: Test Garage Lock

esphome:
  name: $deviceName
  platform: ESP32
  board: featheresp32

wifi:
  ssid: "XXX"
  password: "XXXXXXXX"

debug:

logger:
  level: DEBUG

# Status LED
# status_led:
#   pin:
#     number: GPIO13
#     inverted: yes

output:
  - platform: ledc
    id: lock_one_output  
    pin: GPIO13
    frequency: 50 Hz
    min_power: 5.0%    # 5% at 50Hz is 1mS  (20mS cycles)
    max_power: 10.0%   # 10% at 50Hz is 2mS (20mS cycles)
  - platform: gpio
    pin: GPIO27
    id: junk_output

switch:
  - platform: output
    id: test_lock_one
    name: Test Lock One
    output: junk_output
    icon: mdi:lock
    on_turn_on:
      then:
        - output.turn_on: lock_one_output
        - delay: 2s
        - output.turn_off: lock_one_output

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      inverted: True
    name: Test Lock One Status
    device_class: lock
  - platform: gpio
    pin:
      number: GPIO15
      inverted: True
    name: Test Door Status
    device_class: garage_door
```

Log output
[... 30 lines cut ...]
←[33mWARNING Decoded 0x401602f1: esphome::output::TurnOnAction<>::play() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/esphome/components/output/automation.h:25←[0m
←[33mWARNING Decoded 0x400dee83: esphome::Action<>::play_complex() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/main.cpp:255←[0m
←[33mWARNING Decoded 0x401602a6: esphome::ActionList<>::play() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/esphome/core/automation.h:162
 (inlined by) esphome::Automation<>::trigger() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/esphome/core/automation.h:200
 (inlined by) esphome::Trigger<>::trigger() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/esphome/core/automation.h:57
 (inlined by) esphome::switch_::SwitchTurnOnTrigger::SwitchTurnOnTrigger(esphome::switch_::Switch*)::{lambda(bool)#1}::operator()(bool) const at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src\esphome/components/switch/automation.h:55
 (inlined by) std::_Function_handler<void (bool), esphome::switch_::SwitchTurnOnTrigger::SwitchTurnOnTrigger(esphome::switch_::Switch*)::{lambda(bool)#1}>::_M_invoke(std::_Any_data const&, bool&&) at c:\users\jingleheimer\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871←[0m
←[33mWARNING Decoded 0x400d990d: std::function<void (bool)>::operator()(bool) const at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src\esphome\components\switch/switch.cpp:53
 (inlined by) esphome::CallbackManager<void (bool)>::call(bool) at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/esphome/core/helpers.h:211
 (inlined by) esphome::switch_::Switch::publish_state(bool) at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src\esphome\components\switch/switch.cpp:46←[0m
←[33mWARNING Decoded 0x400d8efa: esphome::output::OutputSwitch::write_state(bool) at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src\esphome\components\output\switch/output_switch.cpp:27←[0m
←[33mWARNING Decoded 0x400d97e3: esphome::switch_::Switch::turn_on() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src\esphome\components\switch/switch.cpp:53←[0m
←[33mWARNING Decoded 0x400d8ec2: esphome::output::OutputSwitch::setup() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src\esphome\components\output\switch/output_switch.cpp:16←[0m
←[33mWARNING Decoded 0x400d8ed6: non-virtual thunk to esphome::output::OutputSwitch::setup()←[0m
←[33mWARNING Decoded 0x4015ffdd: esphome::Component::call_setup() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/esphome/core/component.cpp:111←[0m
←[33mWARNING Decoded 0x401600c5: esphome::Component::call() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/esphome/core/component.cpp:111←[0m
←[33mWARNING Decoded 0x400dd821: esphome::Application::setup() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src\esphome\core/application.cpp:38←[0m
←[33mWARNING Decoded 0x400dfd8d: setup() at S:\Athena\Projects\Electronics & Software\IoT\ESPHome-Configuration\test_garage_lock/src/main.cpp:250←[0m
←[33mWARNING Decoded 0x400eaad7: loopTask(void*) at C:\Users\Jingleheimer\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:14←[0m
←[33mWARNING Decoded 0x40088e7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)←[0m
[15:47:14]
[15:47:14]Rebooting...

Bootloop...
```

roquestrongo, 2021-04-29:
I ran in a similar problem when using Remote transmitter + FastLEDlight at the same time , my issue was closed unsolved recently. However toelke's fix worked for me. Hopefully someone can thinker out a permanent fix. Maybe it would be possible to initialize the variables in the order of how the yaml is created or include an "initialization" order option.

toelke, 2021-04-29:
I have updated my branch at https://github.com/toelke/esphome/tree/fix-uninitialized-ledc.

toelke, 2021-04-29:
https://github.com/esphome/esphome/pull/1732

closed 2021-09-10 by stale[bot]