esphome/issues #1733: color_interlock conflicts with random effect.

MarcHagen, 2020-12-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Home Assistant Core 2020.12.2 
HassIO 2020.12.7
Home Assistant OS 5.9

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32-WROOM-32 Development Board

**ESPHome version (latest production, beta, dev branch)**
ESPHome 1.15.3 (HassIO)

**Affected component:**
https://esphome.io/components/light/rgbww.html
https://esphome.io/components/light/index.html

**Description of problem:**
A fast random effect with the option `color_interlock: true` will lock the white channel to full.
Resulting in no RGB (nor white) effect. It will be full blast white

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: boekenkast
  platform: ESP32
  board: mhetesp32devkit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
  fast_connect: true

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: !secret ota_password

# Enable over-the-air updates.
ota:
  password: !secret ota_password

dallas:
  - pin: GPIO32

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

status_led:
  pin:
    number: GPIO2
    inverted: True

output:
  - platform: ledc
    pin: GPIO25
    frequency: 20000Hz
    id: LED_gpio_25
  - platform: ledc
    pin: GPIO26
    frequency: 20000Hz
    id: LED_gpio_26
  - platform: ledc
    pin: GPIO27
    frequency: 20000Hz
    id: LED_gpio_27
  - platform: ledc
    pin: GPIO14
    frequency: 20000Hz
    id: LED_gpio_14
  - platform: ledc
    pin: GPIO12
    frequency: 20000Hz
    id: LED_gpio_12

#  - platform: ledc
#    pin: GPIO16
#    frequency: 20000Hz
#    id: LED_gpio_16
#  - platform: ledc
#    pin: GPIO17
#    frequency: 20000Hz
#    id: LED_gpio_17
#  - platform: ledc
#    pin: GPIO18
#    frequency: 20000Hz
#    id: LED_gpio_18
#  - platform: ledc
#    pin: GPIO19
#    frequency: 20000Hz
#    id: LED_gpio_19
#  - platform: ledc
#    pin: GPIO21
#    frequency: 20000Hz
#    id: LED_gpio_21

light:
  - platform: rgbww
    name: "boekenkast_RGBCCT"
    red: LED_gpio_14
    green: LED_gpio_12
    blue: LED_gpio_27
    cold_white: LED_gpio_25
    warm_white: LED_gpio_26
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 3000 K
    color_interlock: true
    effects:
      - random:          
          name: "My Slow Random Effect"
          transition_length: 20s
          update_interval: 20s
      - random:
          name: "My Fast Random Effect"
          transition_length: 4s
          update_interval: 5s
      - random:
          name: "My Insane Random Effect"
          transition_length: 1s
          update_interval: 1s
      - random:          
          name: "Totally loco"
          transition_length: 300ms
          update_interval: 250ms

sensor:
  - platform: dallas
    address: 0xEF3C01D0756E5728
    name: "boekenkast_tempsensor"
```

**Logs (if applicable):**

```
INFO Successfully connected to boekenkast.local
[16:11:45][I][app:105]: ESPHome version 1.15.3 compiled on Dec 31 2020, 15:57:52
[16:11:45][C][status_led:019]: Status LED:
[16:11:45][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[16:11:45][C][wifi:443]: WiFi:
[16:11:45][C][wifi:303]:   SSID: [redacted]
[16:11:45][C][wifi:304]:   IP Address: 172.16.3.24
[16:11:45][C][wifi:306]:   BSSID: [redacted]
[16:11:45][C][wifi:307]:   Hostname: 'boekenkast'
[16:11:45][C][wifi:311]:   Signal strength: -74 dB ▂▄▆█
[16:11:45][C][wifi:315]:   Channel: 11
[16:11:45][C][wifi:316]:   Subnet: 255.255.255.192
[16:11:45][C][wifi:317]:   Gateway: 172.16.3.1
[16:11:45][C][wifi:318]:   DNS1: 172.16.3.1
[16:11:45][C][wifi:319]:   DNS2: 0.0.0.0
[16:11:45][C][ledc.output:032]: LEDC Output:
[16:11:45][C][ledc.output:033]:   Pin GPIO25 (Mode: OUTPUT)
[16:11:45][C][ledc.output:034]:   LEDC Channel: 0
[16:11:45][C][ledc.output:035]:   Frequency: 20000.0 Hz
[16:11:45][C][ledc.output:032]: LEDC Output:
[16:11:45][C][ledc.output:033]:   Pin GPIO26 (Mode: OUTPUT)
[16:11:45][C][ledc.output:034]:   LEDC Channel: 1
[16:11:45][C][ledc.output:035]:   Frequency: 20000.0 Hz
[16:11:45][C][ledc.output:032]: LEDC Output:
[16:11:45][C][ledc.output:033]:   Pin GPIO27 (Mode: OUTPUT)
[16:11:45][C][ledc.output:034]:   LEDC Channel: 2
[16:11:45][C][ledc.output:035]:   Frequency: 20000.0 Hz
[16:11:45][C][ledc.output:032]: LEDC Output:
[16:11:45][C][ledc.output:033]:   Pin GPIO14 (Mode: OUTPUT)
[16:11:45][C][ledc.output:034]:   LEDC Channel: 3
[16:11:45][C][ledc.output:035]:   Frequency: 20000.0 Hz
[16:11:45][C][ledc.output:032]: LEDC Output:
[16:11:45][C][ledc.output:033]:   Pin GPIO12 (Mode: OUTPUT)
[16:11:45][C][ledc.output:034]:   LEDC Channel: 4
[16:11:45][C][ledc.output:035]:   Frequency: 20000.0 Hz
[16:11:45][C][logger:185]: Logger:
[16:11:45][C][logger:186]:   Level: DEBUG
[16:11:45][C][logger:187]:   Log Baud Rate: 115200
[16:11:45][C][logger:188]:   Hardware UART: UART0
[16:11:45][C][light:178]: Light 'boekenkast_RGBCCT'
[16:11:45][C][light:180]:   Default Transition Length: 1.0s
[16:11:45][C][light:181]:   Gamma Correct: 2.80
[16:11:45][C][light:184]:   Min Mireds: 153.8
[16:11:45][C][light:185]:   Max Mireds: 333.3
[16:11:45][C][dallas.sensor:072]: DallasComponent:
[16:11:45][C][dallas.sensor:073]:   Pin: GPIO32 (Mode: INPUT)
[16:11:45][C][dallas.sensor:074]:   Update Interval: 60.0s
[16:11:45][W][dallas.sensor:077]:   Found no sensors!
[16:11:45][C][dallas.sensor:087]:   Device 'boekenkast_tempsensor'
[16:11:45][C][dallas.sensor:087]:     Unit of Measurement: '°C'
[16:11:45][C][dallas.sensor:087]:     Accuracy Decimals: 1
[16:11:45][C][dallas.sensor:087]:     Icon: 'mdi:thermometer'
[16:11:45][C][dallas.sensor:095]:     Address: 0xEF3C01D0756E5728
[16:11:45][C][dallas.sensor:096]:     Resolution: 12
[16:11:45][C][homeassistant.time:010]: Home Assistant Time:
[16:11:45][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
[16:11:45][C][ota:029]: Over-The-Air Updates:
[16:11:45][C][ota:030]:   Address: boekenkast.local:3232
[16:11:45][C][ota:032]:   Using Password.
[16:11:45][C][api:095]: API Server:
[16:11:45][C][api:096]:   Address: boekenkast.local:6053
[16:11:46][D][light:265]: 'boekenkast_RGBCCT' Setting:
[16:11:46][D][light:282]:   Color Temperature: 0.5 mireds
[16:11:46][D][light:287]:   Red=95%, Green=82%, Blue=88%
[16:11:46][D][light:290]:   White Value: 100%
[16:11:46][D][light:304]:   Transition Length: 4.0s
[16:11:51][D][light:265]: 'boekenkast_RGBCCT' Setting:
[16:11:51][D][light:282]:   Color Temperature: 0.4 mireds
[16:11:51][D][light:287]:   Red=92%, Green=47%, Blue=49%
[16:11:51][D][light:290]:   White Value: 100%
[16:11:51][D][light:304]:   Transition Length: 4.0s
[16:11:56][D][light:265]: 'boekenkast_RGBCCT' Setting:
[16:11:56][D][light:282]:   Color Temperature: 1.0 mireds
[16:11:56][D][light:287]:   Red=100%, Green=17%, Blue=27%
[16:11:56][D][light:290]:   White Value: 100%
[16:11:56][D][light:304]:   Transition Length: 4.0s
[16:11:58][W][dallas.sensor:146]: 'boekenkast_tempsensor' - Scratch pad checksum invalid!
[16:11:58][D][sensor:092]: 'boekenkast_tempsensor': Sending state nan °C with 1 decimals of accuracy
[16:12:01][D][light:265]: 'boekenkast_RGBCCT' Setting:
[16:12:01][D][light:282]:   Color Temperature: 0.4 mireds
[16:12:01][D][light:287]:   Red=98%, Green=19%, Blue=64%
[16:12:01][D][light:290]:   White Value: 100%
[16:12:01][D][light:304]:   Transition Length: 4.0s
[16:12:06][D][light:265]: 'boekenkast_RGBCCT' Setting:
[16:12:06][D][light:282]:   Color Temperature: 0.5 mireds
[16:12:06][D][light:287]:   Red=86%, Green=36%, Blue=25%
[16:12:06][D][light:290]:   White Value: 100%
[16:12:06][D][light:304]:   Transition Length: 4.0s
[16:12:11][D][light:265]: 'boekenkast_RGBCCT' Setting:
[16:12:11][D][light:282]:   Color Temperature: 0.4 mireds
[16:12:11][D][light:287]:   Red=38%, Green=98%, Blue=60%
[16:12:11][D][light:290]:   White Value: 100%
[16:12:11][D][light:304]:   Transition Length: 4.0s
[16:12:16][D][light:265]: 'boekenkast_RGBCCT' Setting:
[16:12:16][D][light:282]:   Color Temperature: 0.8 mireds
[16:12:16][D][light:287]:   Red=54%, Green=99%, Blue=47%
[16:12:16][D][light:290]:   White Value: 100%
[16:12:16][D][light:304]:   Transition Length: 4.0s
[16:12:21][D][light:265]: 'boekenkast_RGBCCT' Setting:
[16:12:21][D][light:282]:   Color Temperature: 0.4 mireds
[16:12:21][D][light:287]:   Red=30%, Green=25%, Blue=15%
[16:12:21][D][light:290]:   White Value: 100%
[16:12:21][D][light:304]:   Transition Length: 4.0s
```

**Additional information and things you've tried:**

Jab2870, 2021-03-03:
I also have this issue. Did you find a solution @MarcHagen ?

MarcHagen, 2021-03-03:
Unfortunately not. A option is to separate the CCT and RGB... or interlock off

Jab2870, 2021-07-02:
I don't believe this has been fixed. This comment is just to prevent the issue
going stale