esphome/issues #1735: Unable to find action with the name 'http_request.post'.

littleyoda, 2021-01-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Testing with hass.io and esphome (pip/cli version/ESPHome v1.15.3)

**ESP (ESP32/ESP8266, Board/Sonoff):**
n/a, because problem with the config file. (Tested with ESP32 and ESP8266)

**ESPHome version (latest production, beta, dev branch)**
ESPHome v1.15.3

**Affected component:**
https://esphome.io/components/http_request.html

**Description of problem:**
trying to use http_request.post in a on_press-Action, results in this error message (cli-version):
```
Failed config
[...]
  on_press:  [source simple.yaml:12]
    - [source simple.yaml:12]
      
      Unable to find action with the name 'http_request.post'.
 http_request.post: http://XXXXX/mypostrequest [source simple.yaml:12]
```

In hass.io:
```
Unable to find action with the name 'http_request.post'
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ha_az_3dcontroll
  platform: ESP32
  board: mhetesp32minikit

binary_sensor:
  - platform: gpio
    pin: GPIO18
    name: "PIN 18"
    on_press:
      - http_request.post: http://XXXXX/mypostrequest

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Also tested with http_request.post and http_request.get

arduino_version: latest

glmnet, 2021-01-05:
That's weird. It should be checked on every build automatically. 
Did you try a clean re install?

littleyoda, 2021-01-05:
```
pip3 uninstall esphome
pip3 install --no-cache-dir esphome
esphome sample.yaml run
```

same result:

```
Failed config

binary_sensor.gpio: [source sample.yaml:8]
  platform: gpio
  pin: GPIO18
  name: PIN 18
  on_press:  [source sample.yaml:12]
    - [source sample.yaml:12]
      
      Unable to find action with the name 'http_request.post'.
      http_request.post: http://XXXXX/mypostrequest [source sample.yaml:12]

```

glmnet, 2021-01-05:
did you setup the component? 

e.g.

````yaml
# Example configuration entry
http_request:
  useragent: esphome/device
  timeout: 10s
````

littleyoda, 2021-01-05:
great. It works now.
 
Unfortunately, I have overlooked the hint about setting up the component in the instructions.

closed 2021-01-05 by littleyoda