esphome/issues #1737: ESP-12 keeps rebooting when not logging - but works fine while logging

Jojo-A, 2021-01-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 0.118.4 (not relevant to this issue)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 ESP-12 on NodeMCU Dev Board

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

something in the core?!

**Description of problem:**
Very basic setup: nothing but a pulse counter and a filter.
When I upload the config OTA or via serial, everything works fine, which means the the sensor for pulse counter gets filtered correctly and reaches the expected value after the expected time BUT ONLY while I am connected to the board for logging (keeping the terminal window open).
NOW: when I close the terminal from where I uploaded the config, the board seems to reboot after some time and keeps rebooting after ~15 minutes or so. This can be seen in the filtered sensor value, which gets reset over and over again.
I strongly guess, that this is related to the problem described here: https://github.com/esphome/esphome-core/issues/394 which has not been solved as how I understand it...

[image]
The graphic shows:
- the boards seems to reboot until ~13:05
- at ~13:05 I re-uploaded the config OTA without applying any changes
- terminal window kept open until ~ 13:40
- at 13:40 I closed the terminal window from where I uploaded the config
- at ~14:00 the board reboots and keeps hanging in this circle... 
This is totally reproducible.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: geigercounter
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "WiFi SSID"
  password: "verysecretwifipassword"

captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

mqtt:
  discovery: true
  discovery_prefix: homeassistant
  broker: xxx.xxx.xxx.xxx
  username: username
  password: verysecretpassword

sensor:
  - platform: pulse_counter
    pin: D2
    unit_of_measurement: 'µSv/h'
    name: 'radiation'
    update_interval: 60s
    filters:
      - multiply: 0.00812
      - offset: -0.087696
      - exponential_moving_average:
          alpha: 0.0667
          send_every: 1
    state_topic: "sensor/geigercounter"
    retain: True

```

**Logs (if applicable):**

```
I can not provide any logs of the error event, because it works while I am staying connected to it. It only reboots when I disconnect... :(

```

**Additional information and things you've tried:**
- different logging levels -> no change
- different filter configs -> no change
- different boards (!) -> no change
- with or w/o webserver -> no change
- with or w/o MQTT discovery -> no change

Alex9779, 2021-01-06:
My first thought is the API watchdog timer. I don't get from your description if you use the API to HA actually or MQTT.
If the API is active but no HA server connects then there is a watchdog timer to reboot.
Strange that this does not happen if you connect to the serial...
So if you use just MQTT and did not use the esphome integration for the API in HA then try without "api:"...

Jojo-A, 2021-01-06:
Thanks a ton for taking care of my issue.
True, basically I only use MQTT, so I could indeed try what happens if I deactivate the API. I will try tomorrow and let you know.
But yes, it is strange that this does not happen as long as I am connected for logging.

Do you have an idea about the issue I referred to in my opening post? It seems to me as if it is also somehow affected (but not caused) by the logging. My "idea" why this and my issue might be related to each other is, that the logging puts some additional load to the CPU which keeps it busy before a watchdog might kick in. Of course I have no detailed knowledge about the software architecture of the ESP firmware (+ esphome), but at least this is something that the other issue and my have in common...

plastus1, 2021-01-07:
i hew that same issue i use api and dont find any solution any ideas whats cod by cousin that

that my sketch

esphome:
  name: temp
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "xxxxxxxxxxxx"
  password: "xxxxxxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temp Fallback Hotspot"
    password: "rcwro9vSPceL"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:

sensor:
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
    humidity:
      name: "BME280 Humidity"
    address: 0x76
    update_interval: 60s

just my on when i disconnect serial port its loose connection straightaway
and sorry for my English

Jojo-A, 2021-01-07:
@plastus1 
Thanks for sharing.

About your code:
please use the correct formatting to improve readability -> preview before posting.

About English:
maybe [google translate](https://translate.google.com/) or [deepl](https://www.deepl.com/translator) might be an option for you.

plastus1, 2021-01-07:
how to do that i try to find it just now luck
i new in that 
thx for you answer

plastus1, 2021-01-07:
i got it sorted out thx for you help anyway

Jojo-A, 2021-01-08:
Hi and thanks again for caring, @Alex9779 .

For now, my device seems to work fine without regular reboots. Indeed, the API timeout must have caused it, because I do not have the corresponding integration in my Home Assistant server.

So /maybe/ I can close this issue. But on the other hand it is still strange, that this does not happen while keeping the logging terminal open, isn't it?

Greetings

closed 2021-05-23 by stale[bot]