esphome/issues #1739: ESP32 Disconnect/Reconnect to HA and Webserver with text_sensor providing IP Address enabled.

jmoery, 2021-01-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10 Pip command line 
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32. Boards: esp32doit-devkit-v1 & lolin-d32 (Wemos Lolin32 with OLED built-in)

**ESPHome version (latest production, beta, dev branch)**

v1.15.3 

**Affected component:**

text_sensor, platform: wifi_info

**Description of problem:**
Using the wifi_info via the text-sensor to return IP address of esp causes the ESPs to constantly disconnect and reconnect from Home assistant as well as the internal webserver.  This only seems to be an issue on ESP32 as I have used the same configs on ESP8266 and it does not have the same issue.  Removing the text-sensor for id_address fixes the issue.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: garage_esp
  platform: ESP32
  board: lolin32

wifi:
  ssid: "***"
  password: "***"

# Enable logging
logger:

web_server:
  port: 80

# Enable Home Assistant API
api:
  password: "***"

ota:
  password: "***"

switch:
  - platform: gpio
    pin: 12    
    id: garage_relay
  - platform: template
    name: "Garage Door"
    turn_on_action:
    - switch.turn_on: garage_relay
    - delay: 500ms
    - switch.turn_off: garage_relay
  - platform: restart
    name: "Garage Air Quality Restart"

i2c:
  sda: 5
  scl: 4
  frequency: 400khz

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"    
    address: 0x3C
    lambda: |-
      it.printf(0 ,10, id(arial_2),"AQI: %.0f", id(aqi1).state);         
      it.printf(0, 50, id(arial_1),"%s", id(esp_ip).state.c_str()); 

font:
  - file: "fonts/arial.ttf"
    id: arial_1
    size: 10
    glyphs:
      ['&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z', 'å', 'ä', 'ö', '/','³','μ']
  - file: "fonts/arial.ttf"
    id: arial_2
    size: 20

sensor:
  - platform: hm3301
    pm_1_0:
      name: "PM1.0"
      id: pm1_0
    pm_2_5:
      name: "PM2.5"
      id: pm2_5
    pm_10_0:
      name: "PM10.0"
      id: pm10
    aqi:
      name: "AQI"
      calculation_type: "AQI"
      id: aqi1

text_sensor:
  - platform: wifi_info
    ip_address:
      name: Garage ESP IP
      id: esp_ip
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I also tried the fix to previous issues with the HA API disconnecting and reconnecting using the esphome: on_loop delay(5) method.  This didn't fix the issue.  I also tried adding a static IP which also did not fix the issue.  The switch(relay) is also very slow to respond with the text_sensor enabled.  Disabling the text_sensor it becomes almost immediate again when triggered via HA or webserver.  None of these issues are present using the same config on Nodemcu esp8266.

jmoery, 2021-01-05:
Update:  Just added a binary sensor to the previously working yaml that doesn't  include the text_sensor and the issue started to occur again.  It seems it may not be related to the text_sensor itself but the number of inputs.  Here is the new yaml that now causes the issue again even with the text_sensor removed.

`esphome:
  name: garage_esp
  platform: ESP32
  board: lolin32

wifi:
  ssid: "***"
  password: "***"

# Enable logging
logger:

web_server:
  port: 80

# Enable Home Assistant API
api:
  password: "**"

ota:
  password: "***"

switch:
  - platform: gpio
    pin: 12    
    id: garage_relay
  - platform: template
    name: "Garage Door"
    turn_on_action:
    - switch.turn_on: garage_relay
    - delay: 500ms
    - switch.turn_off: garage_relay
  - platform: restart
    name: "Garage Air Quality Restart"

binary_sensor:
  - platform: gpio
    pin: 
      number: 13
      mode: INPUT_PULLUP
      inverted: True
    name: "Garage Door State"
    id: garage_state
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms

i2c:
  sda: 5
  scl: 4
  frequency: 400khz

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"   
    update_interval: 3s 
    address: 0x3C
    lambda: |-
      it.printf(0 ,10, id(arial_2),"AQI: %.0f", id(aqi1).state);  
      it.printf(0, 30, id(arial_1), "PM2.5: %.1f μg/m³", id(pm2_5).state);      
     

font:
  - file: "fonts/arial.ttf"
    id: arial_1
    size: 12
    glyphs:
      ['&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z', 'å', 'ä', 'ö', '/','³','μ']
  - file: "fonts/arial.ttf"
    id: arial_2
    size: 24

sensor:
  - platform: hm3301
    pm_1_0:
      name: "PM1.0"
      id: pm1_0
    pm_2_5:
      name: "PM2.5"
      id: pm2_5
    pm_10_0:
      name: "PM10.0"
      id: pm10
    aqi:
      name: "AQI"
      calculation_type: "AQI"
      id: aqi1
`

hawkeye217, 2021-01-11:
Like you, I also have random disconnects from the API on my ESP32 with a fair amount of sensors.  My ESP8266 nodes don't have that issue, it seems.  I've tried ``on_loop`` delay and ``power_save_mode: none`` to no effect.  Sometimes I can go hours without a disconnect, other times it disconnects/reconnects every minute or two.

jmoery, 2021-01-11:
I'm so glad it's not just me!  I spent a good day trying to figure out what it might be to no avail.  It seems it comes down to having a certain amount of sensors.  I ended up reverting to my 8266 for the project I was working on, but the 32 would be such a better board for it.  Hopefully this can get sorted soon.

realroywalker, 2021-02-07:
Would like to add a me too for this issue.. I've been troubleshooting an issue with esp32 (wrover-b) for the best part of today. 
Random disconnects from HA sometimes a few minutes others can go 10 - 20 mins without a disconnect.
My uptime sensor shows the device isn't restarting, just disconnecting.
Tried similar things to the posts above (disable WiFi powersave etc. ) To no avail.

hawkeye217, 2021-02-07:
I actually solved my issue by getting a new router and moving the router closer to the device. In my case it turns out the problem was was just a weak wifi signal.

realroywalker, 2021-02-07:
Thanks for the update, sounds like mine could be a different issue then. I've tried flashing with tasmota firmware this afternoon and so far rock solid (max 20 mins before disconnect with esphome).
Unfortunately tasmota won't do everything I need though, so need to try and work out what causes it.

jmoery, 2021-02-08:
I don't think mine is a router issue either for a couple of reasons.  The biggest one is I have an AP about 2 feet from my test setup.  If it was a router issue as well it would be the same problem with less sensors.  It would also prevalent with other esp's but considering I can take the same sketch and flash it to a 8266 and it works fine, it leads me to believe it's something in the code that is causing the hangups.

realroywalker, 2021-02-09:
So after some trial and error I've managed to get the esp32 to stay up for 10 hours and counting.. I disabled native API and used MQTT instead, so far that seems to have stopped the disconnect. So may be worth trying!

jmoery, 2021-02-09:
I could see that removing the API could fix the issue as it seems like there is something going on in that part of the code, but kind of defeats the point of having the API option as it should be faster and more consistent than MQTT method.

realroywalker, 2021-02-09:
Indeed, entirely sure why API seems to cause the issue as I have about 8 ESP 8266 with the same option and it works fine.
20 hrs and counting for the ESP32 and MQTT now, so definitely seems to have fixed it for me.

randybb, 2021-02-09:
What is your [signal strength](https://esphome.io/components/sensor/wifi_signal.html)?

jmoery, 2021-02-09:
I'm getting -66 dB in the test location I've been using.  It's pretty much
right next to an access point.

jmoery, 2021-02-09:
Sorry I was looking at the wrong value.  That was the strength to another
AP.  The one it is connected to is -38 dB

randybb, 2021-02-09:
But from the client site - it is different from the signal strength visible on AP.

jmoery, 2021-02-09:
Yes that is what the ESP is reporting.  I am using a test esp and currently
right now it is -41 dB.  All the sensors I had before are included but not
attached to the board.

randybb, 2021-02-09:
And you are still having issues? Because -41 dB is really good.

I am currently testing it on my esp32 dev board with a bunch of i2c sensors, two OLED displays, classic 4x20 lcd display,... and added this sensor that is reporting its IP address without any connection issues through native API (I use only this everywhere without any problems). I even enabled web server and it is still stable. The only difference is in version - I have the latest DEV release.

jmoery, 2021-02-10:
I think I was running the dev release but it might be old right now.  I
haven't really tested it for a few weeks now as I just switched to running
on the 8266, but would like to having everything running correctly on a 32
also.  i'll have to test it.  Thanks for the info.

mmakaay, 2021-03-18:
Random disconnects that vary with the workload on the system, sound like they might be caused by an issue that I tracked down over the last days. Check out [this comment to another bug report about such issues](https://github.com/esphome/issues/issues/1729#issuecomment-802019178).

closed 2021-07-31 by stale[bot]