esphome/issues #1742: AM2320 not found on cold boot

soostdijck, 2021-01-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

"Home assistant OS" running in a LXC container" (which in turn is running esphome in a docker container)

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU v2 (ESP8266)

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

AM2320

**Description of problem:**
When connecting a AM2320 (temperature and humidity sensor) via I2C the sensor is not found on cold boot. The problem can be reproduced like this:
* Connect a AM2320 to a NodeMCU (with esphome)
* (cold)Boot the NodeMCU and the AM2320 isn't found with the following error: `[E][am2320:065]: Communication with AM2320 failed!`. The sensor won't be found/working at all.
* Hit the reset button on the NodeMCU and the sensor starts working.

Side note: the I2C bus reports no devices found in either situation.
Side note2: search on discord yielded several people with the same problem.

**Problem-relevant YAML-configuration entries:**
The workaround:
Add a `setup_priority: -100` to the sensor yaml config, like this and it starts working on a cold boot as well:
```yaml
sensor:
  - platform: am2320
    setup_priority: -100
    address: 0x5C
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
```

**Logs (if applicable):**

Any of the above solutions keep on saying this, which doesn't seem to cause harm:
```
[W][i2c:070]: Received NACK on transmit of address 0x5C
```

**Additional information and things you've tried:**
n/a

sergeolkhovik, 2021-01-06:
I've got another AM2320 that doesn't appear on cold boot, but sometimes appear on reboot.
I think we should increase the delay (see [link](https://github.com/esphome/issues/issues/192)), I was going to do some tests with that value but didn't work on that yet.

sergeolkhovik, 2021-02-17:
> I've got another AM2320 that doesn't appear on cold boot, but sometimes appear on reboot.
> I think we should increase the delay (see [link](https://github.com/esphome/issues/issues/192)), I was going to do some tests with that value but didn't work on that yet.

I found that SI7021 has built-in pull-up resistors (Chineese GY-21, 10k), and AM2320 doesn't, after I connected resistors my AM2320 started working.

closed 2021-07-02 by stale[bot]