esphome/issues #1743: Mqtt clean fail

raidnet-ms, 2021-01-06:
Hello,

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I'm using Hassio on Raspberry Pi4 8Gb (64bit version)
ESPhome configured with SSL with no error
Mqtt is configured with SSL and ACL with no error from MQTT supervisor log (other esp devices are working properly and so Zigbee devices used with Conbee).

**ESP (ESP32/ESP8266, Board/Sonoff):**
The problem exist with every board i tested: esp01, nodemcu, d1mini

ESPHome version latest production

**Affected component:**

**Description of problem:**
While trying to activa an "MQTT CLEAN" from the ESP dashboard i get this error, with every yaml config, even with a basic file (just wifi, api ad ota config):

(CLEAN MQTT LOG FROM ESP dashboard):

`INFO Reading configuration /config/esphome/test.yaml...
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 11, in <module>
    load_entry_point('esphome', 'console_scripts', 'esphome')()
  File "/opt/esphome/esphome/__main__.py", line 554, in main
    return run_esphome(sys.argv)
  File "/opt/esphome/esphome/__main__.py", line 541, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/opt/esphome/esphome/__main__.py", line 328, in command_clean_mqtt
    return clean_mqtt(config, args)
  File "/opt/esphome/esphome/__main__.py", line 219, in clean_mqtt
    return mqtt.clear_topic(config, args.topic, args.username, args.password, args.client_id)
  File "/opt/esphome/esphome/mqtt.py", line 106, in clear_topic
    discovery_prefix = config[CONF_MQTT].get(CONF_DISCOVERY_PREFIX, 'homeassistant')
KeyError: 'mqtt'`

**Problem-relevant YAML-configuration entries:**
```yaml
###############################
esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "$devicename"
    password: !secret ap_password
captive_portal:
# Enable logging
logger:
# Enable Home Assistant API
api:
  password: !secret api_password
ota:
  password: !secret ota_password
#################################################################

status_led:
  pin:
    number: 1
    inverted: false
  id: led

switch:
    platform: gpio
    name: "heater"
[... 101 lines cut ...]
`[13:57:28] INFO: Setup mosquitto configuration
[13:57:28] INFO: Found local users inside config
[13:57:28] INFO: Initialize Hass.io Add-on services
[13:57:28] INFO: Initialize Home Assistant discovery
[13:57:28] INFO: Start Mosquitto daemon
1609937848: mosquitto version 1.6.3 starting
1609937848: Config loaded from /etc/mosquitto.conf.
1609937848: Loading plugin: /usr/share/mosquitto/auth-plug.so
1609937848:  ├── Username/password checking enabled.
1609937848:  ├── TLS-PSK checking enabled.
1609937848:  └── Extended authentication not enabled.
1609937848: |-- *** auth-plug: startup
1609937848: Opening ipv4 listen socket on port 1883.
1609937848: Opening ipv6 listen socket on port 1883.
1609937848: Opening websockets listen socket on port 1884.
1609937848: Opening ipv4 listen socket on port 8883.
1609937848: Opening ipv6 listen socket on port 8883.
1609937848: Opening websockets listen socket on port 8884.
1609937849: Warning: Mosquitto should not be run as root/administrator.
1609937892: New connection from 172.30.32.1 on port 1883.
[INFO] found mqtt on local database
1609937892: New client connected from 172.30.32.1 as 3V1QifV6c05hAN80N7IX4j (p2, c1, k60, u'mqtt').
1609939649: Saving in-memory database to /data/mosquitto.db.`

thank you

yahel2410, 2021-03-31:
Hi @raidnet-ms, did you find a solution for this? I'm facing the same issue

raidnet-ms, 2021-03-31:
I was never able to make mqtt clean working, furthermore mqtt 5.1.1 has stopped working with esphome.
I partially solved the problem of zombie things with persistent will on mqtt integration settings.
Let me know if works for you.

raidnet-ms, 2021-07-30:
Well this problem is persistent