esphome/issues #1744: dht22 wrongly reported negative temperatures

nettle-m, 2021-01-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
sensor is working fine when temp is above 0C
but after temp drops  below zero it starts to output wrong values. for example -3276.80005C but those values are correlated with temperature. I think it is some kind of conversion/overflow bug.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: puda_mala_teplomer
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: "***"
  password: "***"

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.2.41
    gateway: 192.168.2.254
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Garaz Teplomer Fallback Hotspot"
    password: "***"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "***"

ota:
  password: "***"
  
  
sensor:
  - platform: dht
    pin: GPIO05
    temperature:
      name: "mapa_puda Temperature"
    humidity:
      name: "mala_puda Humidity"
    update_interval: 20s
    model: DHT22

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/puda_mala_teplomer.yaml...
INFO Starting log output from 192.168.2.41 using esphome API
INFO Connecting to 192.168.2.41:6053 (192.168.2.41)
INFO Successfully connected to 192.168.2.41
[17:47:40][I][app:105]: ESPHome version 1.15.3 compiled on Jan  7 2021, 17:09:49
[17:47:40][C][wifi:443]: WiFi:
[17:47:40][C][wifi:303]:   SSID: 

[... 107 lines cut ...]
[17:55:46][D][sensor:092]: 'mapa_puda Temperature': Sending state -3276.40015 °C with 1 decimals of accuracy
[17:55:46][D][sensor:092]: 'mala_puda Humidity': Sending state 80.00000 % with 0 decimals of accuracy
[17:56:06][D][dht:048]: Got Temperature=-3276.3°C Humidity=80.4%
[17:56:06][D][sensor:092]: 'mapa_puda Temperature': Sending state -3276.30005 °C with 1 decimals of accuracy
[17:56:06][D][sensor:092]: 'mala_puda Humidity': Sending state 80.40000 % with 0 decimals of accuracy
[17:56:26][D][dht:048]: Got Temperature=-3276.2°C Humidity=80.7%
[17:56:26][D][sensor:092]: 'mapa_puda Temperature': Sending state -3276.19995 °C with 1 decimals of accuracy
[17:56:26][D][sensor:092]: 'mala_puda Humidity': Sending state 80.70000 % with 0 decimals of accuracy
[17:56:46][D][dht:048]: Got Temperature=-3276.2°C Humidity=81.2%
[17:56:46][D][sensor:092]: 'mapa_puda Temperature': Sending state -3276.19995 °C with 1 decimals of accuracy
[17:56:46][D][sensor:092]: 'mala_puda Humidity': Sending state 81.20000 % with 0 decimals of accuracy
[17:57:06][D][dht:048]: Got Temperature=-3276.1°C Humidity=82.0%
[17:57:06][D][sensor:092]: 'mapa_puda Temperature': Sending state -3276.10010 °C with 1 decimals of accuracy
[17:57:06][D][sensor:092]: 'mala_puda Humidity': Sending state 82.00000 % with 0 decimals of accuracy
[17:57:26][D][dht:048]: Got Temperature=-3276.0°C Humidity=82.3%
[17:57:26][D][sensor:092]: 'mapa_puda Temperature': Sending state -3276.00000 °C with 1 decimals of accuracy
[17:57:26][D][sensor:092]: 'mala_puda Humidity': Sending state 82.30000 % with 0 decimals of accuracy

```

**Additional information and things you've tried:**
I tried to change sensor type to all possible values:
AUTO_DETECT, DHT11, DHT22, DHT22_TYPE2, AM2302, RHT03, SI7021

none of them solved the issue

nettle-m, 2021-01-07:
as workaround i used this filter:

```
filters:
        - lambda: !lambda if ( x < -1000.0 ) { return (-1) * x - 3277.0; } else { return x; }
```

now temperatures in home-assistant show as they should

nettle-m, 2021-02-07:
just update:
problem is still present in ESPHome version 1.16.0

SenexCrenshaw, 2021-02-20:
Can you try the latest dev branch?

nettle-m, 2021-02-23:
Hello, i dont want to mess with my current Home Assistant i will try to setup new instance with ESPHome dev version, but it will take some time

nettle-m, 2021-02-24:
i was unsuccessful in dev esphome deployment. Could you guide me how can i upgrade ESPHome HO addon to dev version?

SenexCrenshaw, 2021-02-25:
Have you tried adding a 4.7k pullup like in the docs?
https://esphome.io/components/sensor/dht.html

nettle-m, 2021-02-26:
yes, i have dht22 connected with pull up resistor from the start

agners, 2021-06-12:
This very much matches what is described in this comment: https://github.com/esphome/esphome/pull/926#issuecomment-602110509

That got fixed with the DHT22 Type 2 introduction. Are you sure `DHT22_TYPE2` doesn't fix your problem?

gb53smith, 2021-06-19:
You may want try this custom sensor from my [post](https://github.com/esphome/issues/issues/2102#issuecomment-860109021).
It uses a newer and perhaps better DHT library.
Your issue appears to be the same as found in a much older DHT library.
Esphome folks may have missed a bug fix and not copied it into their DHT component.

gb53smith, 2021-06-25:
@nettle-m You may want to test your DHT using the new external component integration
```external_components:
  - source: github://gb53smith/esphome@devsweep
    components: [dht]
```
 It is based on github://esphome/esphome@dev
with the added test of sweeping the start delay between 800 and 20000 us.
(To prove a theory that the start delay was never the true cause bad readings.)
The model: must be set to AM2302 for the test to be activated.

When github://esphome/esphome@dev get released use model: DHT22
Never use model: DHT22_TYPE2 unless your DHT datasheet indicates 2's complement representation of negative temperatures

closed 2021-11-26 by github-actions[bot]