esphome/issues #1749: Dallas - Having a DS18S20 sets status to error

MichaelAhlberg, 2021-01-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome integration in Home Assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 nodemcuv2

**ESPHome version (latest production, beta, dev branch)**

v1.15.3

**Affected component:**

https://esphome.io/components/sensor/dallas.html

**Description of problem:**
When connecting a DS18S20 to the bus the status_led feature shows an error. The sensor works but the status indication becomes useless because of this component asserting an error.

I have tracked it down to the following lines of code in Dallas_component.cpp (around line 200)

```
  if (this->get_address8()[0] == DALLAS_MODEL_DS18S20) {
    // DS18S20 doesn't support resolution.
    ESP_LOGW(TAG, "DS18S20 doesn't support setting resolution.");
    return false;
  }
```

And I believe that returning true instead of false would solve the problem. I do not have any suitable development environment for esphome so I cannot test this myself.

**Problem-relevant YAML-configuration entries:**
```yaml
dallas:
  - pin: GPIO4

# Individual sensors
sensor:
  - platform: dallas
    address: 0x34000801042F7E10
    name: "Temperature"

status_led:
  pin:
    number: D0
    inverted: True

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

rradar, 2021-01-08:
@MichaelAhlberg are you aware that the pins for dallas (**GPIO4**) and status_led (**D0**) is actually the **same pin**?

MichaelAhlberg, 2021-01-08:
@rradar On my nodemcu board D0 is GPIO16 (nodemcuv2 with ESP8266) it has the red onboard led connected to it And I've verified that it is not the same pin after your comment.

rradar, 2021-01-09:
@MichaelAhlberg you are right, I mixed up the pins... 

D0 = GPIO16
D2 = GPIO4
D4 = GPIO2 -> Typically the onboard status led on esp12x module (often use on a nodemcu or d1mini)

closed 2021-05-28 by stale[bot]