esphome/issues #1750: PID Climate Controller hvac_mode issues in HASS

fabianschaefer, 2021-01-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome integration in Home Assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (Lolin D32)

**ESPHome version (latest production, beta, dev branch)**

v1.15.3

**Affected component:**

https://esphome.io/components/climate/pid.html

**Description of problem:**
When setting up a PID Controller with only heat_output configured (no cool_output), it shows up in Home Assistant with the following hvac modes:
- off
- heat: Heater is permanently turned on, even if temperature is far over target
- heat_cool: Does actually regulate heater based on the temperature as expected

From my understanding there are 2 issues here:
1. If only a heat but no cool output is defined, the hvac mode heat_cool should not even be visible in Home assistant
2. If hvac mode is set to heat in Home assistant, the heater should be regulated based on the temperature instead of always on

**Problem-relevant YAML-configuration entries:**
```yaml
climate:
  - platform: pid
    name: sauna_temperature
    visual:
      min_temperature: 40 °C
      max_temperature: 100 °C
      temperature_step: 5 °C
    sensor: temperature_room
    default_target_temperature: 80°C
    heat_output: heater
    control_parameters:
      kp: 0.2
```

**Logs (if applicable):**

N/A

**Additional information and things you've tried:**
I can remove unneeded hvac modes in the home assistant frontend by editing the customize.yaml file:
```yaml
climate.sauna_temperature:
  hvac_modes:
  - heat_cool
  - 'off'
```

But then there is still the issue, that I have to select Heat/Cool (instead of Heat) in the frontend which doesn't make sense because there is no cooling device. So this is more of a cosmetic issue.

fabianschaefer, 2021-01-09:
Just for comparison, I added a generic_thermostat component in Home Assistant:
```yaml
climate:
  - platform: generic_thermostat
    name: Sauna_Test
    heater: switch.sauna_main_switch
    target_sensor: sensor.temperature_2
```
So the 2 Climate Cards look as follows:
[image]

fabianschaefer, 2021-05-10:
Any update on this? The issue still exists.

AndreFvdMerwe, 2021-07-08:
I have the same issue - when HEAT is selected in "climate PID", the climate output is switched on and runs away, which is different from how Climate normally works.

carstenschroeder, 2021-07-09:
Same here but with COOL only mode. Cools always regardless of current temp.

closed 2021-07-10 by OttoWinter