esphome/issues #1752: IR remote climate Toshiba some AC options not exposed

taifunk, 2021-01-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

homeassistant ESPHome add-on

**ESP (ESP32/ESP8266, Board/Sonoff):**

YTF IR bridge

**ESPHome version (latest production, beta, dev branch)**

ESPHome DEV 1.1.17

**Affected component:**

IR Remote Climate

**Description of problem:**
Toshiba platform doesn't expose fan speed and swing options.
Otherwise the IR commands work and the AC reacts correctly.

In the library the needed options are defined: https://github.com/esphome/esphome/blob/dev/esphome/components/toshiba/toshiba.cpp

**Problem-relevant YAML-configuration entries:**
```yaml
/.../
# Use the blue LED as a status light.
#   Blink slowly = Temporary issue (WiFi/MQTT disconnected, sensor unavailable, ...)
#   Blink quickly = Error encountered
status_led:
  pin: GPIO4

# Configure the IR receiver. Handy to pickup confirmation messages
# from your AC (or capture commands from the actual remote)
remote_receiver:
  id: rcvr
  pin: 
    number: GPIO5
    inverted: True
  dump: all

# Configure the IR LED: this one sends commands to other devices
remote_transmitter:
  pin: GPIO14
  carrier_duty_percent: 50%
  
# Configure the AC unit you have.
# https://esphome.io/components/climate/ir_climate.html?highlight=climate
#
climate:
  - platform: toshiba
    name: "Elutoa soojuspump"
    receiver_id: rcvr
    sensor: temp
    visual:
      min_temperature: 16 °C
      max_temperature: 32°C
      temperature_step: 1 °C

      
sensor:
  - platform: homeassistant
    name: "Current temperature"
    entity_id: sensor.elutoa_anduri_temperatuur
    id: temp
    internal: true

```

**Logs (if applicable):**

```
[17:04:31][C][climate_ir:113]: IR Climate 'Elutoa soojuspump'
[17:04:31][C][climate_ir:114]:   Min. Temperature: 17.0°C
[17:04:31][C][climate_ir:115]:   Max. Temperature: 30.0°C
[17:04:31][C][climate_ir:116]:   Supports HEAT: YES
[17:04:31][C][climate_ir:117]:   Supports COOL: YES
[17:04:31][C][remote_transmitter:018]: Remote Transmitter...
[17:04:31][C][remote_transmitter:019]:   Carrier Duty: 50%
[17:04:31][C][remote_transmitter:020]:   Pin: GPIO14 (Mode: OUTPUT)
[17:04:31][C][remote_receiver.esp8266:059]: Remote Receiver:
[17:04:31][C][remote_receiver.esp8266:060]:   Pin: GPIO5 (Mode: INPUT, INVERTED)
[17:04:31][C][remote_receiver.esp8266:065]:   Buffer Size: 1000
[17:04:31][C][remote_receiver.esp8266:066]:   Tolerance: 25%
[17:04:31][C][remote_receiver.esp8266:067]:   Filter out pulses shorter than: 50 us
[17:04:31][C][remote_receiver.esp8266:068]:   Signal is done after 10000 us of no changes
```

**Additional information and things you've tried:**

[image]
[image]

taifunk, 2021-04-06:
@JoppyFurr any hope you could update the toshiba integration code base to add support for fan speeds?

JoppyFurr, 2021-04-07:
I'll take a look this weekend

cslux, 2021-04-08:
I'm currently working on that! It's already working for me. I will create a PR (WIP) this weekend.

JoppyFurr, 2021-04-11:
Okay, I'll leave it to @cslux

cslux, 2021-04-14:
The temperature is also currently limited to 17 °C min and 30 °C max in the Toshiba component. So, do we need to adjust this too? My unit only supports the range 17 °C - 30 °C.

JoppyFurr, 2021-04-14:
My remote has the same limit, 17-30°C

cslux, 2021-04-14:
@taifunk What temperature range does your Toshiba unit/remote support?

taifunk, 2021-04-14:
The correct temperature range in my unit is also 17 - 30. In the ESP config I made a mistake by defining different range.
My remote for reference:
[image]

cslux, 2021-04-14:
@taifunk Perfect, so 17-30 seems to be the default range. If you find some time and could test the PR https://github.com/esphome/esphome/pull/1689 and report back if something is for you different, that would be highly appreciated. Especially the "Frame Toshiba" log output when you are pressing the preset button would be good to know.

taifunk, 2021-04-14:
@cslux could you give some help, how to do it?
I have bunch of errors, if I just try to use custom component folder and add the .cpp and .h files there. Not sure if this is the right way.

cslux, 2021-04-15:
Puh, good question. I don't if this is the way to do it. I thought you have checked out the esphome project locally based on your first post (ESPHome DEV 1.1.17). Do have it checked out locally?

off112a, 2021-04-19:
hi @cslux 

I have a question about the fan modes. You have named them auto, low, medium, high, middle, focus, diffuse. 

But in the owner manual it looks like this. See picture. 

Shouldn't it be something like this: auto, lowest, low, medium, high, highest, auto quiet?

[image]

cslux, 2021-04-19:
Hey @off112a, you are right, but esphome only provides auto, low, medium, high, middle, focus, diffuse (https://github.com/esphome/esphome/blob/dev/esphome/components/climate/climate_mode.h). I'm not sure if it can easily be modified in a way that allows us to use different names.

off112a, 2021-04-19:
ok, that's to bad. 

When I tested the update I only got off and vertical on swing mode. My Toshiba supports off, both, vertical and horizontal. Is that something you will add?

cslux, 2021-04-19:
> When I tested the update I only got off and vertical on swing mode. My Toshiba supports off, both, vertical and horizontal. Is that something you will add?

Sure, I just left it out because my AC unit only supports VERTICAL, OFF and FIXED. Hence, I don't know the correct numbers which need to send in order to activate BOTH or HORIZONTAL mode. Do you have already uploaded to PR version of the toshiba climate component to your ESP device? If yes, can you post the log output of esphome when it receives the swing mode command for both variants HORIZONTAL and BOTH, thanks. Do the swing commands VERTICAL and OFF work for you? Some remotes use different codes...

off112a, 2021-04-19:
@cslux 

Your OFF is working. But your VERTICAL is BOTH on my Tohisba.

Here is the log output from when I press the swing button on my remote. 

VERTICAL:
[20:35:51][D][toshiba.climate:070]: Frame toshiba: 0xF20D01FE210425
[20:35:51][D][toshiba.climate:071]:   header:      0xF20D
[20:35:51][D][toshiba.climate:072]:   data_size:   0x01
[20:35:51][D][toshiba.climate:073]:   1_checksum:  0xFE
[20:35:51][D][toshiba.climate:074]:   command:     0x21
[20:35:51][D][toshiba.climate:075]:   data:        0x04
[20:35:51][D][toshiba.climate:076]:   2_checksum:  0x25

HORIZONTAL:
[20:37:42][D][toshiba.climate:070]: Frame toshiba: 0xF20D01FE210425
[20:37:42][D][toshiba.climate:071]:   header:      0xF20D
[20:37:42][D][toshiba.climate:072]:   data_size:   0x01
[20:37:42][D][toshiba.climate:073]:   1_checksum:  0xFE
[20:37:42][D][toshiba.climate:074]:   command:     0x21
[20:37:42][D][toshiba.climate:075]:   data:        0x04
[20:37:42][D][toshiba.climate:076]:   2_checksum:  0x25

BOTH:
[20:38:46][D][toshiba.climate:070]: Frame toshiba: 0xF20D01FE210425
[20:38:46][D][toshiba.climate:071]:   header:      0xF20D
[20:38:46][D][toshiba.climate:072]:   data_size:   0x01
[20:38:46][D][toshiba.climate:073]:   1_checksum:  0xFE
[20:38:46][D][toshiba.climate:074]:   command:     0x21
[20:38:46][D][toshiba.climate:075]:   data:        0x04
[20:38:46][D][toshiba.climate:076]:   2_checksum:  0x25

OFF:
[20:39:42][D][toshiba.climate:070]: Frame toshiba: 0xF20D01FE210425
[20:39:42][D][toshiba.climate:071]:   header:      0xF20D
[20:39:42][D][toshiba.climate:072]:   data_size:   0x01
[20:39:42][D][toshiba.climate:073]:   1_checksum:  0xFE
[20:39:42][D][toshiba.climate:074]:   command:     0x21
[20:39:42][D][toshiba.climate:075]:   data:        0x04
[20:39:42][D][toshiba.climate:076]:   2_checksum:  0x25

cslux, 2021-04-19:
Thanks! But your data seems to be always 0x04, is this correct? I'm confused right now.

off112a, 2021-04-19:
I noticed that too. 
I have tried once more and it's still 0x04 when I press the swing button on the remote.

cslux, 2021-04-19:
Hmm...

But if you use esphome to set VERTICAL (BOTH) or OFF it works? Can you also post the model number of your remote and maybe a picture too?

off112a, 2021-04-19:
Yes, that works perfect. 

Remote Model number: WH-TA01LE

[image]

cslux, 2021-04-19:
I'll update the code with HORIZONTAL and BOTH variables you can than change the values until it works. Obviously it can't be 0x04 for all cases, can it? 

Currently the following values work for your remote:
OFF: 0x02
BOTH: 0x01
VERTICAL: 0x0? (maybe 0x03)
HORIZONTAL: 0x0? (maybe 0x04)
FIX: 0x00? -> not supported by esphome

cslux, 2021-04-19:
@off112a I have just pushed a new commit, see https://github.com/esphome/esphome/pull/1689/commits/39624a54c3a8d887ef63a17ed65f5f8c170c6ffe

JoppyFurr also reported that his remote sends 0x04 for SWING and OFF, maybe it really sends 0x04 for all swing modes and just cycles between them, see https://github.com/esphome/esphome/pull/1689/commits/205fe58d9fca7b671bc9cf4c49f7f87bf6bf75b4#r612825154. When this is the case we also need to know the order, than we need to send the command N time depending on the current swing mode and the selected.

off112a, 2021-04-20:
@cslux 

BOTH and OFF is working. But when I use HORIZONTAL (0x04) it just cycles to the next swing mode. 

Here is a picture from the manual with the cycle order.

[image]

cslux, 2021-04-20:
Have you also tried to change the numbers in the toshiba.cpp file? 
https://github.com/cslux/esphome/blob/39624a54c3a8d887ef63a17ed65f5f8c170c6ffe/esphome/components/toshiba/toshiba.cpp#L50

0x04 seems than indeed to be the number for cycle I guess.

off112a, 2021-04-20:
I have now tested different numbers and found out that:

VERTICAL = 0x08
HORIZONTAL = 0x05

cslux, 2021-04-20:
> I have now tested different numbers and found out that:
> 
> VERTICAL = 0x08
> HORIZONTAL = 0x05

Perfect! Thanks!

off112a, 2021-07-22:
Hi @cslux,

I tried the original branch after I updated my Esphome and noticed that all AC options still aren't exposed. 

I then tried to use the files I got from you last time and only get errors when trying to install it. See attached file.

Do you maybe have any updated files?

[error.txt](https://github.com/esphome/issues/files/6862085/error.txt)

cslux, 2021-07-22:
Hey @off112a! Unfortunatly not, I will try my best to get even with the dev branch in the next days. I'll let you know!

off112a, 2021-07-22:
Ok, then I will wait. Thank you 🙂

DearSwitch, 2021-08-01:
Hi @off112a!
Only need to do some corrections:
1. toshiba.h 
          std::**vector**<climate::ClimateFanMode> change to std::**set**<climate::ClimateFanMode>
          std::**vector**<climate::ClimateSwingMode> change to std::**set**<climate::ClimateSwingMode>
2. toshiba.cpp
         switch (this->fan_mode) change to switch (this->fan_mode.value()) in string 129
         this->fan_mode; change to this->fan_mode.value(); in string 255 and 386

off112a, 2021-08-02:
Hi @DearSwitch 

I tried to change **vector** to **set** instead and only get errors.

Errors:
`
Compiling /data/esp_ir_ac/.pioenvs/esp_ir_ac/src/esphome/components/toshiba1/toshiba.cpp.o
src/esphome/components/toshiba1/toshiba.cpp: In member function 'void esphome::toshiba::ToshibaClimate::add_default_command_data_(uint8_t*)':
src/esphome/components/toshiba1/toshiba.cpp:123:106: error: no matching function for call to 'clamp(float&, const uint8_t&, const uint8_t&)'
     temperature = static_cast<uint8_t>(clamp(this->target_temperature, TOSHIBA_TEMP_MIN, TOSHIBA_TEMP_MAX));
                                                                                                          ^
src/esphome/components/toshiba1/toshiba.cpp:123:106: note: candidate is:
In file included from src/esphome/components/climate/climate.h:4:0,
                 from src/esphome/components/climate_ir/climate_ir.h:5,
                 from src/esphome/components/toshiba1/toshiba.h:3,
                 from src/esphome/components/toshiba1/toshiba.cpp:1:
src/esphome/core/helpers.h:83:24: note: template<class T> T esphome::clamp(T, T, T)
 template<typename T> T clamp(T val, T min, T max);
                        ^
src/esphome/core/helpers.h:83:24: note:   template argument deduction/substitution failed:
src/esphome/components/toshiba1/toshiba.cpp:123:106: note:   deduced conflicting types for parameter 'T' ('float' and 'unsigned char')
     temperature = static_cast<uint8_t>(clamp(this->target_temperature, TOSHIBA_TEMP_MIN, TOSHIBA_TEMP_MAX));
                                                                                                          ^
*** [/data/esp_ir_ac/.pioenvs/esp_ir_ac/src/esphome/components/toshiba1/toshiba.cpp.o] Error 1`

DearSwitch, 2021-08-02:
@off112a check with attached files, maybe there are another changes ))

[toshiba.h.txt](https://github.com/esphome/issues/files/6916628/toshiba.h.txt)
[toshiba.cpp.txt](https://github.com/esphome/issues/files/6916629/toshiba.cpp.txt)
[climate.py.txt](https://github.com/esphome/issues/files/6916631/climate.py.txt)

off112a, 2021-08-03:
@DearSwitch Thanks.  Now it works again 🙂

DearSwitch, 2021-08-04:
@off112a you are welcome

wltng, 2022-02-21:
Hi @cslux ,
Thanks for your work on this. You think there is a possibility this will be added to the dev branch? Or is there another way to install this and keep my esphome updated?
Thanks!

MGlants, 2022-06-18:
Up really need that guys

magnusoverli, 2023-11-28:
Swing modes for the Toshiba protocol is something I have also been waiting for.
Is there a reason this is not merged yet?

Also; my remote is the WH-TA15PE. How can I contribute the remote data that corresponds to its swing modes?

casch-at, 2024-02-13:
> Old thread, sorry for the ping all!
> 
> But I just migrated from Tasmote to ESPHome, and probably the only feature I have automated from my AC is swing mode.
> I saw that the PR got closed for inactivity, would it be possible to reopen it? 
> 
> I can update this myself also this week if the author does not have time/interest.

Sorry, it is the AC unit from my ex, so don't expect anything anytime soon from my site 😬