esphome/issues #1756: wifi connection fails when `esphome::on_boot` is declared using 1.17.0-dev

gretel, 2021-01-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

OSX

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff BASIC

**ESPHome version (latest production, beta, dev branch)**

1.17.0-dev compiled on Jan 12 2021

**Affected component:**

wifi

**Description of problem:**
flashing away from `espurna` all devices tried could not connect to (proven working) accesspoint anymore.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m
  on_boot:
    then:
      - if:
          condition:
            wifi.connected:
          then:
            - logger.log: wifi is connected!
  on_shutdown:
    then:
      - switch.turn_off: relay
      - logger.log: shutting down!

```

**Logs (if applicable):**

```
␛[0;32m[I][wifi:194]: WiFi Connecting to 'PlanetFall'...␛[0m
␛[0;33m[W][wifi_esp8266:444]: Event: Disconnected ssid='PlanetFall' bssid=␛[5m74:2F:68:4A:E0:AB␛[6m reason='Association Expired'␛[0m
␛[0;33m[W][wifi:485]: Error while connecting to network.␛[0m
␛[0;33m[W][wifi:522]: Restarting WiFi adapter...␛[0m

```

**Additional information and things you've tried:**
__removing__ the `on_boot:` block fixes this issue:

```yaml
esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m
  on_shutdown:
    then:
      - switch.turn_off: relay
      - logger.log: shutting down!
```

gretel, 2021-02-05:
using the `wifi.connected` condition is just a factor - the root cause is having `status_led` declared.

think there is a race condition lurking. when `status_led` is used even uploading firmware via the fallback is not working as the connection gets interrupted. `esp32` seems to be more tolerant due to more cycles available.

took me another day to figure. disappointing to see so many reports even mentioning this change got closed.

relates https://github.com/esphome/issues/issues/163, https://github.com/esphome/issues/issues/432, https://github.com/esphome/issues/issues/455, https://github.com/esphome/issues/issues/1120, https://github.com/esphome/issues/issues/1658 at least.

pki791, 2021-03-14:
Hi, i have a similar issue. I also had if earlier but i was not sure what the cause was.

Today i changed
`time:
  - platform: homeassistant
    on_time:
    - seconds: 0
      minutes: 30
      hours: 0
      then:
        - light.turn_off: lightbar
    - seconds: 0
      minutes: 0
      hours: 6
      then:
        - light.turn_on: lightbar
`
to

`time:
  - platform: homeassistant
    on_time:
    - seconds: 0
      minutes: 30
      hours: 0
      then:
        - light.turn_off: lightbar
    - seconds: 0
      minutes: 0
      hours: 6
      then:
        - if:
            condition:
              - sun.is_below_horizon:
                 elevation: 6
            then:
            - light.turn_on: lightbar
`

It causes the device failed to connect wifi. Logs before and after OTA:
`[13:28:06][D][wifi:339]: Found networks:
[13:28:06][I][wifi:385]: - 'MEW' (CC:2D:E0:E5:1F:81) ?-?-
[13:28:06][D][wifi:386]:     Channel: 6
[13:28:06][D][wifi:387]:     RSSI: -68 dB
[13:28:06][D][wifi:389]: - 'MEW-X' (9C:B2:B2:EA:22:71) ?-?-
[13:28:06][I][wifi:194]: WiFi Connecting to 'MEW'...
[13:28:07][I][wifi:457]: WiFi Connected!
[13:28:07][C][wifi:303]:   SSID: 'MEW'
[13:28:07][C][wifi:304]:   IP Address: 192.168.1.211
[13:28:07][C][wifi:306]:   BSSID: CC:2D:E0:E5:1F:81
[13:28:07][C][wifi:307]:   Hostname: 'swiatlo_wejscie1'
[13:28:07][C][wifi:311]:   Signal strength: -78 dB ?-?-
[13:28:07][C][wifi:315]:   Channel: 6
[13:28:07][C][wifi:316]:   Subnet: 255.255.255.0
[13:28:07][C][wifi:317]:   Gateway: 192.168.1.50
[13:28:07][C][wifi:318]:   DNS1: 192.168.1.50
[13:28:07][C][wifi:319]:   DNS2: (IP unset)
[13:28:07][D][wifi:466]: Disabling AP...
[13:28:07][C][ota:029]: Over-The-Air Updates:
[13:28:07][C][ota:030]:   Address: swiatlo_wejscie1.local:8266
[13:28:07][C][ota:032]:   Using Password.
[13:28:07][C][api:022]: Setting up Home Assistant API server...
[13:28:07][I][app:059]: setup() finished successfully!
[13:28:07][I][app:105]: ESPHome version 1.16.2 compiled on Mar 14 2021, 13:26:17
[13:28:07][C][wifi:443]: WiFi:
[13:28:07][C][wifi:303]:   SSID: 'MEW'
[13:28:07][C][wifi:304]:   IP Address: 192.168.1.211
[13:28:07][C][wifi:306]:   BSSID: CC:2D:E0:E5:1F:81
[13:28:07][C][wifi:307]:   Hostname: 'swiatlo_wejscie1'
[13:28:07][C][wifi:311]:   Signal strength: -78 dB ?-?-
[13:28:07][C][wifi:315]:   Channel: 6
[13:28:07][C][wifi:316]:   Subnet: 255.255.255.0
[13:28:07][C][wifi:317]:   Gateway: 192.168.1.50
[13:28:07][C][wifi:318]:   DNS1: 192.168.1.50
[13:28:07][C][wifi:319]:   DNS2: (IP unset)
[13:28:07][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'swiatlo_wejscie1 Button'
[13:28:07][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[13:28:07][C][gpio.output:010]: GPIO Binary Output:
[13:28:07][C][gpio.output:011]:   Pin: GPIO12 (Mode: OUTPUT)
[13:28:07][C][logger:185]: Logger:
[13:28:07][C][logger:186]:   Level: DEBUG
[13:28:07][C][logger:187]:   Log Baud Rate: 115200
[13:28:07][C][logger:188]:   Hardware UART: UART0
[13:28:07][C][light:178]: Light 'swiatlo_wejscie1'
[13:28:07][C][homeassistant.time:010]: Home Assistant Time:
[13:28:07][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
[13:28:07][C][captive_portal:169]: Captive Portal:
[13:28:07][C][ota:029]: Over-The-Air Updates:
[13:28:07][C][ota:030]:   Address: swiatlo_wejscie1.local:8266
[13:28:07][C][ota:032]:   Using Password.
[13:28:07][C][api:095]: API Server:
[13:28:07][C][api:096]:   Address: swiatlo_wejscie1.local:6053
[13:29:04][D][api.connection:604]: Client 'Home Assistant 2021.2.3 (192.168.1.11)' connected successfully!
[13:29:04][D][time:040]: Synchronized time: Sun Mar 14 13:29:03 2021
[13:29:34][D][ota:072]: Starting OTA Update from 192.168.1.11...
[13:29:34][D][ota:243]: OTA in progress: 0.2%
[13:29:36][D][ota:243]: OTA in progress: 8.1%
[13:29:37][D][ota:243]: OTA in progress: 11.1%
[13:29:38][D][ota:243]: OTA in progress: 22.7%
[13:29:39][D][ota:243]: OTA in progress: 33.3%
[13:29:40][D][ota:243]: OTA in progress: 44.4%
[13:29:41][D][ota:243]: OTA in progress: 55.5%
[13:29:42][D][ota:243]: OTA in progress: 66.6%
[13:29:43][D][ota:243]: OTA in progress: 77.7%
[13:29:44][D][ota:243]: OTA in progress: 88.0%
[13:29:45][D][ota:243]: OTA in progress: 98.2%
[13:29:45][I][ota:269]: OTA update finished!
[13:29:45][I][app:142]: Rebooting safely...
[13:29:45][W][wifi_esp8266:444]: Event: Disconnected ssid='MEW' bssid=CC:2D:E0:E5:1F:81 reason='Association Leave'
[13:29:45]
[13:29:45] ets Jan  8 2013,rst cause:1, boot mode:(3,6)
[13:29:45]
[13:29:45]load 0x4010f000, len 3584, room 16 
[13:29:45]tail 0
[13:29:45]chksum 0xb0
[13:29:45]csum 0xb0
[13:29:45]v2843a5ac
[13:29:55]@cp:0
[13:29:55]ld
[13:29:55][I][logger:166]: Log initialized
[13:29:55][C][status_led:014]: Setting up Status LED...
[13:29:55][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[13:29:55][I][app:029]: Running through setup()...
[13:29:55][D][binary_sensor:034]: 'swiatlo_wejscie1 Button': Sending initial state OFF
[13:29:55][C][light:097]: Setting up light 'swiatlo_wejscie1'...
[13:29:55][D][light:265]: 'swiatlo_wejscie1' Setting:
[13:29:55][C][wifi:033]: Setting up WiFi...
[13:29:55][D][wifi:324]: Starting scan...
[13:30:01][D][wifi:339]: Found networks:
[13:30:01][D][wifi:341]:   No network found!
[13:30:06][D][wifi:324]: Starting scan...
[13:30:12][D][wifi:339]: Found networks:
[13:30:12][D][wifi:341]:   No network found!
`

After I disabled the status led section it works with the changed conditions.
`#status_led:
#  pin:
#    number: GPIO13
#    inverted: yes
`

Any ideas?

SenexCrenshaw, 2021-03-14:
Id like to see a full yaml

pki791, 2021-03-14:
Here:
```
substitutions:
  device_name: swiatlo_wejscie1

esphome:
  name: '${device_name}'
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "*"
  password: "*"
#  output_power: 17dBm

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "swiatlo wejscie 1 Fa Hotspot"
    password: "*"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: "*"

ota:
  password: "*"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: '${device_name} Button'
    on_press:
      - light.toggle: lightbar

output:
  - platform: gpio
#    name: "Sonoff Basic Relay"
    pin: GPIO12
    id: relay

light:
  - platform: binary
    id: lightbar
    name: '${device_name}'
    output: relay

#status_led:
#  pin:
#    number: GPIO13
#    inverted: yes
    
    
sun:
  latitude: 54.125951
  longitude: 17.978014
  on_sunrise:
  - elevation: -6°
    then:
    - light.turn_off: lightbar
  on_sunset:
  - elevation: -6°
    then:
    - light.turn_on: lightbar

time:
  - platform: homeassistant
    on_time:
    - seconds: 0
      minutes: 30
      hours: 0
      then:
        - light.turn_off: lightbar
    - seconds: 0
      minutes: 0
      hours: 6
      then:
        - if:
            condition:
              - sun.is_below_horizon:
                 elevation: 6
            then:
            - light.turn_on: lightbar
```

I just changed the passwords. If You uncommect the status LED section wifi will stop working.
ESPhome 1.16.2, but was same on 1.15.x.

pki791, 2021-03-21:
Is it possible any of the devs check it?

gretel, 2021-03-21:
> Is it possible any of the devs check it?

think it is advisable to investigate this by ourselves. have been [referencing](https://github.com/esphome/issues/issues/1756#issuecomment-773787239) to numerous other issues in february, but i assume the devs are more interested in implementing new features. who can blame them for that 😏

closed 2021-08-02 by stale[bot]