esphome/issues #1760: esphome-docs: 'Uptime Human Readable' example cannot compiled

mrkeuz, 2021-01-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
N/A

**ESP (ESP32/ESP8266, Board/Sonoff):**
N/A

**ESPHome version (latest production, beta, dev branch)**
N/A

**Affected component:**
esphome-docs -> Uptime Sensor description

**Description of problem:**

Human readable uptime example not compile on board. Missing bracket after 'String(minutes'
See: https://github.com/esphome/esphome-docs/blame/current/components/sensor/uptime.rst#L61

**Problem-relevant YAML-configuration entries:**

```yaml

# Example configuration entry
text_sensor:
  - platform: template
    name: Uptime Human Readable
    id: uptime_human
    icon: mdi:clock-start
sensor:
  - platform: uptime
    name: Uptime Sensor
    id: uptime
    update_interval: 60s
    on_raw_value:
      then:
        - text_sensor.template.publish:
            id: uptime_human
            state: !lambda |-
              int seconds = round(id(uptime).raw_state);
              int days = seconds / (24 * 3600);
              seconds = seconds % (24 * 3600);
              int hours = seconds / 3600;
              seconds = seconds % 3600;
              int minutes = seconds /  60;
              seconds = seconds % 60;
              return (
                (days ? String(days) + "d " : "") +
                (hours ? String(hours) + "h " : "") +
                (minutes ? String(minutes + "m " : "") +
                (String(seconds) + "s")
              ).c_str();

```

**Logs (if applicable):**
Compile logs look like this:

```
...
src/main.cpp: In lambda function:
src/main.cpp:342:26: error: expected primary-expression before '(' token
         (minutes ? String(minutes + "m " : "")) +
                          ^
src/main.cpp:342:42: error: expected ')' before ':' token
         (minutes ? String(minutes + "m " : "")) +
                                          ^
src/main.cpp:342:47: error: expected ':' before ')' token
         (minutes ? String(minutes + "m " : "")) +
                                               ^
src/main.cpp:342:47: error: expected primary-expression before ')' token
*** [.pioenvs/plants/src/main.cpp.o] Error 1
...
```

**Additional information and things you've tried:**

GeneralLuzi, 2021-01-17:
Try it with this:

`    lambda: |-
      uint32_t dur = id(${device_name}_uptime_sensor).state;
      int dys = 0;
      int hrs = 0;
      int mnts = 0;
      if (dur > 86399) {
        dys = trunc(dur / 86400);
        dur = dur - (dys * 86400);
      }
      if (dur > 3599) {
        hrs = trunc(dur / 3600);
        dur = dur - (hrs * 3600);
      }
      if (dur > 59) {
        mnts = trunc(dur / 60);
        dur = dur - (mnts * 60);
      }
      char buffer[17];
      sprintf(buffer, "%ud %02uh %02um %02us", dys, hrs, mnts, dur);
      return {buffer};`

mrkeuz, 2021-01-17:
Thanks! I already fixed lamba. It needs to add bracket immediately after 'String(minutes'. Just create issue for create pull request. 

But anyway, Thanks a lot!

mrkeuz, 2021-01-26:
Fixed in 1.16.0b5

closed 2021-01-26 by mrkeuz