esphome/issues #1764: Nextion touch: second and subsequent button press is not working

GeneralLuzi, 2021-01-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Docker
**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32 Board: esp-wrover-kit
**ESPHome version (latest production, beta, dev branch)** latest dev

**Affected component:** nextion touch

**Description of problem:**
The nextion button should toggle the light on, if the light state is off and vice versa. After OTA, the first button press works great: if the light is on after OTA, it goes off after button press. if the light is off after OTA, it goes onafter button press. 

Unfortunately, the second and subsequent button press are not working. It seams, that the function "on_press" of the binary_sensor is not executed.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: homeassistant   # Kitchen Light state
    id: ${device_name}_top_light_status
    entity_id: light.wohnzimmer_deckenlicht
    on_state:
       then:
          - lambda: |- 
                  if (id(${device_name}_top_light_status).state) {
                    id(${device_name}_nextiondisplay).set_component_text("b1","On");
                  } else {
                    id(${device_name}_nextiondisplay).set_component_text("b1","Off");
                  }
          - component.update: ${device_name}_nextiondisplay
  - platform: nextion
    page_id: 0
    component_id: 6
    id: ${device_name}_top_light_switch
    on_press:
      then:
        if:
          condition:
            lambda: 'return id(${device_name}_top_light_status).state;'
          then:
            - homeassistant.service:
                service: light.turn_off
                data:
                  entity_id: light.wohnzimmer_deckenlicht
            - logger.log: "Button 0_6: Off"
            - component.update: ${device_name}_nextiondisplay
          else:
            - homeassistant.service:
                service: light.turn_on
                data:
                  entity_id: light.wohnzimmer_deckenlicht
            - logger.log: "Button 0_6: On"
            - component.update: ${device_name}_nextiondisplay

display:
  - platform: nextion
    id: ${device_name}_nextiondisplay
    update_interval: 60s
    lambda: |-
            if (id(${device_name}_top_light_status).state) {
              it.set_component_text("b1","On");
            } else {
              it.set_component_text("b1","Off");
            }

```

**Logs (if applicable):**
[20:07:13][D][homeassistant.binary_sensor:021]: 'light.wohnzimmer_deckenlicht': Got state ON
[20:07:13][D][binary_sensor:034]: 'living_room_wall_mount_top_light_status': Sending initial state ON
[20:07:16][D][nextion:209]: Got touch page=0 component=6 type=PRESS
[20:07:16][D][binary_sensor:036]: 'living_room_wall_mount_top_light_switch': Sending state ON
[20:07:16][D][main:522]: Button 0_6: Off
[20:07:17][D][homeassistant.binary_sensor:021]: 'light.wohnzimmer_deckenlicht': Got state OFF
[20:07:17][D][binary_sensor:036]: 'living_room_wall_mount_top_light_status': Sending state OFF
[20:07:51][D][nextion:209]: Got touch page=0 component=6 type=PRESS
[20:07:51][D][nextion:209]: Got touch page=0 component=6 type=PRESS
[20:07:52][D][nextion:209]: Got touch page=0 component=6 type=PRESS

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
n/a

errolsancaktar, 2021-02-03:
I'm running into the same issue, did you find a resolution?

errolsancaktar, 2021-02-03:
See this issue.  solved it for me: https://github.com/esphome/issues/issues/1367

closed 2021-06-18 by stale[bot]