esphome/issues #1765: Sometimes on_time automation doesn't activate

kristjanbjarni, 2021-01-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Official Home Assistant 2021.1.1 running on Raspberry PI 4 Model B 4GB

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic

**ESPHome version (latest production, beta, dev branch)**

Latest production ESPHome 1.15.3

**Affected component:**

https://esphome.io/components/time.html

**Description of problem:**
Sometimes on_time automation doesn't run at all and it seems to be related to when WiFi goes down or has weak signal. This does not happen all the time. I had to try this about 5 times until I was able to hit on this issue. I reproduced the problem in the following manner: I connected the Sonoff to my PC with FTDI (The current time was about 17:50). I set the timer in the Yaml to go off 17:55. I then flashed the Sonoff Basic from ESPHome in Home Assistant with OTA. After the flashing I connected to the logs using port COM12. I then put the Sonoff in a sealed aluminum container to simulate weak WiFi, then monitored the logs until 17:56, the relay didn't fire at 17:55 as it should. At 17:56 I opened the aluminum container and the WiFi started logging a successful connection.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: sbasic_1
  
esphome:
  name: ${devicename}
  comment: Hitaþráður
  platform: ESP8266
  board: esp01_1m
  #on_boot:
  #  then:
  #    - switch.turn_on: relay

wifi:
  reboot_timeout: 0s
  networks:
  - ssid: "Skynet"
    password: !secret wifi
  - ssid: "Skynet_Efri_Haed"
    password: !secret wifi
  ap:
    ssid: "${devicename}_fallback"
    password: !secret wifi

captive_portal:

logger:

api:
  reboot_timeout: 0s

ota:

time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 55
        hours: 17
        then:
          - switch.turn_on: relay
          - logger.log: "Turning on"
      #- seconds: 0
      #  minutes: 44
      #  hours: 15
      #  then:
      #    - switch.turn_off: relay
      #    - logger.log: "Turning off"

binary_sensor:
[... 164 lines cut ...]
[17:51:53][C][api:096]:   Address: sbasic_1.local:6053
[17:52:06][D][api:067]: Disconnecting Home Assistant 2021.1.1 (192.168.3.60)
[17:53:37][I][ota:046]: Boot seems successful, resetting boot loop counter.
[17:56:04][W][wifi_esp8266:395]: Event: Disconnected ssid='Skynet' bssid=F8:23:B2:82:A4:18 reason='Not Associated'
[17:56:04][W][wifi:100]: WiFi Connection lost... Reconnecting...
[17:56:04][W][wifi:522]: Restarting WiFi adapter...
[17:56:04][W][wifi_esp8266:395]: Event: Disconnected ssid='Skynet' bssid=F8:23:B2:82:A4:18 reason='Authentication Failed'
[17:56:04][I][wifi:194]: WiFi Connecting to 'Skynet'...
[17:56:04][W][wifi_esp8266:191]: wifi_apply_hostname_(sbasic_1): lwIP error -16 on interface st (index 0)
[17:56:05][I][wifi:457]: WiFi Connected!
[17:56:05][C][wifi:303]:   SSID: 'Skynet'
[17:56:05][C][wifi:304]:   IP Address: 192.168.3.71
[17:56:05][C][wifi:306]:   BSSID: F8:23:B2:82:A4:18
[17:56:05][C][wifi:307]:   Hostname: 'sbasic_1'
[17:56:05][C][wifi:311]:   Signal strength: -77 dB ▂▄▆█
[17:56:05][C][wifi:315]:   Channel: 13
[17:56:05][C][wifi:316]:   Subnet: 255.255.255.0
[17:56:05][C][wifi:317]:   Gateway: 192.168.3.1
[17:56:05][C][wifi:318]:   DNS1: 192.168.3.1
[17:56:05][C][wifi:319]:   DNS2: 192.168.3.1
[17:56:05][D][wifi:466]: Disabling AP...

```

**Additional information and things you've tried:**

kristjanbjarni, 2021-01-19:
Managed to recreate this bug again and having log level VERBOSE. I created three timers 11:00, 11:05 and 11:10. The first two timers fired but not the third one. I reset the relay by hand each time.

Here is the changed Yaml portion:
```yaml
time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 0
        hours: 11
        then:
          - switch.turn_on: relay
          - logger.log: "Turning on (11:00)"
      - seconds: 0
        minutes: 5
        hours: 11
        then:
          - switch.turn_on: relay
          - logger.log: "Turning on (11:05)"
      - seconds: 0
        minutes: 10
        hours: 11
        then:
          - switch.turn_on: relay
          - logger.log: "Turning on (11:10)"
```

Here is the log.
```
Using 'COM12' as serial port.
Showing logs:
[10:57:59]scandone
[10:57:59][D][wifi:339]: Found networks:
[10:57:59][I][wifi:385]: - 'Skynet_Efri_Haed' (80:1F:02:31:9E:6A) ?_?¦
[10:57:59][D][wifi:386]:     Channel: 11
[10:57:59][D][wifi:387]:     RSSI: -79 dB
[10:57:59][I][wifi:385]: - 'Skynet' (F8:23:B2:82:A4:18) ?_?¦
[10:57:59][D][wifi:386]:     Channel: 13
[10:57:59][D][wifi:387]:     RSSI: -81 dB
[10:57:59][I][wifi:194]: WiFi Connecting to 'Skynet_Efri_Haed'...
[10:57:59][V][wifi:196]: Connection Params:
[10:57:59][V][wifi:197]:   SSID: 'Skynet_Efri_Haed'
[10:57:59][V][wifi:200]:   BSSID: 80:1F:02:31:9E:6A
[10:57:59][V][wifi:220]:   Password: 'XXX'
[10:57:59][V][wifi:225]:   Channel: 11
[10:57:59][V][wifi:235]:   Using DHCP IP
[10:57:59][V][wifi:237]:   Hidden: NO
[10:57:59][V][wifi_esp8266:402]: Event: Changed AuthMode old=OPEN new=WPA2 PSK
[10:57:59]wifi evt: 2
[10:57:59]scandone
[10:58:00]state: 0 -> 2 (b0)
[10:58:00]state: 2 -> 3 (0)
[10:58:00]state: 3 -> 5 (10)
[10:58:00]add 0
[10:58:00]aid 7
[10:58:00]cnt 
[10:58:00]
[10:58:00]connected with Skynet_Efri_Haed, channel 11
[10:58:00]dhcp client start...
[10:58:00][V][wifi_esp8266:383]: Event: Connected ssid='Skynet_Efri_Haed' bssid=80:1F:02:31:9E:6A channel=11
[10:58:00]wifi evt: 0
[10:58:02]ip:192.168.3.71,mask:255.255.255.0,gw:192.168.3.1
[10:58:02][V][wifi_esp8266:417]: Event: Got IP static_ip=192.168.3.71 gateway=192.168.3.1 netmask=255.255.255.0
[10:58:02]wifi evt: 3
[10:58:02][I][wifi:457]: WiFi Connected!
[10:58:02][C][wifi:303]:   SSID: 'Skynet_Efri_Haed'
[10:58:02][C][wifi:304]:   IP Address: 192.168.3.71
[10:58:02][C][wifi:306]:   BSSID: 80:1F:02:31:9E:6A
[10:58:02][C][wifi:307]:   Hostname: 'sbasic_1'
[10:58:02][C][wifi:311]:   Signal strength: -82 dB ?_?¦
[... 948 lines cut ...]
[11:09:41][D][wifi:324]: Starting scan...
[11:09:47]scandone
[11:09:47][D][wifi:339]: Found networks:
[11:09:47][D][wifi:341]:   No network found!
[11:09:52][D][wifi:324]: Starting scan...
[11:09:57]scandone
[11:09:57][D][wifi:339]: Found networks:
[11:09:57][D][wifi:341]:   No network found!
[11:10:02][D][wifi:324]: Starting scan...
[11:10:08]scandone
[11:10:08][D][wifi:339]: Found networks:
[11:10:08][D][wifi:341]:   No network found!
[11:10:13][D][wifi:324]: Starting scan...
[11:10:19]wifi evt: 7
[11:10:19]wifi evt: 7
[11:10:19]scandone
[11:10:19][D][wifi:339]: Found networks:
[11:10:19][D][wifi:341]:   No network found!
[11:10:24][D][wifi:324]: Starting scan...
[11:10:30]scandone
[11:10:30][D][wifi:339]: Found networks:
[11:10:30][D][wifi:341]:   No network found!
```
At one point there is a crash 11:05:24:
Fatal exception 9(LoadStoreAlignmentCause). Is this the culprit?

Smitty357, 2021-02-11:
Similar issue here since early to mid January. I have not been able to get the timed events to work at all.
Everything has been working for over a year now until then. Thinking it was after an update, but cant remember which one since I typically do not update immediately.

closed 2021-06-25 by stale[bot]