esphome/issues #1769: RC522 only works with logging set to verbose

rradar, 2021-01-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

D1 Mini, esp8266

**ESPHome version (latest production, beta, dev branch)**

1.16.0b4 (beta)

**Affected component:**

- RC522 (https://beta.esphome.io/components/binary_sensor/rc522.html)

**Description of problem:**

The reading of tags only works when logging is set to `verbose` for versions 1.16.0b4 and later (1.17.0-dev).

In 1.16.0b3 it was still working when logging was default.

**Problem-relevant YAML-configuration entries:**
```yaml
rc522_spi:
  cs_pin: D8
  update_interval: 1s
  on_tag:
    then:
    - homeassistant.tag_scanned: !lambda 'return x;'
    - if:
        condition:
          switch.is_on: buzzer_enabled
        then:
        - rtttl.play: "success:d=24,o=5,b=100:c,g,b"
    - if:
        condition:
          switch.is_on: led_enabled
        then:
        - light.turn_on:
            id: activity_led
            brightness: 30%
            red: 0%
            green: 100%
            blue: 0%
            flash_length: 500ms
```

**Logs (if applicable):**

With default logging the reading of tags is not working, the logs only show:
```yaml
[23:00:15][D][rc522:121]: Status is now: 1
[23:00:15][D][rc522:121]: Status is now: 3
[23:00:16][D][rc522:121]: Status is now: 1
[23:00:16][D][rc522:121]: Status is now: 3
[23:00:20][D][rc522:121]: Status is now: 1
[23:00:20][D][rc522:121]: Status is now: 3
[23:00:23][D][rc522:121]: Status is now: 1
[23:00:23][D][rc522:121]: Status is now: 3
```

**Additional information and things you've tried:**

Aterfax, 2021-01-17:
Also confirmed by myself on 1.17.0

jesserockz, 2021-02-03:
Can you guys please retest this on the latest beta `1.16.0b8`?
It is working for me now nicely without verbose via esphome/esphome#1506

Aterfax, 2021-02-06:
Just tested this - all working great with:  

`spi:
  clk_pin: D0
  miso_pin: D1
  mosi_pin: D2

rc522_spi:
  cs_pin: D3
  update_interval: 1s
  on_tag:
    then:
    - homeassistant.tag_scanned: !lambda 'return x;'`

glmnet, 2021-02-07:
Working in 1.16

closed 2021-02-07 by glmnet