esphome/issues #1776: Costco Feit Smart Dimmer Stops responding to Commands

ic3man5, 2021-01-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HA

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (TYWE2S)
Intertek 5014010

See here for more details: https://community.home-assistant.io/t/costco-feit-smart-dimmer-tuya-convert-tasmota/151390

**ESPHome version (latest production, beta, dev branch)**
1.15.3

**Affected component:**
Tuya Dimmer

**Description of problem:**
I have 8 of these dimmers all flashed with ESPHOME firmware with the YAML script below. Issue I've found is the secondary chip connected to the ESP8266 stops responding to commands sent from the ESP8266. The ESP8266 can still read the state of the dimmer when physically changing the brightness/state. I suspect the secondary chip can't handle the UART traffic quick enough and the secondary chip's UART controller overflows and isn't smart enough to recover from it. 

**Problem-relevant YAML-configuration entries:**
```esphome:
  name: dimmer
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "asdf"
  password: "asdf"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dimmer 1 Fallback Hotspot"
    password: "dUg1c33SV9mg"

captive_portal:

# Make sure logging is not using the serial port
logger:
  baud_rate: 0
  level: NONE

# Enable Home Assistant API
api:
  password: "asdf"

ota:
  password: "asdf"

# My dimmer used the hardware serial port on the alternate pins
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

# Make the light
light:
  - platform: "tuya"
    name: "dimmer1"
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 0
    max_value: 1444

```

**Logs (if applicable):**

**Additional information and things you've tried:**

I believe if I can figure out how to add a parameter to the YAML config and have `void Tuya::send_command_()` delay on each byte sent this would fix it. I'm not familiar enough with the code yet to be able to modify anything. I'd like to add a yaml parameter `serial_tx_delay_ms: 1` that is configurable (defaults to 0).

```
# Register the Tuya MCU connection
tuya:
 # Delay in between each byte
 serial_tx_delay_ms: 1
 # Delay at the end of the command sent (needs a better name)
 serial_tx_cmd__delay_ms: 50

# Make the light
light:
  - platform: "tuya"
    name: "dimmer1"
    dimmer_datapoint: 2
    switch_datapoint: 1
    min_value: 0
    max_value: 1444
```

ic3man5, 2021-01-22:
I made some initial changes here, I have yet to compile them or test them (Still have no idea how to do that currently - Next on my list of things to learn here).

https://github.com/ic3man5/esphome/commit/64ea6dc4140497dbdf5b2f561d02c2f59cda506d

ic3man5, 2021-01-23:
I managed to get this working, it works better but I'm not completely happy with it. Seems its masking some bigger problem underneath. Extending the delays too long makes the initial problem even worse, but a 100us delay in-between bytes seems to help.

This leads me to believe there is some "race" condition going on here (not sure if ESPHome uses interrupts/thread/etc) or the baudrate isn't exactly 9600 like it should be.

```
# Register the Tuya MCU connection
tuya:
  serial_byte_delay_us: 100
  serial_command_delay_us: 10
```

I found removing this return here https://github.com/esphome/esphome/blob/dev/esphome/components/tuya/tuya.cpp#L263 makes the issue worse also. I'm not 100% sure what is going on yet as I don't have a full understanding of ESPHome or the tuya code (assuming someone reverse engineered it somewhere).

ic3man5, 2021-01-23:
@OttoWinter Seems like you were first to write this module so I'm hoping you'll have some input here.

MattW2, 2021-02-01:
@ic3man5 , I have 10 Costco Feit dimmers installed all with ESPHome running on them. I've been dealing with the same occasional problem and have been looking for a solution for months.

It seems that this pull request directly address the issue. 
https://github.com/esphome/esphome/pull/1404

ic3man5, 2021-02-01:
> @ic3man5 , I have 10 Costco Feit dimmers installed all with ESPHome running on them. I've been dealing with the same occasional problem and have been looking for a solution for months.
> 
> It seems that this pull request directly address the issue.
> [esphome/esphome#1404](https://github.com/esphome/esphome/pull/1404)

Thanks! I'll check this out, I can confirm my fix doesn't work. This seems to happen when there is a lot of physical interaction at the switch.

MattW2, 2021-02-24:
Just want to put this out here. I've been testing https://github.com/esphome/esphome/pull/1404
and https://github.com/TheEebb/tasmota-tuyamcu-fix
and as much as I don't like tasmota. The custom version of tasmota has not failed once in over a week. Whereas I had a single failure of the esphome with pull #1404

jkl1337, 2021-03-02:
looking at the tuya command queue code that was just merged, I think there might be one more unresolved problem.
It looks like the loop handles chars and then processes the command queue. That much is fine, however sending a command also causes a send. There does not seem to be anything that inhibits sending bytes to the MCU while it is in the middle of receiving a packet. I think one some MCU firmware, this lock-step is the only way to completely prevent problems.

freeloader247, 2021-06-06:
I'm on ESPHOME 1.18.0 and still have this issue with the Costco Feit dimmers.  PR esphome/esphome#1404 doesn't seem to do the trick.

MattW2, 2021-06-07:
> I'm on ESPHOME 1.18.0 and still have this issue with the Costco Feit dimmers. PR [esphome/esphome#1404](https://github.com/esphome/esphome/pull/1404) doesn't seem to do the trick.

As much as I really like ESPHome for the Feit dimmers I've given up on using it. I've now flashed all (8 units) with https://github.com/TheEebb/tasmota-tuyamcu-fix this custom version of tasmota has worked without issue for months.

closed 2021-06-18 by jesserockz