esphome/issues #1778: Color logs still broken maybe.

foxsam21, 2021-01-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip installation of esphome in a venv
**ESP (ESP32/ESP8266, Board/Sonoff):**

any
**ESPHome version (latest production, beta, dev branch)**

1.17.0-dev

**Affected component:**

**Description of problem:**
Color logs still missing even after fix.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I saw https://github.com/esphome/esphome/pull/1385 and understand it to be fixed however my experience is that it is only fixed in dashboard not in cli. I tried asking in discord but I think there will be more work and exposure here.
Here is the cli:
[image]

and here is dashboard:
[image]

gretel, 2021-01-22:
same for me.

closed 2021-04-08 by OttoWinter

foxsam21, 2021-04-08:
Works good so far. Thanks for everything.