esphome/issues #1779: Nextion button and display not working from Esphome

mrteausa, 2021-01-22:
am using a wemos D1 mini with a 23.5 inch Nextion Display
Affected component:
Nextion Touch Component/Nextion Display Component
Description of problem:
I am unable to use any buttons with the Nextion Display. esphome doesn't seem to see when I touch the button. Also when I use Lambda to populate some of the components for the display it gives errors but you can see it clearly did the task correctly because you can see it on the display.```
**Additional information and things you've tried:**
I tried examples on the youtube weather station.  Unable to get the temperature to display on the Nextion.  Try for weeks on all examples but non working.   I have to convert the HMI file for the 3.5 inch screen.  I send test display still nothing show up on the Display.  The esphome log does not show anything.

Andoramb, 2021-01-23:
@mrteausa try my example code, [here](https://github.com/Andoramb/Nextion-wall-mount)
It should work

mrteausa, 2021-01-24:
I have already try the examples, it does not work because of HMI file is too old version and does not bring in to the Nextion Editor.  Secondly I get error on  these lines.  When I remove it it does not working.  
on_boot:
    priority: -180
    then:
      - lambda: |-
          id(page1).publish_state(true);
          id(page2).publish_state(false);

My issue is I only put couple lines of test codes with text boxes and it does not display on the Nextion.  What is the issue? I switch the Tx and Rx pins and still same problem communicating.

mrteausa, 2021-01-25:
I figure it out.  Baud_rate need to be 115200.  Any other speed it does not communicate.  Also TX and RX pin need to swap on the board.  Then it should work.

Andoramb, 2021-01-26:
@mrteausa sure: TX-RX needs to be swapped, the baud rates need to match also:
In Esphome YAML file you declare:
```
uart:
  rx_pin: D4   
  tx_pin: D3   
  baud_rate: 9600  //<- This has to match with the display's baud rate
```
You can change the baud rate of the display in 2 ways:
1. Send _bauds=9600_ cmd to the physicaldisplay from the debug menu
2. Add _bauds=9600_ cmd to the _Program.s_ part of the HMI - recommended

In theory this could work with faster baud rates, but slower is safer for regular esp-s/"lots of data".

**Bonus**: I have uploaded a new Nextion version to my repo, there you'll find currently working examples for the dual-state buttons too.

Please close the issue if it's solved :)

mrteausa, 2021-01-26:
The new HMI file works.   When I bought the 3.5 inch Nextion display, I use the sample codes and non are working,  After 1 weeks of testing I figure out that by default all my Nextion display use baud rate is 115200.    You dual-button examples work great for me.  Thanks for your help.    My problem solved..

closed 2021-01-26 by mrteausa