esphome/issues #1782: zyaura: esp32: floating point operation in interrupt routine causes crash

rwhitby, 2021-01-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Command line installation.

**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP32
board: nodemcu-32s

**ESPHome version (latest production, beta, dev branch)**

Version: 1.15.3

**Affected component:**

https://esphome.io/components/sensor/zyaura.html

**Description of problem:**
On the ESP32, performing floating point operations inside an interrupt service routine can cause a Coprocessor Exception.
See https://esp32.com/viewtopic.php?t=1292 for details.
ZyAura component performs floating point operations in an interrupt service routine, and thus crashes on ESP32.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: zyaura
    clock_pin: GPIO22
    data_pin: GPIO23
    co2:
      name: "ZyAura CO2"
    temperature:
      name: "ZyAura Temperature"
    humidity:
      name: "ZyAura Humidity"

```

**Logs (if applicable):**

```
[18:40:52][W][zyaura:034]: Sensor message 109 3129
[18:40:52][W][zyaura:079]: Unknown Type 109: 3129
[18:40:52][W][zyaura:082]: Exit set_data_
[18:40:52][W][zyaura:034]: Sensor message 110 12838
[18:40:52][W][zyaura:079]: Unknown Type 110: 12838
[18:40:52][W][zyaura:082]: Exit set_data_
[18:40:52][W][zyaura:034]: Sensor message 113 543
[18:40:52][W][zyaura:079]: Unknown Type 113: 543
[18:40:52][W][zyaura:082]: Exit set_data_
[18:40:52][W][zyaura:034]: Sensor message 80 552
[18:40:52][W][zyaura:074]: CO2: 552
[18:40:52]Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
[18:40:52]Core 1 register dump:
[18:40:52]PC      : 0x40081149  PS      : 0x00060731  A0      : 0x800811a0  A1      : 0x3ffbe860
WARNING Decoded 0x40081149: esphome::zyaura::ZaSensorStore::set_data_(esphome::zyaura::ZaMessage*) at /Users/rwhitby/Projects/plantmonitor/plantmonitor/src/esphome/components/zyaura/zyaura.cpp:75
[18:40:52]A2      : 0x3ffbb62c  A3      : 0x00000001  A4      : 0x3f404e41  A5      : 0x00000000
[18:40:52]A6      : 0x3ffbaea8  A7      : 0x00000771  A8      : 0x00002710  A9      : 0x3ffbe8c0
[18:40:52]A10     : 0x00000002  A11     : 0x3f404e41  A12     : 0x0000004a  A13     : 0x3f404efa
[18:40:52]A14     : 0x00000228  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000004
[18:40:52]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff
[18:40:52]Core 1 was running in ISR context:
[18:40:52]EPC1    : 0x40081149  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40087105
[18:40:52]
[18:40:52]Backtrace: 0x40081149:0x3ffbe860 0x4008119d:0x3ffbe920 0x4008119d:0x3ffbe940 0x4008149d:0x3ffbe960 0x40084c99:0x3ffbe980 0x4000bfed:0x3ffce3c0 0x4008fca9:0x3ffce3d0 0x4008e521:0x3ffce3f0 0x400d6822:0x3ffce430 0x401e4f8d:0x3ffce4d0 0x401e5059:0x3ffce4f0 0x400ded41:0x3ffce510 0x400e1056:0x3ffce560 0x400ec66b:0x3ffce890 0x4008eb3d:0x3ffce8b0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40081149: esphome::zyaura::ZaSensorStore::set_data_(esphome::zyaura::ZaMessage*) at /Users/rwhitby/Projects/plantmonitor/plantmonitor/src/esphome/components/zyaura/zyaura.cpp:75
WARNING Decoded 0x4008119d: esphome::zyaura::ZaSensorStore::interrupt(esphome::zyaura::ZaSensorStore*) at /Users/rwhitby/Projects/plantmonitor/plantmonitor/src/esphome/components/zyaura/zyaura.cpp:57
WARNING Decoded 0x4008119d: esphome::zyaura::ZaSensorStore::interrupt(esphome::zyaura::ZaSensorStore*) at /Users/rwhitby/Projects/plantmonitor/plantmonitor/src/esphome/components/zyaura/zyaura.cpp:57
WARNING Decoded 0x4008149d: __onPinInterrupt at /Users/rwhitby/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.c:218
WARNING Decoded 0x40084c99: _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x4008fca9: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x4008e521: xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2520
WARNING Decoded 0x400d6822: esphome::esp32_ble_tracker::ESP32BLETracker::loop() at /Users/rwhitby/Projects/plantmonitor/plantmonitor/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:241
WARNING Decoded 0x401e4f8d: esphome::Component::call_loop() at /Users/rwhitby/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1754
WARNING Decoded 0x401e5059: esphome::Component::call() at /Users/rwhitby/Projects/plantmonitor/plantmonitor/src/esphome/core/component.cpp:82
WARNING Decoded 0x400ded41: esphome::Application::setup() at /Users/rwhitby/Projects/plantmonitor/plantmonitor/src/esphome/core/application.cpp:50 (discriminator 2)
WARNING Decoded 0x400e1056: setup() at /Users/rwhitby/Projects/plantmonitor/plantmonitor/src/main.cpp:268
WARNING Decoded 0x400ec66b: loopTask(void*) at /Users/rwhitby/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14
WARNING Decoded 0x4008eb3d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

```

**Additional information and things you've tried:**

Gping tp try converting all floats to doubles.

rwhitby, 2021-01-28:
https://github.com/esphome/esphome/pull/1499

shuuse, 2021-10-04:
Any news here? Experiencing the same.