esphome/issues #1785: Custom Component not showing Entity in HA

toomuchcookies, 2021-01-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Hass.io on a Raspberry Pi 4

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266 (D1 Mini)

**ESPHome version (latest production, beta, dev branch)** v1.15.3

**Affected component:** https://github.com/esphome/feature-requests/issues/85 (zip file from https://github.com/rmorenoramos )

**Description of problem:**
The component in question does show that it's reading the inputs correctly (it's a multiplexer that can multiplex between 16 analog inputs) and this shows in the log files (see below). But the entity does not show up in HA even after deleting the device and readding it to HA. Also, in the log files one can clearly see, that the values of the sensor are not run through the filter and don't get the unit_of_measurement correct.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: testnodeesp8266
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "SSID"
  password: "PASS"
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Testnodeesp8266 Fallback Hotspot"
    password: "PASS"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "**PASS**"

cd74hc4067:
  adc_pin: A0
  s0_pin: D0
  s1_pin: D1
  s2_pin: D2
  s3_pin: D3
  
sensor:
  # component source code: https://github.com/esphome/feature-requests/issues/85
  - platform: cd74hc4067
    number: 0
    name: "Raumtemperatur"
    id: "test166"
    update_interval: 1s
    unit_of_measurement: "°C"
    internal: false
    filters:
      - calibrate_linear:
        - 0.0 -> 0.0
        - 0.75 -> 75.0
      - multiply: 3.3
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/testnodeesp8266.yaml...
INFO Starting log output from testnodeesp8266.local using esphome API
INFO Connecting to testnodeesp8266.local:6053 (192.168.178.55)
INFO Successfully connected to testnodeesp8266.local
[20:52:27][I][app:105]: ESPHome version 1.15.3 compiled on Jan 24 2021, 20:09:02
[20:52:27][C][wifi:443]: WiFi:
[20:52:27][C][wifi:303]:   SSID: [redacted]
[20:52:27][C][wifi:304]:   IP Address: 192.168.178.55
[20:52:27][C][wifi:306]:   BSSID: [redacted]
[20:52:27][C][wifi:307]:   Hostname: 'testnodeesp8266'
[20:52:27][C][wifi:311]:   Signal strength: -41 dB ▂▄▆█
[20:52:27][C][wifi:315]:   Channel: 1
[20:52:27][C][wifi:316]:   Subnet: 255.255.255.0
[20:52:27][C][wifi:317]:   Gateway: 192.168.178.1
[20:52:27][C][wifi:318]:   DNS1: 192.168.178.1
[20:52:27][C][wifi:319]:   DNS2: (IP unset)
[20:52:27][C][cd74hc4067:021]: Setting up CD74HC4067...
[20:52:27][C][logger:185]: Logger:
[20:52:27][C][logger:186]:   Level: DEBUG
[20:52:27][C][logger:187]:   Log Baud Rate: 115200
[20:52:27][C][logger:188]:   Hardware UART: UART0
[20:52:27][D][sensor:092]: 'WiFi Signal Sensor': Sending state -41.00000 dB with 0 decimals of accuracy
[20:52:27][C][captive_portal:169]: Captive Portal:
[20:52:27][C][ota:029]: Over-The-Air Updates:
[20:52:27][C][ota:030]:   Address: testnodeesp8266.local:8266
[20:52:27][C][ota:032]:   Using Password.
[20:52:27][C][api:095]: API Server:
[20:52:27][C][api:096]:   Address: testnodeesp8266.local:6053
[20:52:27][D][sensor:092]: 'Raumtemperatur': Sending state 0.07520  with 0 decimals of accuracy
[20:52:27][C][wifi_signal.sensor:009]: WiFi Signal 'WiFi Signal Sensor'
[20:52:27][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[20:52:27][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[20:52:27][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[20:52:28][D][sensor:092]: 'Raumtemperatur': Sending state 0.07520  with 0 decimals of accuracy
[20:52:29][D][sensor:092]: 'Raumtemperatur': Sending state 0.07422  with 0 decimals of accuracy
[20:52:30][D][sensor:092]: 'Raumtemperatur': Sending state 0.07520  with 0 decimals of accuracy
[20:52:31][D][sensor:092]: 'Raumtemperatur': Sending state 0.07422  with 0 decimals of accuracy
[20:52:32][D][sensor:092]: 'Raumtemperatur': Sending state 0.07422  with 0 decimals of accuracy
```

**Additional information and things you've tried:**
I've tried enabling mqtt, but then the yaml wouldn't compile! (mqtt does work without the custom component).

rmorenoramos, 2021-01-28:
I just saw this...It seems like ti is measuring adequately, but the problem might be in the sensor configuration, or maybe the hardware setting. Can you maybe post the configuration of your setup?

rmorenoramos, 2021-01-28:
Ok, I just took a look at it. It might be that I have forgotten to register the sensor. I have not tried it yet, but maybe you can add a version of this to sensor.py ?

`  def to_code(config):
    parent = yield cg.get_variable(config[CONF_cd74hc4067_ID])
    var = cg.new_Pvariable(config[CONF_ID], parent, config[CONF_NAME], config[CONF_NUMBER])
    yield cg.register_component(var, config)
    yield sensor.register_sensor(var, config)

    cg.add(parent.register_sensor(var))`

toomuchcookies, 2021-01-28:
Yes, thanks. I did find out by chance that i needed to register the sensor. But what does cg.add(parent.register_sensor(var)) do?

toomuchcookies, 2021-01-28:
Oh, and will you by any chance try to make this component into a pull request for homeassistant. I feel, it might benefit alot more, if it was available as normal component. I actually had to ask around to find out, that i needed to copy the files into a newly created folder "custom_components" under /config/esphome/ .

Spirit13, 2021-03-10:
Need help.
1. Uploaded files from zip to /config/esphome/cd74hc4067 folder
2. Restarted server
3. added conig section to device
4. uploaded firmware
5. log is good. i can see pressure_10 values
6. HA found device but no sensors there
7. Added code as @rmorenoramos described 
8. whe compile firmware got error.

```
src/main.cpp: In function 'void setup()':
src/main.cpp:193:35: error: 'class esphome::cd74hc4067::CD74HC4067Component' has no member named 'register_sensor'
   cd74hc4067_cd74hc4067component->register_sensor(pressure_10);
```

rggmiranda, 2021-03-16:
> Need help.
> 
> 1. Uploaded files from zip to /config/esphome/cd74hc4067 folder
> 2. Restarted server
> 3. added config section to device
> 
> ```
> cd74hc4067:
>   s0_pin: D1
>   s1_pin: D2
>   s2_pin: D3
>   s3_pin: D4     
>   adc_pin: A0 
> 
> sensor:
>   - platform: cd74hc4067
>     number: 0
>     id: pressure_10
>     update_interval: 15s
> ```
> 
> 1. uploaded firmware
> 2. log is good. i can see pressure_10 values
> 3. HA found device but no sensors present
> 4. Added code as @rmorenoramos described
> 5. Try to compile firmware, but got error.
> 
> ```
> src/main.cpp: In function 'void setup()':
> src/main.cpp:193:35: error: 'class esphome::cd74hc4067::CD74HC4067Component' has no member named 'register_sensor'
>    cd74hc4067_cd74hc4067component->register_sensor(pressure_10);
> ```

I had the same issue that you have. I solved it by removing the line `cg.add(parent.register_sensor(var))`

my entire `sensor.py` file is:
`import esphome.codegen as cg
import esphome.config_validation as cv
from esphome.components import sensor, voltage_sampler
from esphome.const import CONF_GAIN, CONF_NUMBER, ICON_FLASH, UNIT_VOLT, CONF_ID, CONF_NAME
from . import cd74hc4067_ns, CD74HC4067Component

DEPENDENCIES = ['cd74hc4067']

CD74HC4067Sensor = cd74hc4067_ns.class_('CD74HC4067Sensor', sensor.Sensor, cg.PollingComponent,
                                  voltage_sampler.VoltageSampler)

CONF_cd74hc4067_ID = 'cd74hc4067_id'
CONFIG_SCHEMA = sensor.sensor_schema(UNIT_VOLT, ICON_FLASH, 3).extend({
    cv.GenerateID(): cv.declare_id(CD74HC4067Sensor),
    cv.GenerateID(CONF_cd74hc4067_ID): cv.use_id(CD74HC4067Component),
    cv.Required(CONF_NUMBER): cv.int_range(0, 15)
}).extend(cv.polling_component_schema('60s'))

def to_code(config):
    parent = yield cg.get_variable(config[CONF_cd74hc4067_ID])
    var = cg.new_Pvariable(config[CONF_ID], parent, config[CONF_NAME], config[CONF_NUMBER])
    yield cg.register_component(var, config)
    yield sensor.register_sensor(var, config)`

Spirit13, 2021-03-16:
> > ```
> I had the same issue that you have. I solved it by removing the line `cg.add(parent.register_sensor(var))` ...
> ```

Thx. Better, but only 1 sensor from config added(last one). 
`sensor:
  - platform: cd74hc4067
    name: test 1
    number: 0
    id: pressure_10
    update_interval: 15s
  - platform: cd74hc4067
    name: test 2
    number: 1
    id: pressure_100
    update_interval: 15s 
  - platform: cd74hc4067
    name: Test 3
    number: 2
    id: pressure_1000
    update_interval: 15s `

And id of sensor is not as i defined. but this is not a problem.

rggmiranda, 2021-03-17:
> Thx. Better, but only 1 sensor from config added(last one).
> `sensor:
>
> * platform: cd74hc4067
>   name: test 1
>   number: 0
>   id: pressure_10
>   update_interval: 15s
> * platform: cd74hc4067
>   name: test [...]

I was able to add all the entities, maybe you're missing the `"` in the name, my code is as follows:

```
- platform: cd74hc4067
  number: 0
  id: id0
  name: "adc0"
  update_interval: 100ms
  
- platform: cd74hc4067
  number: 1
  id: id1
  name: "adc1"
  update_interval: 100ms
```

I was not able to get the mux working yet but I bet it is due to an issue with the IC, I'll check it later

nickizgr8, 2021-03-28:
I have added the cd74hc4067 file to my custom components folder and restarted the server. When I insert the code `cd74hc4067:
  s0_pin: D1
  s1_pin: D2
  s2_pin: D3
  s3_pin: D4     
  adc_pin: A0 `
I get component not found error
[image]
What am I do wrong? I am suppose to unzip the file before inserting uploading to the custom components file right? I try uploading it to the components file in ESPHome but the readme file says that anything added like that gets deleted each time ESPhome is reloaded and to make them to the main custom components file. Also I was unable to upload the entire cd74hc4067  file folder. I had to create a new folder in my custom components file named cd74hc4067 then upload the 4 individual files inside that (__init__.py  ,  cd74hc4067.cpp  ,  cd74hc4067.h  ,  sensor.py). Is that how I'm suppose to do it??
[image]

nickizgr8, 2021-03-28:
I have added the cd74hc4067 file to my custom components folder and restarted the server. When I insert the code `cd74hc4067:
  s0_pin: D1
  s1_pin: D2
  s2_pin: D3
  s3_pin: D4     
  adc_pin: A0 `
I get component not found error 
[image]

rggmiranda, 2021-03-29:
> I have added the cd74hc4067 file to my custom components folder and restarted the server. When I insert the code `cd74hc4067: s0_pin: D1 s1_pin: D2 s2_pin: D3 s3_pin: D4 adc_pin: A0 `
> I get [...]

You uploaded the files fine, I think that you may be missing the following:
[image]

nickizgr8, 2021-04-08:
> .....
> You uploaded the files fine, I think that you may be missing the following:
> [image]

Ok I finally  was able to upload the code to my wemos d1 mini without errors (I'll attach my code below) 
My issue now is it does not appear to be providing any [meaningful?] readings. To test it I attached a single battery (18650) via a voltage divider of two 10k ohm resistors. My multimeter is reading 1.88 volts at the center of the divider so I attach this to pin C0 on the multiplexer board. The readings in ESPHome do not change. I've switched to different pins on the cd74hc4067 module but the values remain the same. What am I missing?  
[image]
How does ESPhome know which pins on the multiplexer I'm using? Since I do not specify "C0; C1; C2; etc" as I do with GPIO pins. Code:
`esphome:
  name: bedroom_node
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "aaabbbccc"
  password: "111222333"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Battery Tester Hotspot"
    password: "111222333"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  
ota:

cd74hc4067:
  s0_pin: D1
  s1_pin: D2
  s2_pin: D3
  s3_pin: D4     
  adc_pin: A0 
  
#individual sensors
sensor:
 
  - platform: cd74hc4067
    number: 0
    id: id0
    name: "battery1_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 1
    id: id1
    name: "battery2_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 2
    id: id2
    name: "battery3_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 3
    id: id3
    name: "battery4_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 4
    id: id4
    name: "battery5_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 5
    id: id5
    name: "battery6_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 6
    id: id6
    name: "battery7_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 7
    id: id7
    name: "battery8_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 8
    id: id8
    name: "battery9_V"
    update_interval: 2s
  - platform: cd74hc4067
    number: 9
    id: id9
    name: "battery10_V"
    update_interval: 2s`

nickizgr8, 2021-04-09:
> @nickizgr8 SIG is connected to ADC?
> are GND and VCC (5V) also connected? I have all connected except 'EN'.
> Make sure ADC on the WemosD1 is working fine and not defective... (you could test this easy)
> I have also added a line in the sensor.py file (just to have correct results syntax)
> last 2 lines of my sensor.py
> yield cg.register_component(var, config)
> yield sensor.register_sensor(var, config)
> those files are located here: config\esphome\custom_components\cd74hc4067

@lesleypersyn hey thanks for your reply. I saw a comment above previously where @rggmiranda posted their sensor.py file with that added in so I copy and pasted that into my own sensor.py file a few days ago. Regarding the connections they are as follows 
[image]
When you said to test the ADC pin I did by just taking the blue lead attached to the voltage divider (1.886 volts) and touching it directly to the ADC pin to see if anything changed. The readings in ESPHome were all over the place which was promising since they had all remained the same until then. I reconnected the blue lead to the cd74hc4067 boards C0 and additionally I added some filters to the code. I read that the wemos d1 mini has an internal voltage divider in order to drop 3.3v down to 1v so to get accurate readings I should multiply by 3.3 as described on the ESPHome sensor page: 
[image]
I also have an external voltage divider (two 10k ohm resistors) to cut the 18650 voltage in half so I have multiply the readings again, this time by 2 to bring it back to a full voltage reading (=(Voltage Reading * 3.3) * 2 ). To simplify this for adding the filter in esphome I just multiplied by 6.6 (=3.3*2). My code now looks like this: `
cd74hc4067:
  s0_pin: D1
  s1_pin: D2
  s2_pin: D3
  s3_pin: D4     
  adc_pin: A0 
  
#individual sensors
sensor:
 
  - platform: cd74hc4067
    number: 0
    id: id0
    name: "battery1_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 1
    id: id1
    name: "battery2_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 2
    id: id2
    name: "battery3_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 3
    id: id3
    name: "battery4_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 4
    id: id4
    name: "battery5_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 5
    id: id5
    name: "battery6_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 6
    id: id6
    name: "battery7_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 7
    id: id7
    name: "battery8_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 8
    id: id8
    name: "battery9_V"
    update_interval: 2s
    filters:
      - multiply: 6.6
  - platform: cd74hc4067
    number: 9
    id: id9
    name: "battery10_V"
    update_interval: 2s
    filters:
      - multiply: 6.6`
 
After testing the ADC as you recommended by directly touching the ADC pin with the 1.88 volts and adding the filters to my code I was now starting to get somewhere (and possibly by wiggling some wires around on the breadboard too may be contributing to my new-found voltage readings haha). Voltage is showing *around* the correct voltage but is still not very accurate. To do a simple test I connected the blue lead coming from the 18650 cell to the first 4 pins on the multiplexer so C0, C1, C2, C3 are all measuring the same battery like this and I get the following readings (below):
[image]
[image]
You can see battery 3 is showing way lower than the others and at one point 2 of them were showing ~6 volts for several minutes until I re-uploaded the code??. I have not tried this with separate batteries yet. Does my filter seem correct or is there another way I should be calculating the voltage?

OttoWinter, 2021-04-10:
That is an error in the custom component, please file an issue with the custom component author

closed 2021-04-10 by OttoWinter