esphome/issues #1788: HomeAssistant restarts during upload /compiling

MarcHille, 2021-01-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HomeAssistant (RP3+)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

HomeAssistant ESPhome Plugin

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "XXXX"
  password: "XXXX"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/test.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/test
Processing test (board: d1_mini; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/test/.pioenvs/test/src/main.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiServer.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiServerSecureAxTLS.cpp.o
Compiling /data/test/.pioenvs/test/lib4d9/ESP8266WiFi/WiFiServerSecureBearSSL.cpp.o

```

**Additional information and things you've tried:**
all data downloaded from ha and compiled locally.
Reboot, reinstalled plugin, cleared cache

jperquin, 2021-01-25:
Rpi 3+ is not the most powerful SBC and 1Gb of memory gets eaten up rather quickly when compiling. See if you can temporarily stop all other integrations while compiling. That usually solves the problem.

If you want to get really adventurous try increasing your swap file size.

MarcHille, 2021-01-26:
The Raspberry really seems to be too weak. By disabling the other integrations I was able to continue compiling. Now a new rpi4 will come soon...

closed 2021-01-26 by MarcHille