esphome/issues #1790: Flicker ignoring primary color again

broyuken, 2021-01-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant core? (used to be hassio)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 in a Novostella NTB10 bulb

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

https://esphome.io/components/light/index.html#flicker-effect

**Description of problem:**
The flicker effect ignores the primary color and only flickers white. 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
# Common code for Novostella NTB10 Bulb

esphome:
  name: ${device_name}
  comment: ${device_description}
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: true

wifi:
  ssid: !secret guestwifissid
  password: !secret guestwifipass
  fast_connect: on #we only have one WiFi AP so just use the first one that matches
  power_save_mode: none
  ap: #since we listed an SSID above, this AP mode will only enable if no WiFi connection could be made
    ssid: ${friendly_name}_AP
    password: !secret guestwifipass

captive_portal:

# Enable logging
logger:
  baud_rate: 0 #disable UART logging since we aren't connected to GPIO1 TX

# Enable Home Assistant API
api:
  password: !secret esphome_adminpw
  
# Enable OTA updates
ota:
  password: !secret esphome_adminpw

# Enable web server
web_server:
  port: 80

binary_sensor:
# Reports if this device is Connected or not
- platform: status
  name: ${friendly_name} Status

sensor:
## Reports the WiFi signal strength
#- platform: wifi_signal
#  name: ${friendly_name} Signal
#  update_interval: 60s

# Reports how long the device has been powered (in minutes)
- platform: uptime
  name: ${friendly_name} Uptime
  filters:
    - lambda: return x / 60.0;
  unit_of_measurement: minutes

text_sensor:
# Reports the ESPHome Version with compile date
- platform: version
  name: ${friendly_name} ESPHome Version

output:
- platform: esp8266_pwm
  id: red
  pin: GPIO15
  inverted: False
  min_power: 0.000499
- platform: esp8266_pwm
  id: green
  pin: GPIO5
  inverted: False
  min_power: 0.000499
- platform: esp8266_pwm
  id: blue
  pin: GPIO12
  inverted: False
  min_power: 0.000499
- platform: esp8266_pwm
  id: cold_white
  pin: GPIO14
  inverted: False
  min_power: 0.000499
- platform: esp8266_pwm
  id: warm_white
  pin: GPIO4
  inverted: False
  min_power: 0.000499

light:
- platform: rgbww
  name: ${friendly_name}
  red: red
  green: green
  blue: blue
  cold_white: cold_white
  warm_white: warm_white
  cold_white_color_temperature: 6500 K
  warm_white_color_temperature: 2000 K
  id: thelight
  color_interlock: true
  restore_mode: ALWAYS_ON #Always turn light on when power restored.
  effects:
    - random:
    - strobe:
    - flicker:
        alpha: 95% #The percentage that the last color value should affect the light. More or less the “forget-factor” of an exponential moving average. Defaults to 95%.
        intensity: 1.5% #The intensity of the flickering, basically the maximum amplitude of the random offsets. Defaults to 1.5%.
    - lambda:
        name: Throb
        update_interval: 1s
        lambda: |-
          static int state = 0;
          auto call = id(thelight).turn_on();
          // Transtion of 1000ms = 1s
          call.set_transition_length(1000);
          if (state == 0) {
            call.set_brightness(1.0);
          } else {
            call.set_brightness(0.01);
          }
          call.perform();
          state += 1;
          if (state == 2)
            state = 0;
```

**Logs (if applicable):**

```
Not debugs because its a light but this is from the web interface
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=37%, Green=100%, Blue=0%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:327]:   Effect: 'Flicker'
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=80%, Blue=0%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=0%, Blue=71%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][sensor:092]: 'TBD Light Uptime': Sending state 25.78637 minutes with 0 decimals of accuracy
[D][light:265]: 'TBD Light' Setting:
[D][light:274]:   State: OFF
[D][light:304]:   Transition Length: 1.0s
[D][light:310]:   Effect: 'None'
[D][light:265]: 'TBD Light' Setting:
[D][light:274]:   State: ON
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=65%, Blue=27%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=65%, Blue=26%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=65%, Blue=26%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=70%, Blue=61%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=74%, Blue=65%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=93%, Blue=64%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=85%, Blue=67%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=83%, Blue=63%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=69%, Blue=0%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:287]:   Red=100%, Green=69%, Blue=0%
[D][light:290]:   White Value: 0%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:278]:   Brightness: 37%
[D][light:304]:   Transition Length: 1.0s
[D][light:265]: 'TBD Light' Setting:
[D][light:327]:   Effect: 'Flicker'
[D][light:265]: 'TBD Light' Setting:
[D][light:274]:   State: OFF
[D][light:304]:   Transition Length: 1.0s
[D][light:310]:   Effect: 'None'

```

**Additional information and things you've tried:**

regoras, 2021-01-26:
I think this is related to #1531. I had the same issue and turning off color_interlock made flicker work as expected. Unfortunately it meant that white adjustment became borked since it was mixing RGB into the warm and cool white.

broyuken, 2021-01-26:
yea, that's not an option for me as I use color_interlock. Since that one was closed hopefully it can be revisited in this one.

regoras, 2021-01-26:
Agreed, I left color_interlock: true since I didn't want the broken white values. Will just have to live without effects until this can get revisited.

rozza-m, 2021-03-11:
I closed #1886 as a duplicate of this. More logs available over there

rozza-m, 2021-09-16:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

This is still an issue. The only thing I use esphome for is lights and I'd really like a candlelight flicker again :-/

benni336, 2024-06-04:
I can confirm the issue with a cwww light and version 2024.3.0-dev.