esphome/issues #1791: RGBW light color correction through max_power has no effect

baracoder, 2021-01-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

"MagicHome" LED controller with esp8285

**ESPHome version (latest production, beta, dev branch)**

v1.15.3

**Affected component:**

light rgbw https://esphome.io/components/light/rgbw.html
output config https://esphome.io/components/output/index.html#config-output

**Description of problem:**
An RGBW light strip has blue tint. [Color correction](https://esphome.io/components/light/rgbw.html#color-correction) refers to setting `max_power` for individual outputs to correct. Setting different values does not affect light color, when setting rgb to white through home assistant.
[image]
 

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  - platform: esp8266_pwm
    pin: GPIO14
    id: pwm_r
  - platform: esp8266_pwm
    pin: GPIO5
    id: pwm_g
    max_power: 0.85
  - platform: esp8266_pwm
    pin: GPIO12
    id: pwm_b
    max_power: 0.7
  - platform: esp8266_pwm
    pin: GPIO13
    id: pwm_w

light:
  - platform: rgbw
    id: strip
    name: ${upper_name_1}
    red: pwm_r
    green: pwm_g
    blue: pwm_b
    white: pwm_w
    default_transition_length: 300ms
```

**Logs (if applicable):**

```
[D][light:287]:   Red=100%, Green=85%, Blue=71% <-- used those values for calibration
[D][light:287]:   Red=100%, Green=100%, Blue=100%  <- set to middle of color wheel in home assistant
```

**Additional information and things you've tried:**

CarlosGS, 2021-04-16:
I had the same problem - now adding the following note to the documentation :)

> Remember that ``gamma_correct`` is enabled by default (``γ=2.8``), and you may want take it into account for the calibration. For instance if you command a light to *50%* brightness and want it to be the new maximum: ``max_PWM_power = max_light_power^2.8 = 0.5^2.8 = 0.144``, then you would set ``max_power`` to *14.4%*.

lambdahindiii, 2021-04-17:
> I had the same problem, now adding a note to the documentation [esphome/esphome-docs#742](https://github.com/esphome/esphome-docs/pull/742)
> 
> > Remember that `gamma_correct` is enabled by default (`γ=2.8`), and you may want take it into account for the calibration. For instance if you command a light to _50%_ brightness and want it to be the new maximum: `max_PWM_power = max_light_power^2.8 = 0.5^2.8 = 0.144`, then you would set `max_power` to _14.4%_.
> 
> Also note that ESPHOME is always going to report the command scales from 0 to 100%, `max_power` has its effect internally.
> 
> In your case, when the blue channel is set to 100%, internally it sends 70% of power. That is not a perceptible difference due to gamma correction. For that channel you should be setting `max_power = 0.7^2.8 = 0.368`. Sounds very low, but that's how light perception works. Cheers!

I had the same problem with RGB light and was struggling to figure it out. This solved it for me completely. Thanks!

closed 2021-08-17 by baracoder