esphome/issues #1793: esp-wrover-kit (ttgo t-camera)  disconnects every hour at same time

Necrosaurus, 2021-01-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Core (various versions) on Raspberry pi 4b
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32 esp-wrover-kit (ttgo t-camera) 
**ESPHome version (latest production, beta, dev branch)**

latest production 1.15.3 

**Affected component:**

**Description of problem:**
I initially just thought the (cheap knockoff version) ttgo t-camera was just inherently unstable. removing almost everything (camera and sensors) from the yaml file improved the issue, (less disconnections) until I eventually noticed a pattern. With the most stable configuration i have found (with camera disabled, only a status sensor enabled), it disconnects almost every hour, at exactly the same time (+/- a few seconds). The esp32 usually disconnects for a second or 2 then reconnects. at irregular intervals (still on the same hourly pattern) it disconnects long enough to cause the esp32 to restart - i have set the timeout to 5mins. On a couple of occasions it has lasted many hours (12+) without disconnecting, but then returns by itself to the hourly disconnections.  If i enable the camera or any other sensors, it never exceeds the hourly disconnection mark.

This disconnection time is dependent on when i last restarted the router: ie the esp32 disconnects on the hourly anniversary of the routers wifi connection coming online (a few minutes after router restart). 

i have removed all commented out configuration lines (camera and sensors) from yaml file below.

the esphome (usb) log outputs vary. sometime it just says "Disconnecting Home Assistant" other times is says "_eventCallback(): Event: 5 - STA_DISCONNECTED"

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: affenhausenmonitor
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: ""
  password: ""
  fast_connect: true
  power_save_mode: light
  reboot_timeout: 5min

logger:
  level: verbose 

api:
  reboot_timeout: 5min

ota:

substitutions:
  devicename: esp-wrover-kit
  friendly_name: Affenhausen
  gap: 1min

binary_sensor:
  - platform: status
    name: $friendly_name Monitor Status

esp32_camera:
  name: $friendly_name Camera
  external_clock:
    pin: GPIO32
    frequency: 20MHz
  i2c_pins:
    sda: GPIO13
    scl: GPIO12
  data_pins: [GPIO5, GPIO14, GPIO4, GPIO15, GPIO18, GPIO23, GPIO36, GPIO39]
  vsync_pin: GPIO27
  href_pin: GPIO25
  pixel_clock_pin: GPIO19
  power_down_pin: GPIO26
```

**Logs (if applicable):**

```
[17:57:28][V][esp-idf:000]: I (2446178) wpa: GTK 2 is already in use in entry 3, it may be an attack, ignor it.
[17:57:28]
[17:57:29][V][esp-idf:000]: I (2447675) wpa: GTK 2 is already in use in entry 3, it may be an attack, ignor it.
[17:57:29]
[17:57:30][V][esp-idf:000]: I (2448699) wpa: GTK 2 is already in use in entry 3, it may be an attack, ignor it.
[17:57:31]
[17:57:33][V][esp-idf:000]: I (2451250) wifi: state: run -> init (6c0)
[17:57:33]
[17:57:33][V][esp-idf:000]: I (2451252) wifi: pm stop, total sleep time: 2116152725 us / 2443654829 us
[17:57:33]
[17:57:33]
[17:57:33][V][esp-idf:000]: I (2451256) wifi: n:13 0, o:13 0, ap:255 255, sta:13 0, prof:1
[17:57:33]
[17:57:33][D][api:067]: Disconnecting ESPHome v1.15.3 (192.168.1.13)[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[17:57:33][W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 6 - NOT_AUTHED
[17:57:33][W][wifi_esp32:386]: Event: Disconnected ssid='NSA' bssid=[redacted] reason='Not Authenticated'
[17:57:33]
[17:57:33][W][wifi:100]: WiFi Connection lost... Reconnecting...
[17:57:33][I][wifi:194]: WiFi Connecting to 'NSA'...

[19:56:21][V][api.connection:588]: Hello from client: 'Home Assistant 2021.1.3 (192.168.1.13)'
[19:56:21][D][api.connection:604]: Client 'Home Assistant 2021.1.3 (192.168.1.13)' connected successfully!
[19:56:22][D][api:067]: Disconnecting Home Assistant 2021.1.3 (192.168.1.13)
[19:57:37][W][AsyncTCP.cpp:963] _poll(): ack timeout 4
[19:57:37][V][api.connection:693]: Error: Disconnecting Home Assistant 2021.1.3 (192.168.1.13)
[19:57:37][D][api:067]: Disconnecting Home Assistant 2021.1.3 (192.168.1.13)
[19:57:37][D][binary_sensor:036]: 'Affenhausen Monitor Status': Sending state OFF
[20:02:37][E][api:084]: No client connected to API. Rebooting...

```

**Additional information and things you've tried:**
removing all sensors / camera
changing logger debug/ verbose /none
setting uart
dhcp /manual ip
power_save_mode: none /light
wiping esp32 with expressif downloadtool
using completely new esphome code
connecting over mqtt instead of api

mmakaay, 2021-03-18:
The "ack timeout 4" message followed by a disconnect might be a sign of a race condition that I discovered and [reported](https://github.com/me-no-dev/AsyncTCP/issues/116) for the AsyncTCP library.
I did a fix on the AsyncTCP-esphome library which works great for my devices. No more random disconnects with "ack timeout 4" errors. If you want to try the patch, then [check out this post](https://community.home-assistant.io/t/solved-esphome-dhcp-client-problems-temporary-solution-with-manual-fixed-ip/147312/37?u=mmakaay).

Necrosaurus, 2021-03-23:
Thank you, ill give it a try!

mmakaay, 2021-03-23:
If that is indeed the esphome config directory (sounds reasonable, I presume that is where your device's yaml file is stored?), then you are right. Create a folder /homeassistant/config/esphome/libs and extract my repo into that folder.

What value to use for the `lib_extra_dirs` option depends on how ESPHome is running.
I would recommend to simply use a relative path there, e.g. `lib_extra_dirs: ./libs`, instead of my solution which uses an absolute path. That makes it a lot more portable. I should have done that in my posting too, I'll fix that right away.

During the build you can easily see if my patched library is used or not by inspecting the dependency tree in the build output:

[image]

The red arrow shows the original library, while the green arrow shows the patched version.
If you only see the orginal library, then my version is not picked up somehow.

Necrosaurus, 2021-04-02:
thank you, trying now...

closed 2021-08-14 by stale[bot]