esphome/issues #1794: ESPHome doesn't work with DHT22 temp/humidity sensor

lhanson, 2021-01-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io running in a virtual machine on Proxmox

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 ([this one](https://www.amazon.com/gp/product/B07L8W9SP3))

**ESPHome version (latest production, beta, dev branch)**
1.15.3

**Affected component:**
https://esphome.io/components/sensor/dht.html

**Description of problem:**
I have multiple ESP8266 boards ([these](https://www.amazon.com/gp/product/B07L8W9SP3)) and multiple DHT22 temperature/humidity sensors ([these](https://www.amazon.com/gp/product/B08KRJ14NH))

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esphome_sensor_0
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "redacted"
  password: "redacted"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome Sensor 0"
    password: "redacted"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "[redacted]"

ota:
  password: "[redacted]"

sensor:
  - platform: dht
    model: DHT22
    pin: D5
    temperature:
      name: "Test Device Temperature"
    humidity:
      name: "Test Device Humidity"
    update_interval: 60s
```

**Logs (if applicable):**

Flashing using [esphome-flasher](https://github.com/esphome/esphome-flasher):
```
Using '/dev/ttyUSB0' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting....

Chip Info:
 - Chip Family: ESP8266
 - Chip Model: ESP8266EX
 - Chip ID: 0059366E
 - MAC Address: [redacted]
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
 - Flash Mode: dout
 - Flash Frequency: 40MHz
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.6s
Compressed 402000 bytes to 278900...
Writing at 0x00044000... (100 %)Wrote 402000 bytes (278900 compressed) at 0x00000000 in 6.3 seconds (effective 507.4 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[21:27:51][I][logger:166]: Log initialized
[21:27:51][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[21:27:51][I][app:029]: Running through setup()...
[21:27:51][C][dht:011]: Setting up DHT...
[21:27:51][C][wifi:033]: Setting up WiFi...
[21:27:51][D][wifi:324]: Starting scan...
[21:27:51][W][dht:162]: Requesting data from DHT failed!
[21:27:51][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[21:27:51][D][sensor:092]: 'Test Device Temperature': Sending state nan °C with 1 decimals of accuracy
[21:27:51][D][sensor:092]: 'Test Device Humidity': Sending state nan % with 0 decimals of accuracy
[21:27:57][D][wifi:339]: Found networks:
[21:27:57][I][wifi:385]: - '[redacted]' ▂▄▆█
[21:27:57][D][wifi:386]:     Channel: 11
[21:27:57][D][wifi:387]:     RSSI: -54 dB
[21:27:58][I][wifi:194]: WiFi Connecting to '[redacted]'...
[21:28:00][I][wifi:457]: WiFi Connected!
[21:28:00][C][wifi:303]:   SSID: '[redacted]'
[21:28:00][C][wifi:304]:   IP Address: 192.168.120.135
[21:28:00][C][wifi:306]:   BSSID: ...
[21:28:00][C][wifi:307]:   Hostname: 'esphome_sensor_0'
[21:28:00][C][wifi:311]:   Signal strength: -60 dB ▂▄▆█
[21:28:00][C][wifi:315]:   Channel: 11
[21:28:00][C][wifi:316]:   Subnet: 255.255.255.0
[21:28:00][C][wifi:317]:   Gateway: 192.168.120.1
[21:28:00][C][wifi:318]:   DNS1: 192.168.120.1
[21:28:00][C][wifi:319]:   DNS2: (IP unset)
[21:28:00][D][wifi:466]: Disabling AP...
[21:28:00][C][ota:029]: Over-The-Air Updates:
[21:28:00][C][ota:030]:   Address: esphome_sensor_0.local:8266
[21:28:00][C][ota:032]:   Using Password.
[21:28:00][C][api:022]: Setting up Home Assistant API server...
[21:28:00][I][app:059]: setup() finished successfully!
[21:28:00][I][app:105]: ESPHome version 1.15.3 compiled on Jan 28 2021, 18:33:15
[21:28:00][C][wifi:443]: WiFi:
[21:28:00][C][wifi:303]:   SSID: 'Curvy Fudgehammer'
[21:28:00][C][wifi:304]:   IP Address: 192.168.120.135
[21:28:00][C][wifi:306]:   BSSID: [redacted]
[21:28:00][C][wifi:307]:   Hostname: 'esphome_sensor_0'
[21:28:00][C][wifi:311]:   Signal strength: -60 dB ▂▄▆█
[21:28:00][C][wifi:315]:   Channel: 11
[21:28:00][C][wifi:316]:   Subnet: 255.255.255.0
[21:28:00][C][wifi:317]:   Gateway: 192.168.120.1
[21:28:00][C][wifi:318]:   DNS1: 192.168.120.1
[21:28:00][C][wifi:319]:   DNS2: (IP unset)
[21:28:00][C][logger:185]: Logger:
[21:28:00][C][logger:186]:   Level: DEBUG
[21:28:00][C][logger:187]:   Log Baud Rate: 115200
[21:28:00][C][logger:188]:   Hardware UART: UART0
[21:28:00][C][dht:017]: DHT:
[21:28:00][C][dht:018]:   Pin: GPIO14 (Mode: INPUT)
[21:28:00][C][dht:024]:   Model: DHT22 (or equivalent)
[21:28:00][C][dht:027]:   Update Interval: 60.0s
[21:28:00][C][dht:029]:   Temperature 'Test Device Temperature'
[21:28:00][C][dht:029]:     Unit of Measurement: '°C'
[21:28:00][C][dht:029]:     Accuracy Decimals: 1
[21:28:00][C][dht:029]:     Icon: 'mdi:thermometer'
[21:28:00][C][dht:030]:   Humidity 'Test Device Humidity'
[21:28:00][C][dht:030]:     Unit of Measurement: '%'
[21:28:00][C][dht:030]:     Accuracy Decimals: 0
[21:28:00][C][dht:030]:     Icon: 'mdi:water-percent'
[21:28:00][C][captive_portal:169]: Captive Portal:
[21:28:00][C][ota:029]: Over-The-Air Updates:
[21:28:00][C][ota:030]:   Address: esphome_sensor_0.local:8266
[21:28:00][C][ota:032]:   Using Password.
[21:28:00][C][api:095]: API Server:
[21:28:00][C][api:096]:   Address: esphome_sensor_0.local:6053
[21:28:45][W][dht:162]: Requesting data from DHT failed!
[21:28:45][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[21:28:45][D][sensor:092]: 'Test Device Temperature': Sending state nan °C with 1 decimals of accuracy
[21:28:45][D][sensor:092]: 'Test Device Humidity': Sending state nan % with 0 decimals of accuracy
```

**Additional information and things you've tried:**
This is very similar, if not identical, to https://github.com/esphome/issues/issues/792 (closed without resolution). Following in the footsteps of the reporter of that ticket (@WMTaylor3), I've tried several different scenarios without success:
* Specifying the model of the DHT. I've attempted implicit autodetection, DHT22, and AM2302
* Tried various iterations of the same configuration on different ESP8266 boards with multiple DHT22 sensors.
* Uploading OTA vs using a USB serial adaptor.
* Trying different pins on the board
* Per the comments in https://github.com/esphome/issues/issues/792, I've tried power cycling after flashing

I have been able to successfully read values from same hardware using [my own custom firmware based on platformio](https://gitlab.com/lyle_/esp8266-temp-humidity), so I'm reasonably confident that this is a software issue particular to ESPHome.

This is my first foray into ESPHome so I'm not very familiar with its internals, but I've got a little bit of familiarity with writing code for these boards on PlatformIO so I'm happy to assist in debugging however I can.

pvk071, 2021-01-29:
According to reviews on the sensor seller's page, this is not a DHT22 / AM2302.
Have you tried specifying other types of sensors?

I have a lot of DHT22 / AM2302 sensors, they all work fine with ESPHome 1.15.3 / NodeMcu v2. Autodetection does not work for me the same as yours, I needed to specify them as AM2302. You can try to specify other types of sensors, this should not damage your hardware.

lhanson, 2021-02-03:
@pvk071 Thanks, I think you're correct that these are knockoff sensors of some kind and I'm slightly embarrassed that I didn't read those same reviews on Amazon and come to a similar conclusion. I can confirm that these don't work as any other type of sensor in ESPHome. The fact that they work with other code makes me think there's something slightly flaky in their timing that I'm still curious to uncover and may capture some oscilloscope traces to see if there's a way to make ESPHome a little more resilient to shoddy sensors, so hopefully we can keep this issue open for discussion for a few weeks while I get new hardware and compare.

To others who may come across this issue (as I did with #792): if no software solution follows here eventually in the ESPHome ecosystem and you're still trying to get a temperature/humidity sensor to work, here's what you do:

* Leave a bad review wherever you bought these trashy sensors because they don't conform to the most basic [protocol specs](https://web.archive.org/web/20210203012445/http://www.ocfreaks.com/basics-interfacing-dht11-dht22-humidity-temperature-sensor-mcu/)
* Either throw away your fake sensor or try using different firmware which might be more tolerant of its shortcomings (the [Adafruit library](https://github.com/adafruit/DHT-sensor-library) used with either the Arduino or PlatformIO ecosystems seems to be excellent).

stonics, 2021-02-17:
I had a similar issue with an AM2302 sensor that I bought from [ebay](https://www.ebay.com/itm/DHT22-Digital-Temperature-Humidity-Sensor-AM2302-Measure-Module-SHT15-TE907/224260046517?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D20160908105057%26meid%3D6f17dec824c24465acab1593290140d4%26pid%3D100675%26rk%3D1%26rkt%3D15%26mehot%3Dnone%26sd%3D224260046517%26itm%3D224260046517%26pmt%3D1%26noa%3D1%26pg%3D2380057%26brand%3DUnbranded&_trksid=p2380057.c100675.m4236&_trkparms=pageci%3Ab0643d5f-70b6-11eb-8442-ee7c9584accd%7Cparentrq%3Aad60fe821770a9e0f27d25f7ffd84682%7Ciid%3A1). I flashed esphome to a NodeMCU ESP32 dev board and could not get it to read. I tried to specifying DHT22 and AM2302 as a sensor model, both of which did not work. 

The sensor reads correctly if I connect it to a raspberry pi or the ESP32 flashed with tasmota.

SenexCrenshaw, 2021-02-20:
This is such a common component and I am sure its just a sequence issue. I bought one of these from eBay (comes in a week) and I will investigate.  I bet the tasmota developer had one of these when it was developed.

stonics, 2021-02-20:
I got it working on mine. Turns out I needed to set the sensor model to DHT22_TYPE2

SenexCrenshaw, 2021-02-22:
@lhanson please try model DHT22_TYPE2

ragenhe, 2021-02-23:
Defining sensor model as DHT22_TYPE2 fixed my issue, which was on cold boot of module, it wasn't reporting temp accurately.  Now it does.

lhanson, 2021-02-27:
That works for me as well. Thanks!

closed 2021-02-27 by lhanson