esphome/issues #1801: OTA failing with error "[Error 32] Broken pipe" on ESP8266 device when captive_portal option is active

jojo141185, 2021-01-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome latest beta running on Linux Kubuntu 20.04 and stable in docker on Ubuntu 18.04. 

**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP8266 
board: modwifi (also tried with esp01_1m)

**ESPHome version (latest production, beta, dev branch)**

- Testing: Version: 1.16.0b4 (installed with pip)
- Stable: Docker esphome/esphome

**Affected component:**

The device is a Shelly 2.5 with 2 MB flash.

**Description of problem:**
The problem occurs with the attached yaml configuration once uploaded with enabled the captive_portal option.
After a successful firmware build, OTA update with ultra slow progress fails (in almost all cases randomly between 8-50%). The ESP reboots and I receive error message "**[Error 32] Broken pipe**". Unfortunately I can't flash via usb or create a serial log because the device is already fixed in the wall. Huge effort to remove it again.
Luckily, in very rare cases the OTA upload runs through completely. It feels like it takes about 1000 tries - I could tell that it works best right after the device reconnects to WIFI.

**Problem-relevant YAML-configuration entries:**
```yaml
https://gist.github.com/jojo141185/1a7bc1f9359056eae4d1c53ed6535434

```

**Logs (if applicable):**

Maybe someone with a Shelly 2.5 can try to reproduce the problem and upload a serial log.
```

```

**Additional information and things you've tried:**
No problem with disabled captive_portal option. WIFI connection is strong.
Problem seems to be related to yaml configuration or device. OTA runs without problems on other boards (ESP8266) on same esphome installation also with captive_portal activated. In case of error I could not access the captive portal either. After logging into the AP of ESP8266 a blank page appears.

It seems like something is interfering with the connection or interrupting the update process. Maybe the device with active captive portal is overloaded. 

Do you have an explanation for the problem?

gretel, 2021-02-05:
do you have `status_led` declared?

jojo141185, 2021-02-05:
@gretel Yes, I did on line 288 in my config.
Could this lead to the described problem in connection with active captive_portal option?

gretel, 2021-02-05:
@jojo141185 just an idea cause of https://github.com/esphome/issues/issues/1756

jojo141185, 2021-02-07:
I can't see any disconnections with active status_led declaration in operation, as in the issues linked by gretel. 
I was brave and uploaded my configuration with removed status_led declaration and active captive_portal. OTA ran fortunately again at the first attempt but the process was still unusually slow.
So removing the status_led seems to improve the situation, but the OTA process still seems to be disturbed in some way.

ElGranLoky, 2021-03-04:
Same problem with a sonoff bridge 433 with captive portal active.

pki791, 2021-03-14:
> Same problem with a sonoff bridge 433.
> 
> After Disable:
> 
> * Captive portal
> * status led
> * binary_sensor status
> 
> Works fine.

One or all of these?

ElGranLoky, 2021-03-14:
At first disable these options:

- Captive portal
- status led

And then finish adding binary_sensor status, so that it will work.

closed 2021-07-26 by stale[bot]

tescher, 2021-09-01:
I realize this is closed, but I still see this issue randomly on my devices at 2 different locations. Was it resolved?

mkono87, 2021-09-22:
> 
> 
> I realize this is closed, but I still see this issue randomly on my devices at 2 different locations. Was it resolved?

Also have this issue with a plug I have not used for a few months. Powers and operates but does not display wattage anymore nor OTA updates